iiab/iiab #1673: Gitea 1.8.0+ install on Raspbian but don't run

holta, 2019-05-17:
@aidan-fitz Gitea {1.7.5, 1.7.6, 1.8.0, 1.8.1} all work on Ubuntu 18.04, and all also install on Raspbian on RPi 3 B+ &mdash; but versions 1.8.0 and 1.8.1 fail to run on Raspbian on RPi 3 B+ &mdash; with the following message, showing Gitea's systemd service appears to be constantly restarting, i.e. thrashing:
```
root@box:~# systemctl status gitea
● gitea.service - Gitea (Git with a cup of tea)
   Loaded: loaded (/etc/systemd/system/gitea.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Fri 2019-05-17 01:55:45 EDT; 279ms ago
  Process: 2853 ExecStart=/library/gitea/bin/gitea-1.8.1 web -c /etc/gitea/app.ini (code=killed, signal=BUS)
 Main PID: 2853 (code=killed, signal=BUS)

May 17 01:55:45 box.lan systemd[1]: gitea.service: Unit entered failed state.
May 17 01:55:45 box.lan systemd[1]: gitea.service: Failed with result 'signal'.
```
In comparison, Gitea 1.7.6 (like 1.7.5) both installs and work on Raspbian on RPi 3 B+, showing the following:
```
root@box:~# systemctl status gitea
● gitea.service - Gitea (Git with a cup of tea)
   Loaded: loaded (/etc/systemd/system/gitea.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-05-17 02:06:11 EDT; 1min 3s ago
 Main PID: 8781 (gitea-1.7.6)
   CGroup: /system.slice/gitea.service
           └─8781 /library/gitea/bin/gitea-1.7.6 web -c /etc/gitea/app.ini

May 17 02:06:11 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:11 [I] [SQL] SELECT `user`.id FROM `user` WHERE `user`.num_repos!=(SELECT COUNT(*) FROM `repository` WHERE o
May 17 02:06:11 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:11 [I] [SQL] SELECT `issue`.id FROM `issue` WHERE `issue`.num_comments!=(SELECT COUNT(*) FROM `comment` WHER
May 17 02:06:13 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:13 [I] [SQL] SELECT repo.id FROM `repository` repo WHERE repo.num_closed_issues!=(SELECT COUNT(*) FROM `issu
May 17 02:06:13 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:13 [I] [SQL] SELECT repo.id FROM `repository` repo WHERE repo.num_forks!=(SELECT COUNT(*) FROM `repository`
May 17 02:06:14 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:14 [I] SQLite3 Supported
May 17 02:06:14 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:14 [I] Run Mode: Development
May 17 02:06:14 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:14 [I] [SQL] SELECT `id`, `repo_id`, `hook_id`, `uuid`, `type`, `url`, `payload_content`, `content_type`, `e
May 17 02:06:14 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:14 [I] [SQL] SELECT `id`, `type`, `status`, `issue_id`, `index`, `head_repo_id`, `base_repo_id`, `head_user_
May 17 02:06:16 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:16 [I] Listen: http://0.0.0.0:61734/gitea
May 17 02:06:16 box.lan gitea-1.7.6[8781]: 2019/05/17 02:06:16 Serving [::]:61734 with pid 8781
```
Is this an upstream failure possibly, and if so can you report it?

FYI /library/gitea/log/http.log is empty &mdash; what log files are likely most useful in solving this, to inform the above bug report?

(And/Or should we consider reverting to Gitea 1.7.6 for [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes)'s release in June?)

holta, 2019-05-30:
@aidan-fitz do you know if yesterday's Gitea 1.8.2 has any better hope of working on Rapberry Pi?

Can you interpret the ongoing situation at https://github.com/go-gitea/gitea/pull/7037 and https://github.com/go-gitea/gitea/pull/7042 for us?

aidan-fitz, 2019-05-31:
Sorry, I don't actually have a Raspberry Pi on me that I can test with. I've been away from GitHub for a while, and I'll let them know about the problem ASAP.

aidan-fitz, 2019-06-01:
@holta Their response: Use the ARM6 build. 1.8.2 should run correctly.

aidan-fitz, 2019-06-01:
@holta Is this issue safe to close now?

holta, 2019-06-01:
> @holta Is this issue safe to close now?

Yes, thanks.  PR #1713 does the job.

closed 2019-06-01 by holta