iiab/iiab #1675: Validate Kolibri 0.12.4 on its release "June" ?

holta, 2019-05-18:
Building on PR #1608 (Kolibri 0.12.3)

Kolibri 0.12.4 Milestone:
https://github.com/learningequality/kolibri/milestone/36

holta, 2019-06-12:
Kolibri 0.12.4 was released earlier today:
https://github.com/learningequality/kolibri/releases/tag/v0.12.4

It will likely take a few days for them to publish their .pex (pip installer) as usual: https://learningequality.org/download/

holta, 2019-06-13:
> Kolibri 0.12.4 was released earlier today:
> https://github.com/learningequality/kolibri/releases/tag/v0.12.4
> 
> It will likely take a few days for them to publish their .pex (pip installer) as usual: https://learningequality.org/download/

@benjaoming Kolibri 0.12.4's .pex (pip install) is now available, and installs on Ubuntu 18.04, but fails completely to start its systemd service:

> TASK [kolibri : Enable & (Re)Start kolibri service] ****************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service kolibri: Job for kolibri.service failed because the control process exited with error code.\nSee \"systemctl status kolibri.service\" and \"journalctl -xe\" for details.\n"}

A bit more detail here:
```
root@box:~# systemctl start kolibri
Job for kolibri.service failed because the control process exited with error code.
See "systemctl status kolibri.service" and "journalctl -xe" for details.

root@box:~# systemctl restart kolibri
Job for kolibri.service failed because the control process exited with error code.
See "systemctl status kolibri.service" and "journalctl -xe" for details.

root@box:~# systemctl status kolibri
● kolibri.service - Kolibri
   Loaded: loaded (/etc/systemd/system/kolibri.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-06-13 15:44:55 EDT; 1s ago
  Process: 18855 ExecStart=/usr/local/kolibri/bin/kolibri start (code=exited, status=1/FAILURE)
 Main PID: 1585 (code=killed, signal=TERM)

Jun 13 15:44:55 box.lan kolibri[18855]:   File "/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/apps/registry.py", line 116, in populate
Jun 13 15:44:55 box.lan kolibri[18855]:     app_config.ready()
Jun 13 15:44:55 box.lan kolibri[18855]:   File "/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/morango/apps.py", line 23, in ready
Jun 13 15:44:55 box.lan kolibri[18855]:     max_parameter_substitution()
Jun 13 15:44:55 box.lan kolibri[18855]:   File "/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/morango/util.py", line 106, in max_parameter_substitution
Jun 13 15:44:55 box.lan kolibri[18855]:     with open(SQLITE_VARIABLE_FILE_CACHE, 'w') as file:
Jun 13 15:44:55 box.lan kolibri[18855]: IOError: [Errno 2] No such file or directory: '/home/kolibri/SQLITE_MAX_VARIABLE_NUMBER.cache'
Jun 13 15:44:55 box.lan systemd[1]: kolibri.service: Control process exited, code=exited status=1
Jun 13 15:44:55 box.lan systemd[1]: kolibri.service: Failed with result 'exit-code'.
Jun 13 15:44:55 box.lan systemd[1]: Failed to start Kolibri.
```

Note this was installing Kolibri 0.12.4 on top of 0.12.3 &mdash; as part of IIAB 7.0 &mdash; using the same approach that always worked in the past:
```
cd /opt/iiab/iiab
sudo ./runrole kolibri
```

Aside, @KW4NP & all please also continue to keep an eye on Kolibri's longstanding login blockage:

#1532 http://box/kolibri causes browser to infinitely loop, preventing login (works initially...server reboot does not help...happens with all client OS's & all server OS's)

holta, 2019-06-13:
I've mentioned this upstream too, just in case: https://github.com/learningequality/kolibri/issues/5664

benjaoming, 2019-06-13:
As a work-around, could you create a home directory for your `kolibri` user `/home/kolibri` and chown it to that user? (I saw you mentioned creating the home directory in the upstream issue, but it also needs to be writable).

holta, 2019-06-13:
> As a work-around, could you create a home directory for your `kolibri` user `/home/kolibri` and chown it to that user? (I saw you mentioned creating the home directory in the upstream issue, but it also needs to be writable).

FWIW that seemed to work (when @lyw07 asked me to do that earlier) even without the directory being writable:
```
root@box:~# ls -ld /home/kolibri/
drwxr-xr-x 2 kolibri root 4096 Jun 13 17:47 /home/kolibri/
root@box:~# ls -l /home/kolibri/
total 4
-rw-r--r-- 1 kolibri www-data 3 Jun 13 17:47 SQLITE_MAX_VARIABLE_NUMBER.cache
```

(In any case, great news that 0.12.5 should be released next week, Thanks All!)

holta, 2019-06-14:
Closing this tkt in favor in anticipation of Kolibri 0.17.5 (#1736)

closed 2019-06-14 by holta

benjaoming, 2019-06-14:
@holta take this lightly, I know you have a very complex setup -- but it's normal for a process to expect that its environment has a writable and existing home directory :)

benjaoming, 2019-06-14:
@holta glad you got it working, looking forwards to more IIAB updates :heart: