iiab/iiab #1680: Tracking tkt for "set system time from browser time"

holta, 2019-05-19:
https://github.com/iiab/iiab-admin-console/pull/204

holta, 2019-05-23:
@tim-moody, to document optional disabling of this auto-time-fixing feature in http://FAQ.IIAB.IO, what should we write?

@avni, please see the revised testing procedure @ https://github.com/iiab/iiab-admin-console/pull/204#issuecomment-495379846

holta, 2019-05-30:
@avni please see Item 2.d.ii. in today's (May 30, 2019) http://minutes.iiab.io

In short...we'd really like you to help make sure Randy Tate & all in Uganda are happy with IIAB's experimental new feature whereby RPi/IIAB servers' clocks are now auto-set by nearby iPhones, Androids, etc &mdash; this works when the IIAB is not using ntp i.e. truly offline &mdash; and the IIAB's clock appears more than 2 min behind the times etc.

Note the RPi/IIAB's clock is never moved backward, only forward, per the current implementation.  So there will occasionally be a risk that its clock will "permanently" jump ahead too many hours/days/years if a nearby phone has its clock set wrong (!?)

So...after you've tested this feature (instructions @ #1680) please also try disabling the feature entirely as follows, e.g. if deployments/schools want a more unhackable-if-often-tardy clock:

- Admin Console (http://box.lan/admin) > Content Menus > Menu Properties > Properties of Current Menu > uncheck "Check to Allow Power Off Link." (does not require Ansible run thankfully!)
- in the unlikely event apache_allow_sudo really was False in /etc/iiab/local_vars.yml, then you’d need an Ansible run

RSVP @avni please if possible!  Then we can document to http://FAQ.IIAB.IO with your impressions well prior to IIAB 7.0's release around June 20th.

PS @tim-moody does Linux's `fakeclock` presumably continue to work, even when you disable the above, such that the system clock is saved to disk during most poweroffs for the next bootup?

holta, 2019-06-27:
@avni thanks for please testing the above on your new Raspberry Pi 4 Friday night (in ~48h from now) if you can!

holta, 2019-07-01:
@avni can you follow the instructions @ https://github.com/iiab/iiab-admin-console/pull/204#issuecomment-495379846 prior to IIAB 7.0's release, to test out & confirm that auto-sync'ing of RPi 4's clock works, when a nearby iPhone begins browsing?

avni, 2019-07-07:
Did the 10 minute turn-off test today. Some interesting results. 

Prior to shutting off: 
pi@box:~ $ date
Sun 07 Jul 2019 04:02:07 PM EDT

After restarting, prior to browsing: 
pi@box:~ $ date
Sun 07 Jul 2019 03:20:12 PM EDT

After browsing: 
pi@box:~ $ date
Sun 07 Jul 2019 04:32:18 PM EDT

avni, 2019-07-07:
Useragent: AppleWebKit/605.1.15 (KHTML, like Gecko) Version/12.1.1 Safari/605.1.15

avni, 2019-07-07:
Did another 10 minute test with iOS. Same results:

after turning on: pi@box:~ $ date
Sun 07 Jul 2019 04:32:58 PM EDT

after accessing via iPhone iOS Safari:
pi@box:~ $ date
Sun 07 Jul 2019 05:25:23 PM EDT

holta, 2019-07-08:
@avni thanks so much for having confirmed that RPi IIAB clocks are updated automatically by nearby (recent?) iPhones and MacBook both, when the standard browser on those client devices are used!

If a mischievous kid abuses this feature (e.g. any kid can set the school's clock to 2030, or just an hour forward...saved by the bell ;) please note you can turn it off this entire feature in http://box.lan/admin > `Content Menus` > `Menu Properties` > `Check to Allow Browser to Update Server Time.`

@tim-moody:

1) Are there any other browsers/devices that should be tested?

2) Does Linux's `fake-hwclock` respect manual uses of the `date` command bringing the time "back into the present", if the clock has been mischievously set "back to the future" by mischievous students/pranksters?

3) Is it true that most all Linux browsers are ignored and/or are there others too?

holta, 2019-07-09:
> @tim-moody:
>
> 1. Are there any other browsers/devices that should be tested?
> 2. Does Linux's `fake-hwclock` respect manual uses of the `date` command bringing the time "back into the present", if the [...]

4. Does clock-updating work with all port 80 URL's?  Or only a subset, like IIAB's main page i.e. http://box, http://box.lan, and http://172.18.96.1 ?

holta, 2019-07-09:
The following paragraph was just added to [IIAB 7.0's DRAFT Release Notes](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes):

> Update your IIAB's clock automatically even when it's offline, when nearby smartphones or laptops/devices browse to your IIAB. This helps low-end Raspberry Pi servers that contain no RTC (real-time clock), so they don't completely lose track of the current day and time.

tim-moody, 2019-07-09:
any mobile devices can be tried.
fake clock writes the time every 10 min I think. http://manpages.ubuntu.com/manpages/xenial/man8/fake-hwclock.8.html
3. yes
updating only works on the home page (if js-menu used) not on any others

holta, 2019-07-09:
> fake clock writes the time every 10 min I think. http://manpages.ubuntu.com/manpages/xenial/man8/fake-hwclock.8.html

Small Correction: on Raspbian `fake-hwclock` runs 17 min past every hour:

```
root@box:~# more /etc/cron.hourly/fake-hwclock
#!/bin/sh
#
# Simple cron script - save the current clock periodically in case of
# a power failure or other crash

if (command -v fake-hwclock >/dev/null 2>&1) ; then
  fake-hwclock save
fi
```
```
root@box:~# grep hourly /etc/crontab
17 *    * * *   root    cd / && run-parts --report /etc/cron.hourly
```
> 2. Does Linux's `fake-hwclock` respect manual uses of the `date` command bringing the time "back into the present", if the clock has been mischievously set "back to the future" by mischievous students/pranksters?

@avni @tim-moody let's test this some day, just to be 100% sure!

holta, 2019-07-09:
FYI @avni's evidence suggests that `fake-hwclock` fails to run during graceful shutdowns (certainly on her RPi 4) even despite its `man fake-hwclock` documentation advertising that feature prominently:

> The design expectation of fake-hwclock is that it will be run very late at shutdown and very early at boot.

Curious Aside: RPi's will almost always boot 17min past the hour, as a consequence of the above (:

holta, 2019-07-09:
> updating only works on the home page (if js-menu used) not on any others

@tim-moody does RPi's clock get updated by browsers coming in over Ethernet/localhost too?  (Or just browsers coming in over WiFi?)

jvonau, 2019-07-09:
I can't reproduce /etc/fake-hwclock.data failing to update with a reboot. As a data point could someone compare the behavior between reboot, shutdown, poweroff?

holta, 2019-07-09:
> I can't reproduce /etc/fake-hwclock.data failing to update with a reboot. As a data point could someone compare the behavior between reboot, shutdown, poweroff?

Good idea.  Let's try on 10.8.0.78 later, which is an RPi 4 essentially identical to @avni's with the #1855 issue.

tim-moody, 2019-07-09:
> does RPi's clock get updated by browsers coming in over Ethernet/localhost too? (Or just browsers coming in over WiFi?)

any access to the page

holta, 2019-07-15:
This design/operations tkt is very useful to help gather feedback on how "offline" communities are facing laggardly clock issues, with or without an RTC (real-time clock) installed.

So I'm keeping it open and moving it to the IIAB 7.1 Milestone to help us keep refining this feature and its associated documentation.

holta, 2019-12-21:
Certainly good enough for now!

closed 2019-12-21 by holta