iiab/iiab #1687: dnsmasq won't start on Anish's Ubuntu Server 18.04.2 NUC

holta, 2019-05-22:
As far as I know the main symptoms/regressions compared to February 2019 are:

- dnsmasq does not start on @m-anish's NUC ()
- pbx.lan no longer resolves to 172.18.96.1

Thanks @m-anish for explaining more the symptoms you're experiencing, alongside these diagnostic/log files:

/opt/iiab/iiab/iiab-install.log including results of the network role which ran successfully on 1st install:
http://paste.ubuntu.com/p/kkpQyzyxdJ/

/etc/iiab/iiab.env
http://paste.ubuntu.com/p/Q25tjwskwk/

/etc/iiab/iiab.ini
http://paste.ubuntu.com/p/wv76mKxgMq/

/etc/iiab/local_vars.yml
http://paste.ubuntu.com/p/yDZHDfqY9J/

ip a
http://paste.ubuntu.com/p/3DBrVdFxv3/

brctl show
http://paste.ubuntu.com/p/p3vk3t4WFk/

journalctl -u dnsmasq

holta, 2019-05-22:
@m-anish @jvonau: can you figure out why `unknown interface br0` keeps appearing in this NUC's "journalctl -y dnsmasq" @ http://paste.ubuntu.com/p/3xmyV6PMPM/ ?

FWIW `brctl show` output (later?) appears as: (according to http://paste.ubuntu.com/p/p3vk3t4WFk/)
```
bridge name	bridge id		STP enabled	interfaces
br0		8000.b20d7b660c33	no		eno1
							wlp58s0
```

holta, 2019-05-22:
@m-anish: can you clarify more if dnsmasq continues to run after you manually start it?

(Putting your paste of `systemctl status dnsmasq` at http://paste.ubuntu.com/p/stM6vH9qxD/ in further context please?)

PS what aspects of dnsmasq work / don't work when it **is** running?

m-anish, 2019-05-23:
> @m-anish: can you clarify more if/how dnsmasq continues to run after you manually start it?
> 
> (Putting your paste of `systemctl status dnsmasq` at http://paste.ubuntu.com/p/stM6vH9qxD/ in further context please?)
> 
> PS what aspects of dnsmasq work / don't work when it **is** running?

http://paste.ubuntu.com/p/qSPzHMm9PB/ is the status when dnsmasq starts

I think dhcp works when it starts. Still can't resolve pbx.lan. What other things should I test. 

Also, it seems that the resolution of box.lan also takes a few seconds.

m-anish, 2019-05-23:
I should also mention that when the NUC is booting up, it spends upto two minutes at the following status:

A start job is running for Wait for network to be configured (XXX/no limit)

holta, 2019-05-23:
> I should also mention that when the NUC is booting up, it spends upto two minutes at the following status:
> 
> A start job is running for Wait for network to be configured (XXX/no limit)

FWIW this is the exact same message we saw on Ubuntu 19.04 a month ago, prior to @floydianslips & @jvonau's workaround for #1586 here: PR https://github.com/iiab/iiab/pull/1633, PR https://github.com/iiab/iiab/pull/1634

holta, 2019-05-23:
@m-anish are any of these below related?

#1306 IIAB 6.7/master fails to install on latest Ubuntu 18.04: "Unable to start service dnsmasq"
#1549 dnsmasq issue(s) on IIAB 7.0/master e.g. Ubuntu 18.04.2 on ThinkPads?
#1569 dnsmasq sometimes fails to start on Raspbian Desktop & Lite
#1572 LAN issue with bridge (br0) on Ubuntu 18.04
PR #1636 Increase hostapd_wait from 1 to 5 sec (?) at end of roles/network (restart.yml) for dnsmasq

holta, 2019-05-23:
1) @m-anish since you had this all working 3-4 months ago in January/February, can you recreate that using 18.04.1 (not 18.04.2) *without* running `apt update`, to see what the differences are?

2) It likely doesn't matter, but it remains confusing why you're kernel is 4.15 rather than 4.18 (which we were told would be installed as a part of a fresh installs of Ubuntu 18.04.2 &mdash; whereas upgrades to 18.04.2 definitely continue to run 4.15 just like 18.04.0 and 18.04.1).  So please do post the exact ISO or .img you're using to install your OS.

holta, 2019-05-23:
We're trying to understand right now during our live weekly IIAB call (http://minutes.iiab.io) to see if Ubuntu 18.04.2 itself contain some networking glitches.

@m-anish can you post onto this ticket?
- /etc/resolv.conf
- /etc/hosts
- /etc/systemd/system/hostapd.service
- /lib/systemd/system/hostapd.service
- ls -l /etc/systemd/system | pastebinit

m-anish, 2019-05-23:
Sure. will do!

holta, 2019-05-23:
> Sure. will do!

Thanks.  @m-anish you said 2min boot delay occurred as part of Ubuntu 18.04.2 even prior to installing IIAB &mdash; as such can you please describe exactly how you installed the [apparently glitchy] OS?

holta, 2019-05-23:
Also @m-anish keep in this mind https://github.com/iiab/iiab/wiki/IIAB-Networking#dns that @jvonau helped with from 2018-12-16 to 2019-01-17:

"some Linux distributions do not function well with dnsmasq's default service. Several of these systemd unit files have a timing problem. As a result, IIAB uses the **iiab-dnsmasq**[*] service instead — to start dnsmasq at the correct time during boot. IIAB disables the OS's stock dnsmasq unit file across all distros to be safe."

[*] https://github.com/iiab/iiab/blob/master/roles/network/tasks/enable_services.yml#L51-L73

m-anish, 2019-05-24:
> > Sure. will do!
> 
> Thanks. @m-anish you said the 2 min boot delay occurred as part of Ubuntu 18.04.2 even prior to installing IIAB — as such can you please describe exactly how you installed the [apparently glitchy] OS?

It was a standard liveusb install. I first created the liveusb from iso, then booted the PC from it. During install, it detected two network interfaces (LAN, USB-LAN), and I just did the base install + openssh server.

holta, 2019-05-30:
FYI @m-anish will try to download a new ISO of Ubuntu Server 18.04.2 and try again, sometime in the coming days.

(Apparently installing in this method gets you kernel 4.18, instead installing older ISO's like 18.04.1 and then upgrading, which gets you kernel 4.15)

holta, 2019-06-18:
@m-anish we need to close out [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) issues now as final release is imminent &mdash; is it time to defer this dnsmasq issue to IIAB 7.1 &mdash; or is this now sufficiently resolved?

holta, 2019-06-23:
@m-anish please run `/usr/bin/iiab-diagnostics` to post more detailed diagnostics where possible / when possible towards solving this in coming days?

As explained here:
https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md

Per PR #1763

holta, 2019-06-27:
@m-anish please re-open if this is still an issue 5 weeks later?  Thanks!

closed 2019-06-27 by holta