iiab/iiab #1689: IIAB main page (http://box) sometimes shows nothing, expect the "turn off" button

holta, 2019-05-23:
Partial report from @mir123 in Panama:

> Sometimes opening the main page showed nothing, except the "turn off" button! Refreshing the page would sometimes work to bring up the page. So in all cases the box ended up off after some random new user clicked that hoping it would do something. Haven't yet tried to reproduce the issue, probably happened when too many users in at the same time? Trying to remove the turn off link from admin console did not work. Also haven't had time to reproduce issue.

holta, 2019-05-23:
@mir123 you should be able to disable the "Power Off" button by setting `apache_allow_sudo: False` within `/etc/iiab/local_vars.yml` just prior to IIAB install.  If IIAB is already installed, you may choose to reinstall by running: `cd /opt/iiab/iiab; ./iiab-install --reinstall`

Or use IIAB's Admin Console (http://box.lan/admin) to hide the Power Off button (which might not also disable the URL action, but @tim-moody can clarify?)

Both approaches are summarized here:
http://FAQ.IIAB.IO Item 24 ('Is a "Rapid Power Off" button possible for low-electricity environments?')

tim-moody, 2019-05-24:
I just tried turning off the poweroff button in the admin console and it worked. make sure you refresh the page and make sure it does not load from cache. N.B. that removing the poweroff link via the admin console does not change the apache allow sudo settings, the link still works. it just isn't on the menu.

holta, 2019-06-27:
@mir123 is there any news here?

(Or should we close this tkt in coming weeks, if you're not encountering this any longer?)

holta, 2019-07-02:
> @mir123 is there any news here?
> 
> (Or should we close this tkt in coming weeks, if you're not encountering this any longer?)

@mir123 am moving this to the IIAB 7.1 milestone.

Consider closing this tkt if it's not a serious/common/reproducible issue?

holta, 2019-10-11:
@mir123 please re-open & explain if this is still an issue!

closed 2019-10-11 by holta