iiab/iiab #1699: IIAB Install Fails on node.js Step Even With node.js Installed

Tjstanton, 2019-05-26:
Running a Rasberry Pi Zero W with ARM6l running Rasberian Stretch v. April 2019, Kernel 4.14. When we install IIAB v7 we get the following error:

fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "curl -sL https://deb.nodesource.com/setup_10.x | bash -", "delta": "0:00:01.369624", "end": "2019-05-26 11:38:46.313227", "msg": "non-zero return code", "rc": 1, "start": "2019-05-26 11:38:44.943603", "stderr": "", "stderr_lines": [], "stdout": "\n## Installing the NodeSource Node.js 10.x repo...\n\n\n## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.", "stdout_lines": ["", "## Installing the NodeSource Node.js 10.x repo...", "", "", "## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later."]}

We have installed node.js from binary tarballs (https://nodejs.org/dist/v10.15.3/node-v10.15.3-linux-armv6l.tar.gz) as instructed and still get the error. Latest version is 10.15.3 and still failed. Downgraded to node-v6.17.1 and still get the same message. Tried IIAB v 6.7 and still getting the same error.

Stumped.

holta, 2019-05-26:
Hi @Tjstanton, which IIAB services/apps are you installing that require Node.js on RPi Zero W?

I don't have the answer but want to understand better what you're trying to achieve (in what country, for what kinds of students / doctors, with what organizations etc) so that others can possibly help you try to work this out.  Perhaps you're already aware of the "Known Issue" at the bottom of IIAB 7.0's DRAFT Release Notes here?

https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes#known-issues

PS you clearly know what you're doing, so please also write me (holt @ laptop . org) if you also want to join our email/discussion lists for IIAB implementers/installers, where others might possibly know more about this.

holta, 2019-05-26:
> Hi @Tjstanton, which IIAB services/apps are you installing that require Node.js on RPi Zero W?

Background: Typically IIAB is installed onto a microSD card that is (temporarily) inside an RPi 3 or RPi 3 B+ during installation and content  curation — and then later, the microSD is moved into a RPi Zero W — sometime prior to delivery/deployment.

Can you do that?
(If not, what's the reason?)

Tjstanton, 2019-05-27:
We have ordered a RPi3 and we will try that approach when it arrives. We ordered the RPi Zero W based on the IIAB/FAQ (http://wiki.laptop.org/go/IIAB/FAQ#What_hardware_should_I_use.3F). We tried a clean install yesterday (wipe card, re-install Rasberian, updated the node.js, ran the install, and still got the same error so we will go this route. Right now we are simply looking to build a prototype platform to distribute open source scholarly content so our goals are pretty modest.

Tjstanton, 2019-05-30:
We got both RPi 3 and RPi B+ and tried installs, had numerous crashes that I attributed to a possible fault in how we formatted the SSD, but we also had a number of fails having to do with tarball corruption. Notably, node.js was not an issue. Rather frustrated, we tried a new SSD format tool (SDCard Formatter) and after yet another failure on IIAB 6.7 install decide we'd shoot the moon on 7, went to dinner, returned, and low and behold it works like a charm. No idea why. I have the prior error msgs if anyone cares. Will carefully document what we did for future users. Arduous install, possibly more a RPi OS issue than IIAB but that's our story at this moment.

Tjstanton, 2019-05-30:
RPi 3 had no issues with node.js install, this can be closed.

closed 2019-05-30 by Tjstanton

holta, 2019-05-30:
> RPi 3 had no issues with node.js install, this can be closed.

Thanks @Tjstanton &mdash; keep in touch with your IIAB needs.

Particularly after IIAB 7.0 Preview 1 is released tomorrow, showcasing a couple of [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes)'s new Vector Maps etc!