iiab/iiab #1704: Satellite tiles in z0-z9 package missing -- cause local blur

georgejhunt, 2019-05-30:
I wrote a script to validate each blob in the sqlite3 database, to see if it could be loaded successfully as a png. The results were informative:
Opening database ../output/stage2/satellite_z0-z9.mbtiles
bad 0 ok 0 empty 0 html 0 unfixable 0 zoom 0 replaced 0
bad 0 ok 1 empty 0 html 0 unfixable 0 zoom 1 replaced 0
bad 0 ok 10 empty 0 html 0 unfixable 0 zoom 2 replaced 0
bad 32 ok 27 empty 0 html 32 unfixable 0 zoom 3 replaced 0
bad 179 ok 105 empty 0 html 179 unfixable 0 zoom 4 replaced 0
bad 510 ok 735 empty 0 html 510 unfixable 0 zoom 5 replaced 0
bad 2008 ok 3206 empty 0 html 2008 unfixable 0 zoom 6 replaced 0
bad 6830 ok 14513 empty 0 html 6830 unfixable 0 zoom 7 replaced 0
bad 24398 ok 61970 empty 0 html 24398 unfixable 0 zoom 8 replaced 0
bad 75202 ok 272287 empty 0 html 75202 unfixable 0 zoom 9 replaced 0
bad 75202 ok 272287 empty 0 html 75202 unfixable 0
The number after indicates the problem. I counted the number of tiles which had the html header "DOCTYPE html". There were 75K out of 272K that were not really images. Some, maybe all, may have been ocean, and not really relevant.  But when I tried to write the program to fetch only the ones which seemed to have html content rather than image content, the program failed to function as expected. 

There are 3 possibilities going forward:

1. Live with the current level of blurring, hoping that in the next two weeks I can understand, and fix, the current z0-z9 file.
2. Dial back the satellite file  to z0-z7, which seems less frustraing, if the area of interest is blurred.
3. Go back to the "ocean.mbtiles" based upon a world map png, which gets pretty blurred at z5, and becomes just a background color at higher zooms. -- And remove the layer switcher entirely.

holta, 2019-05-30:
> 1. Live with the current level of blurring, hoping that in the next two weeks I can understand, and fix, the current z0-z9 file.

Seems reasonable, especially as user surveys might also determine that occasional blurring is ugly but acceptable for IIAB 7.0?

holta, 2019-05-31:
Related: #1711 Vector Map double click unreliable

holta, 2019-06-06:
@georgejhunt has fixed this, as will be reflected in new map packs that will contain revised satellite layers 0-9 (or similar!) based on his latest here:

https://archive.org/download/satellite_z0-z9_v3.mbtiles/satellite_z0-z9_v3.mbtiles (931MB)

closed 2019-06-06 by holta