iiab/iiab #1707: failed: "Add any MAPS that are installed to home/menu.js"

holta, 2019-05-30:
@georgejhunt @tim-moody do you have a workaround to get me get past this?  Context: I'm running `./install` in `/opt/iiab/iiab-admin-console` with latest master &mdash; and this failure occurs every timeon an Ubuntu 18.04 NUC:

> TASK [js-menu : Add any MAPS that are installed to home/menu.js] ***************
[DEPRECATION WARNING]: evaluating osm_vector_maps_enabled as a bare variable,
this behaviour will go away and you might need to add |bool to the expression
in the future. Also see CONDITIONAL_BARE_VARS configuration toggle.. This
feature will be removed in version 2.12. Deprecation warnings can be disabled
by setting deprecation_warnings=False in ansible.cfg.
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/usr/bin/iiab-update-map", "delta": "0:00:00.206343", "end": "2019-05-30 12:42:48.601849", "msg": "non-zero return code", "rc": 1, "start": "2019-05-30 12:42:48.395506", "stderr": "Traceback (most recent call last):\n  File \"/usr/bin/iiab-update-map\", line 204, in <module>\n    main()\n  File \"/usr/bin/iiab-update-map\", line 64, in main\n    item = map_catalog['regions'][region]\nKeyError: 'min'", "stderr_lines": ["Traceback (most recent call last):", "  File \"/usr/bin/iiab-update-map\", line 204, in <module>", "    main()", "  File \"/usr/bin/iiab-update-map\", line 64, in main", "    item = map_catalog['regions'][region]", "KeyError: 'min'"], "stdout": "[\n  \"en-osm-omt_north_asia\", \n  \"en-map_test\", \n  \"en-osm-omt_south_america\", \n  \"en-osm-omt_central_america\", \n  \"en-osm-omt_south_asia\", \n  \"en-osm-omt_oceania\", \n  \"en-osm-omt_san_jose\"\n]\n['min_2017-07-03_v0.1']", "stdout_lines": ["[", "  \"en-osm-omt_north_asia\", ", "  \"en-map_test\", ", "  \"en-osm-omt_south_america\", ", "  \"en-osm-omt_central_america\", ", "  \"en-osm-omt_south_asia\", ", "  \"en-osm-omt_oceania\", ", "  \"en-osm-omt_san_jose\"", "]", "['min_2017-07-03_v0.1']"]}

tim-moody, 2019-05-30:
The problem is a mismatch between min, San Jose, test

holta, 2019-05-30:
When I ran `/usr/bin/iiab-update-map` on @tim-moody's suggestion, the result was:

> root@box:~# /usr/bin/iiab-update-map
[
  "en-osm-omt_north_asia",
  "en-map_test",
  "en-osm-omt_south_america",
  "en-osm-omt_central_america",
  "en-osm-omt_south_asia",
  "en-osm-omt_oceania",
  "en-osm-omt_san_jose"
]
['min_2017-07-03_v0.1']
Traceback (most recent call last):
  File "/usr/bin/iiab-update-map", line 204, in <module>
    main()
  File "/usr/bin/iiab-update-map", line 64, in main
    item = map_catalog['regions'][region]
KeyError: 'min'

holta, 2019-05-30:
Map Packs appears to be folders like `en-osm-omt_min_2017-07-03_v0.1` stored in:
```
/library/www/osm-vector-maps
```
So @tim-moody recommended I delete all/old map packs and try `./install` again...

This worked, Thanks!

_Where's the best document this in the original error message &mdash; to keep others on track the next time this happens &dash; so we (like Ansible) move to more actionable error messages?_

tim-moody, 2019-05-30:
line 64 add check for region in map_catalog['regions']
if not issue skipping message and continue

holta, 2019-05-30:
> line 64 add check for region in map_catalog['regions']
> if not issue skipping message and continue

Line 64 of what file?

(Are you talking about https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/tasks/main.yml#L64 ?)

holta, 2019-05-31:
Here's an _Early Rough Draft_ of George's techniques to manually update an IIAB map (pack, typically one of 10 "continents") so implementers don't get stuck.

Part 1:

> 1. ./runrole osm-vector-maps to download the new catalog.
> 2. delete the entry for the map that should be replaced in /library/www/html/common/assets/vector-map-idx.json 
> 3. reboot server
> 4. Check the checkbox in admin console and click the download button
> 
> Tim has explained that our code quietly skips installing items that are already installed -- and for kiwix, that seems like the best strategy, given that the number of items, and long filenames, might easily lead to duplicate installs, eating up valuable SD card space.
>
> Some other regions, have had their titles changed by Adam [and LinuxFest Northwest volunteers], but the download url remains the same and the older versions of the those regions work. -- did not try world, North America, or Europe. 
>
> I've pinged Tim so that we can identify why the backend refuses to update a map without removal of an item from vecter-map-idx.json. I think we understand it now. We can discuss what to do about it next week.

Part 2:

> 1. download a new map catalog: "wget http://download.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json -O /library/www/osm-vector-maps/maplist/assets/regions.json"
> 2. Download a new bboxes.geojson: "wget http://download.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/bboxes.geojson -O /library/www/osm-vector-maps/maplist/assets/bboxes.geojson"
> 3. Manually edit the file /library/www/html/common/assets/vector-map-idx.json, removing the item that needs to be updated.

@tim-moody & @georgejhunt clarify:

>> deleting from  vector-map-idx.json doesn't remove the map from the file system. is the new download a direct replacement, does it have the same name?
>
> The replacement map has a new url: and therefore a new filename within /library/www/osm-vector-maps. 
>
> > with kiwix the content of a zim with a given name will always be the same. the perma ref or generic name can refer to more than one zim, but a zim file is always one version. if there is a new version of  Africa, Middle East, and Central America do the zips have new names?
>
> Yes

Ref: #1710 Living Docs for IIAB 7.0 Maps

holta, 2019-06-07:
@georgejhunt just explained to me that:
- Deleting Map Pack(s) from /library/www/osm-vectors/maps is not quite enough is not enough, as upgrading a map pack might still be blocked by Admin Console if name's are the same
- George has a PR he's working on, that will compare Map Pack names, for Tim to review to add to Admin Console
- For now, one also needs to edit /library/www/html/common/assets/vector-map-idx.json &mdash; an example of which is here:
```
{
  "en-osm-omt_africa": {
    "menu_item": "en-osm-omt_africa",
    "language": "en",
    "file_name": "en-osm-omt_africa_2017-07-03_v0.2",
    "region": "africa",
    "date": "2017-07-03",
    "size": "10890784768"
  },
  "en-osm-omt_central_america": {
    "menu_item": "en-osm-omt_central_america",
    "language": "en",
    "file_name": "en-osm-omt_central_america_2017-07-03_v0.21",
    "region": "central_america",
    "date": "2017-07-03",
    "size": "3768221696"
  },
  "en-osm-omt_middle_east": {
    "menu_item": "en-osm-omt_middle_east",
    "language": "en",
    "file_name": "en-osm-omt_middle_east_2017-07-03_v0.2",
    "region": "middle_east",
    "date": "2017-07-03",
    "size": "7851638784"
  },
  "en-osm-omt_san_jose": {
    "menu_item": "en-osm-omt_san_jose",
    "language": "en",
    "file_name": "en-osm-omt_san_jose_2017-07-03_v0.2",
    "region": "san_jose",
    "date": "2017-07-03",
    "size": "1415360512"
  }
}
```
Refs:
#1710 Living Docs for IIAB 7.0 Maps [OSM]
#1725 http://iiab.me/maps (or similar!) public demo site needed for latest/evolving IIAB 7.0 Maps

holta, 2019-06-07:
Related stuff:

PR https://github.com/iiab/iiab-admin-console/pull/211 Download if filename is changed
#1612 How can IIAB operators install the latest /library/www/osm-vector/maplist/assets/regions.json ? (catalog of downloadable maps, e.g. from d.iiab.io) [document workarounds for IIAB 7.0]

holta, 2019-06-18:
For IIAB 7.0, the instructions in this new "IIAB Maps" doc are likely sufficient:

https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack

holta, 2019-07-14:
This bug is largely solved.

Remaining issues can be handled in #1791 #1793 #1800

closed 2019-07-14 by holta