iiab/iiab #1708: Understanding IIAB's Admin Console (box.lan/admin) > Content Menus (tab)

holta, 2019-05-30:
- [ ] Are the 2 buttons on the left (under "Content Menu Tasks") supposed to do something when clicked?  What are they supposed to do?
  - Edit Content Menus
  - Edit Menu Items
- [ ] Are these 2 orange/white tabs near the top supposed do something when clicked?  What are they supposed to do?
  - Menu Properties
  - Content Item List
- [ ] Can the 2 words "Check box" be removed from all 12 checkboxes under Admin Console (http://box.lan/admin) > Content Menus > Menu Properties > Properties of Current Menu ?  (These 12 sentences are quite long, can they be made shorter?)
  - [ ] Also consider dropping the period from the end of each of those 12 sentences?

holta, 2019-05-31:
Ok I've partially figured out items 1 and 4 above, i.e. what to do when the "Select Menu Item" orange/white tab at the bottom of http://box.lan/admin > Content Menus does nothing but show you "current menu list here":

- scroll back up to the top of the page
- click "Edit Menu Items" on the left
- scroll all the way back to the bottom of the page
- blue buttons appear, allowing you to "Edit" or "Clone" each menu items

Quite confusing...maybe that dummy/stub text ("current menu list here") could be revised to tell you what to do?

tim-moody, 2019-05-31:
I now understand what you're talking about. This does not behave at all as it used to. Somewhere there is a code regression.

holta, 2019-06-03:
Many of the above issues were solved on Friday by:

https://github.com/iiab/iiab-admin-console/pull/205

holta, 2019-07-15:
Refinements & more importantly screencasts/videos are needed here, perhaps in the month after IIAB 7.0 is released in July 2019 hopefully?

In any case, I'm moving this tkt to the [IIAB 7.1 Milestone](https://github.com/iiab/iiab/milestone/6).