iiab/iiab #1711: Vector Map double click unreliable

georgejhunt, 2019-05-31:
I think this is related to the asynchronous nature of openlayers javascript library.  When there are many requests pending for tiles, the check for existence of a tile at the next zoom level does not get a timely response, and double-click does not function properly (though plus and minus at upper left do work, and have been my fall back). 

In my testing the failure to respond to a double-click at zoom less than 9 has been self correcting.

I think the queue of requested tiles may also be the underlying cause of the blurring, which I find objectionable for some adjacent tiles.  But they also seem to self correct most of the time. I think the problem of html content in a sqlite3 blob, where there should be image content, is the next technical problem to address. 

I don't yet know how to verify a tile blob that should contain vectors. But to improve the user experience, the ratio of good to bad tiles needs to be addressed, and perhaps improved. But it's is also very confusing. I have concluded in one context that a tile was bad, that worked on another machine, in another network context -- so timing, and server loading may be involved. 

I'm wondering if maps will work at all in a classroom of 30 students all hitting the same server at the same time with time requests.

holta, 2019-05-31:
Related: #1704 Satellite tiles in z0-z9 package missing -- cause local blur

holta, 2019-06-06:
@georgejhunt the IIAB coming together for Ghana earlier this week seems to have this problem.

Even with a single user, our Ghana partner was getting frustrated with double clicks (seemingly) not working when trying to zoom into the latest "Africa" map packs.

Is the underlying pattern sufficiently understood?  Or are there diagnostics recipes/pastebins that others might provide you & all in order to understand this issue better?

georgejhunt, 2019-06-06:
There is a javascript async call to exists.php service on the server. That
"exists.php" server responds yes/no. I could change the logic such that if
there was no timely response, zoom went ahead ( the current strategy is
only to permit a zoom if there is a yes response).

holta, 2019-06-06:
> There is a javascript async call to exists.php service on the server. That "exists.php" server responds yes/no. I could change the logic such that if there was no timely response, zoom went ahead ( the current strategy is only to permit a zoom if there is a yes response).

Go wild if you think this has a shot!

PS let's post this or similar to a public demo site (as outlined in #1725) so that we can cast an increasingly wider, net with brand new people of different kinds from a growing number of countries, delivering their most sincere feedback from IIAB 7.0 Preview 1 to be announced in coming days.

georgejhunt, 2019-06-07:
I updated iiab.me with the new satellite image file, and the fix for
unreliable zooming. There are africa, central_america, san_jose, and
middle_east regions all accessible from the home page.

holta, 2019-06-07:
> I updated iiab.me with the new satellite image file, and the fix for unreliable zooming. There are africa, central_america, san_jose, and middle_east regions all accessible from the home page.

This solves #1725, Thanks @georgejhunt!  Having all 4 clickable at the top of http://iiab.me/maps (as you've set up) will make IIAB 7.0 community/beta testing (coming soon) dramatically more meaningful.

Out of curiosity...
- Why is [Central America](http://iiab.me/maps/en-osm-omt_central_america_2017-07-03_v0.21/) version v0.21 while the other 3 map packs are v0.2 ?
- When you first visit [Central America v0.21](http://iiab.me/maps/en-osm-omt_central_america_2017-07-03_v0.21/), is it possible the displayed bounding box (high-resolution region, that is much more zoomable) can be placed in the center of the page?  In my experience on desktop & mobile, the displayed bounding box appears at the bottom of the screen.
- Zooming in about 10-or-15 levels causes Chrome 74.0.3729.157 on Android 8.1.0 to crash every time, do you know if this is a JavaScript issue?  (Typically 2-to-8 zoom levels prior to the browser crashing, the visible map region shrinks to 3/4 of the screen, with the right-side and bottom-side of the screen being all white.)

georgejhunt, 2019-06-07:
> I updated iiab.me with the new satellite image file, and the fix for
> unreliable zooming. There are africa, central_america, san_jose, and
> middle_east regions all accessible from the home page.
>
> This solves #1725 <https://github.com/iiab/iiab/issues/1725>, Thanks
> @georgejhunt <https://github.com/georgejhunt>! Having all 4 clickable at
> the top of http://iiab.me/maps (as you've set up) will make IIAB 7.0
> community/beta testing (coming soon) dramatically more meaningful.
>
> Out of curiosity...
>
>    - Why is Central America
>    <http://iiab.me/maps/en-osm-omt_central_america_2017-07-03_v0.21/>
>    version v0.21 while the other 3 map packs are v0.2 ?
>
> I had already published a central america image as v0.2, and uploaded the
region.json to ukids.
-- so I needed to bump the version so the new satellite.mbtiles would be
downloaded (because the filename needs to be unique).

>
>    - When you first visit Central America v0.21
>    <http://iiab.me/maps/en-osm-omt_central_america_2017-07-03_v0.21/>, is
>    it possible the displayed bounding box (high-resolution region, that is
>    much more zoomable) can be placed in the center of the page? In my
>    experience on desktop & mobile, the displayed bounding box appears at the
>    bottom of the screen.
>
> I can set the center, and zoom level for each region. But screen size and
centering is also affected by OS.

>
>    - Zooming in about 10-or-15 levels causes Chrome 74.0.3729.157 on
>    Android 8.1.0 to crash every time, do you know if this is a JavaScript
>    issue? (Typically 2-to-8 zoom levels prior to the browser crashing, the
>    visible map region shrinks to 3/4 of the screen, with the right-side and
>    bottom-side of the screen being all white.)
>
> I have not seen my chrome get confused zooming.

I set iiab.me central america one initial zoom higher, and recentered on my
screen (the green bbox is not visible initially because it is zoomed one
notch higher)

holta, 2019-06-07:
> - Zooming in about 10-or-15 levels causes Chrome 74.0.3729.157 on Android 8.1.0 to crash every time, do you know if this is a JavaScript issue? (Typically 2-to-8 zoom levels prior to the browser [...]

> I have not seen my chrome get confused zooming.

Have you tried Chrome on Android rather than on a laptop?  (As folks use in the developing world, which seems to crash every time, for me anyway!)

georgejhunt, 2019-06-07:
Yes my chrome on android crashes also. And the whole experience on my
android is disappointing. the search window occupies the whole top of the
page. And the popup keyboard prevents selection of dropdown found cities.
Maybe search does not make sense for smart phones.

holta, 2019-06-27:
@georgejhunt can you provide a quick update on where we stand here?

Thanks!  (FYI I've just moved this tkt to the [IIAB 7.1 Milestone](https://github.com/iiab/iiab/milestone/6))

holta, 2019-12-27:
@georgejhunt what's the status here?

Can we close this or do we leave this open as an IIAB 7.2 target?

holta, 2020-09-12:
@georgejhunt can we close this ticket?

(And/or open another ticket specifically for Android/mobile browsing if OSM maps / satellite photos as appropriate!)

georgejhunt, 2020-09-12:
no longer a problem

closed 2020-09-12 by georgejhunt