iiab/iiab #1715: 8 map packs show green rectangles, 3 others do not (North Asia, South Asia, Oceania)

holta, 2019-06-02:
@georgejhunt is it expected that 3 "continents" {North Asia, South Asia, Oceania} behave differently than the other 8 "continents", when hovering over (mousing over) their names on the lest side of http://box.lan/admin -> Install Content -> Get Map Region ?

FWIW this is a fresh install of IIAB on Ubuntu 19.04 VM 10.8.0.38, as outlined at #1714.

(CLARIF: please do not start an actual map download to this VM, as this is hosted by family who pays a large amount per gigabyte downloaded!)

georgejhunt, 2019-06-02:
Yes it is expected. When you changed the regions.json, you broke everything
that was already uploaded to archive.org with respect to the bounding boxes.

holta, 2019-06-02:
Bounding boxes changed April 26-28 thanks to hard work by volunteers at https://linuxfestnorthwest.org thinking through many field ramifications.

Catalog transitional phases are fully normal during such times, so that the global public gets increasingly more of the map packs they're asking for.

A less confusing downloader experience during these multi-week transition periods should be a part of IIAB 7.1's Admin Console.  It should probably start by delineating just one and only one thing:

(a) Which maps are available to download at that moment in time.

(All other text and bounding box visuals confuse end-user implementers.)

(b) Whereas which maps are already installed...can effectively be communicated in plain English, no fancy visuals required:

e.g. a prominent sentence at the top of the Admin Console "Get Map Region" page saying something like "To see what Map Packs you already have installed, please look in /library/www/osm-vector-maps [or wherever!]  Also browse http://box/maps to explore what regions are zoomable to how many levels."

IIAB ~7.2 can improve on this if really absolutely required, with visual bounding boxes in a different color showing what's already installed.  @tim-moody has ideas to substantially improve in this with multiple colors, should that be necessary.

(c) Maps that are not available to download (for whatever reason) should not be listed or shown in the downloader.

Regardless whether Map Pack was renamed, bounding box improved, became stale/deprecated/replaced, was wrong to begin with, or is a new region not yet uploaded to archive.org (or any other reason map cannot or ought not be downloaded) — these details greatly confuse end-user implementers.

Of course if a map is *accidentally* not downloadable or not findable for one of any number of different reasons, despite our best attempts, IIAB 7.1 could produce human-readable / actionable error messages explaining the probable cause(s), and if any recourse might be possible.

tim-moody, 2019-06-02:
a) if not available for download should not be in catalog and so will not appear in download list

georgejhunt, 2019-06-02:
as far as I know, all of the regions visible on the admin console are
downloadable

holta, 2019-06-02:
> as far as I know, all of the regions visible on the admin console are downloadable

Great.  Whether you mean 8 visible-in-green or 11 visible-in-text-on-left, can we sync up the catalog now with what is downloadable, so beta-testers can understand this?

i.e. can we make some kind of precedent/policy that green rectangles + text-on-left should both correspond to what's actually downloadable going forward?

georgejhunt, 2019-06-02:
If you are asking me to remove the regions from the catalog, where the
region names have been changed, I can do that.

holta, 2019-06-02:
> If you are asking me to remove the regions from the catalog, where the region names have been changed, I can do that.

Great: whether Admin Console offers 8 map packs or 11 map packs doesn't much matter during Beta 1. 
 Either is fine.  Let's just set a precedent going forward that the the text-on-left and the green-rectangles-on-the-right (which I assume are both coming from the catalog, currently regions.json) should both represent what's actually offered for download, at any particular moment in time.

For Beta 2 / Preview 2 we should likely try to get most/all 11 map packs in place for final testing.

For IIAB 7.1 or 7.2 (?) the catalog should likely be downloadable directly from a version-controlled GitHub-like site?  That would seem wise in order to...

1. Foster increased community engagement with direct transparency, bringing down barriers to entry so that others learn [how to make/contribute/evolve diverse Map Packs](https://github.com/georgejhunt/maps/blob/simple/generate-regions/readme.md) (if we play our cards right!)
2. Increase uptime + significantly reduce the frequency of infra exploits, as offered by large-scale cloud providers and their 24x7 security/IT teams, so that IIAB and our contributors can focus on delivering much higher value humanitarian/learning tools (low-level IT/devops/systems administration tasks can be all-consuming; where possible anyway let's delegate menial tasks to enterprise platforms that do a better job than we do!)

Ref: #1625

georgejhunt, 2019-06-13:
This issue should be resolved with the new archive.org packages, and the corresponding changes to regions.json

holta, 2019-06-13:
> This issue should be resolved with the new archive.org packages, and the corresponding changes to regions.json

Great news.  [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) Preview 1 (or Release Candidate 1?) arriving in coming days will be very impressive as a result of @georgejhunt's mapmaking work across many months here.

closed 2019-06-13 by holta