iiab/iiab #1717: openvpn@xscenet fails to use /etc/iiab/openvpn_handle with OpenVPN 2.4.6 ?

holta, 2019-06-03:
Despite /etc/iiab/openvpn_handle being properly set on Ubuntu Server 19.04 IIAB install 10.8.0.42, OpenVPN handle (e.g. its machine name set for the VPN) is unfortunately not being circulated within @georgejhunt's XSCENET infrastructure &mdash; despite several reboots and several attempts at commands like:

- systemctl restart openvpn
- systemctl restart openvpn@xscenet

Anything else that should be tried on this 10.8.0.42 machine?!  Background:

- Ubuntu 19.04 uses OpenVPN 2.4.6
- Ubuntu 18.04 uses OpenVPN 2.4.4
- Raspbian used OpenVPN 2.4.0

holta, 2019-06-03:
If someone can test IIAB's OpenVPN on Debian 10 ([that contains the most recent OpenVPN 2.4.7](https://packages.debian.org/buster/openvpn)) that would be tremendously useful !!

Ref: #1387 "Debian 10 Buster: remaining issues/workarounds e.g. with Kolibri & MongoDB"

georgejhunt, 2019-06-03:
I fixed 10.8.0.42 by running "apt install ncat". This suggests to me that
ubuntu 19 has broken "ncat" out of the "nmap" package which we install.
I we are not to break other OS's, we need to do more of the same
"is_ubuntu19", or as Tim suggested, come up with a better strategy.

holta, 2019-06-03:
> I fixed 10.8.0.42 by running "apt install ncat". This suggests to me that ubuntu 19 has broken "ncat" out of the "nmap" package which we install. [If] we are not to break other OS's, we need to do more of the same "is_ubuntu19" [or similar]

Miraculous.  I'll work on this when I get back in front of a more real computer in coming days.

holta, 2019-06-03:
Self-Reminder: let's test IIAB & its OpenVPN on the latest daily/nightly build of Debian 10 too.

Per: #1387

holta, 2019-06-12:
1) @georgejhunt can you call me today to outline what's likely needed in a PR here?

2) Debian 10 Buster will be released in a few short weeks FYI ([likely July 6th](https://www.phoronix.com/scan.php?page=news_item&px=Debian-10.0-Buster-Release-Date)) so I will be asking everyone who can to test here too.

holta, 2019-06-12:
Many Thanks to @georgejhunt who's helping to uncover the situation here:

- it's likely there are 2+ somewhat different versions of ncat, competing to be the most viable (as packaged by different folk)
- George will try to make sense of the situation and make a recommendation in coming days/coming weeks

holta, 2019-06-13:
See possible fix from @georgejhunt::

"Ncat"
PR #1732

holta, 2019-06-16:
Appears fixed by PR #1732

closed 2019-06-16 by holta