iiab/iiab #1719: All 15 links are broken in OER2Go/RACHEL module menu-defs/en-afristory.html

holta, 2019-06-04:
Good news is this 1 single high-level links works to showcase 5 stories: (in English)

http://iiab.me/modules/en-afristory/

Bad news is all 15 links displayed by its menuDef here are broken -- likely this file needs fixing or replacing:

https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-afristory.html

FYI these 15 broken links are quite prominent as a result of http://oer2go.org/viewmod/en-afristory being the #1 OER2Go/RACHEL module offered right on top of Admin Console's downloader, i.e. http://box.lan/admin > Install Content > Get OER2Go(RACHEL) Modules

tim-moody, 2019-06-04:
don't know where the html file came from, but completely unrelated to this module.

tim-moody, 2019-06-04:
should be

<ul>
<li><a href="<?php echo $dir ?>/firstwords.html">First Words</a></li>
<li><a href="<?php echo $dir ?>/firstsentences.html">First Sentences</a></li>
<li><a href="<?php echo $dir ?>/firstparagraphs.html">First Paragraphs</a></li>
<li><a href="<?php echo $dir ?>/longerparagraphs.html">Longer Paragraphs</a></li>
<li><a href="<?php echo $dir ?>/readaloud.html">Read Aloud</a></li>
</ul>

holta, 2019-06-16:
Appears fixed by https://github.com/iiab/iiab-admin-console/pull/210

closed 2019-06-16 by holta