iiab/iiab #1721: Logos for Wikiquote & Wikispecies appear broken at http://box; Wiktionary shows too few entries

holta, 2019-06-04:
Likely the 2 missing/broken logo images have something to do with these menuDefs, which also need explanatory/clean descriptions:  (what end users will see on their schools IIAB home page, e.g. http://box )

https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wikiquote_en_all.json
https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wikispecies_en_all.json

Wiktionary's incorrect count of total entries shows "5,309,521" whereas the ZIM itself shows "5,630,670" &mdash; the description etc should be suitably adjust here:

https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wiktionary_en_all_novid.json

Likewise this one shows "4,772,815" and should be suitable adjusted:

https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wiktionary_en_all.json

tim-moody, 2019-06-05:
> (2) Wiktionary's incorrect count of total entries shows "5,309,521" whereas the ZIM itself shows "5,630,670" — the description etc should be suitably adjust here:

These have the wrong number because they were hard coded. Instead of changing the number use substitution variables or we can never catch up.

Also the convention is to put this in the Footnote.

tim-moody, 2019-06-06:
Ncwikiquote2.png and 102px-Wikispecies_001.png are both in the current repo

holta, 2019-06-15:
Largely solved by https://github.com/iiab/iiab-admin-console/pull/210

Further refined by https://github.com/iiab/iiab-admin-console/pull/212 that will hopefully be merged soon.

closed 2019-06-16 by holta