iiab/iiab #1723: What's the best way to restore Content Pack metadata in small font / human-readable Spanish?

holta, 2019-06-04:
(1) What's the best way to restore small font metadata under most every Content Pack e.g. for IIAB pages like these below?

- http://iiab.me/peru
- http://iiab.me/chiapas

(2) What's the best way to restore metadata descriptions in localized/friendly language that teachers understand?  Examples above show what was refined in 2017 and 2018 across several Latin American countries and Africa too.

Specifically, teachers and implementers' understanding & curricular design greatly benefit when they have immediate access to the 2-or-3 (roughly!) metadata bits that are most appropriate for each particular kind of content pack &mdash; from a longer list of ~10 metadata types described in ways that are most meaningful for educators &mdash; e.g. the above 2018 examples in Spanish use:

- artículos
- medios
- videos
- subtítulos
- cuestionarios
- libros
- artículos ilustrados
- documentos
- simulaciones
- etc

(2B) Some of the above can be counted automatically thankfully &mdash; whereas others cannot, and must be counted far more manually &mdash; in any case, both kinds of counts are needed, having proved extremely useful to educators/librarians both during provisioning and in the field too!

holta, 2019-06-04:
As I sort this out with Ghana, Uganda and a growing nest others during IIAB 7.0's "beta" program this month, a few initial questions I'm trying to sort out for them on the way: (not that we can deliver WYSIWYG obviously, e.g. WordPress 5.2+ continues to struggle to delivering block-editor GUI functionality still today years later!)
- What's the best way to restore metadata to http://box/modules/en-wikem which used to appear as `1.4 GB, 14,098 articles, 17,340 images, May 2017` ?  (as can be seen at http://iiab.me/nigeria)
- What's the easiest way to hide excessive metadata for example if schools want to hide `Tags; [wikiquote;novid], Language: en, Date: 2019-05-09` currently displayed by http://box/kiwix/wikispecies_en_all_novid_2018-08/ ?
  - Likewise if other schools want to hide `Tags; [wikimedia;novid;_ftindex], Language: en` from http://box/kiwix/wikispecies_en_all_novid_2018-08/ what's the best way?  (i.e. to reduce & eliminate clutter, down to those metadata fields that are most meaningful in each case)
- More generally, what are the most efficient-yet-also-friendly channels/pathways for community feedback of upstream front-end UX improvements that we can develop starting with IIAB 7.0, for those implementers that want to "pay it forward" giving back editorial/front-end enhancements to others?  (starting with spelling typos in Content Pack descriptions, and logo fixes, etc!)

Refs: #1719 #1721

tim-moody, 2019-06-05:
Metadata is expected to be in the Footnote field, which can be edited. There are substitution variables for common zim properties. Probably the ones you don't like have been generated and are expected to be edited.  Metadata  is not expected to be in extra_html as before, but can be copied from there.

The generated menu defs contain all fields because it is easier to take out than to add, but I agree most will not want tags, (I argued against them unsuccessfully.)  I'm surprised you don't want the date as you always added it manually. 

Menu Item Definitions are data, not code and can be edited by implementers using the Admin Console. A PR can be made to update the repo.

tim-moody, 2019-06-05:
I'm not sure if the Footnote field is rendered as small, but it should be.

holta, 2019-06-16:
(1) "footnote" variables in [/library/www/html/js-menu/menu-files/menu-defs/*.json](https://github.com/iiab/iiab-admin-console/tree/master/roles/js-menu/files/menu-files/menu-defs) help to solve these issues (in increasingly automatic ways) as seen in:

https://github.com/iiab/iiab-admin-console/pull/210 Fix 5 popular menuDefs (Afristory + Wikibooks, Wikiquote, Wikispecies, Wiktionary) after "_novid" created problems (logos, descriptions, metadata)
https://github.com/iiab/iiab-admin-console/pull/212 Improved descriptions for 3 very common ZIM files

(2) Could `5672183 articles, 4783341 images` be shown much more readably as `5,672,183 articles, 4,783,341 images` in future?

(3) The "Desktop/Mobile" button at the top of IIAB pages like http://iiab.me/nigeria is an absolute lifesaver for teachers preparing their lessons (on Android typically) when they need to be able to flip this details on and off with a single large button.

Let's discuss how to restore this very important functionality for IIAB 7.1 if possible?

holta, 2019-06-18:
> I'm not sure if the Footnote field is rendered as small, but it should be.

The footnote appears in a large font, same as the the description.

@tim-moody can you check if a quick fix might possibly solve this for IIAB 7.0?

tim-moody, 2019-06-26:
> "Mobile/Desktop" button

what does this mean, whatever the implementer set for Mobile and Desktop?

holta, 2019-06-26:
Solved by @tim-moody's https://github.com/iiab/iiab-admin-console/pull/225

We'll never have perfect spacing (e.g. Chrome on Android still has a bit of extra vertical spacing below "Mustard Seed Books" but this is more than good enough!)

closed 2019-06-26 by holta