iiab/iiab #1725: http://iiab.me/maps (or similar!) public demo site needed for latest/evolving IIAB 7.0 Maps

holta, 2019-06-06:
@tim-moody @georgejhunt is http://iiab.me/maps the best place to make this fully public in coming days, with a short+memorable URL?  Or do you recommend another place?

In short: a very public demo site is needed, so that usability testing / acceptance testing (gathering feedback from different kinds of people in different countries) is accelerated both prior to IIAB 7.0's release, and also afterwards -

Refs:
#1704 Satellite tiles in z0-z9 package missing -- cause local blur
#1710 Living Docs for IIAB 7.0 Maps
#1711 Vector Map double click unreliable
#1720 New IIAB installs show "OpenStreetMap Test Page" which should explain what implementers can do
#1724 Map redirect

georgejhunt, 2019-06-06:
I think http://iiab.me/home is a better URL to promote -- because there are
multiple maps installed.

georgejhunt, 2019-06-06:
Do  you  have access to IIab.me? Apparently my credentials are not in .ssh/authorized_keys. I don't see anything on the VPN which announces such identity

holta, 2019-06-07:
> Do you have access to IIab.me? Apparently my credentials are not in .ssh/authorized_keys. I don't see anything on the VPN which announces such identity

1) Yes, but security is very tight on this machine (for good reason!) so ring me if you forget the trick.

2) http://iiab.me/maps should be made to work as well as http://iiab.me &mdash; e.g. for the the 90% of people who inevitably get lost AKA distracted as they click around http://iiab.me on their phones (-:

Others URLs to consider so that we're driving focused testing:
http://iiab.me/africamaps
http://iiab.me/testmaps
http://iiab.me/doubleclickmaps
etc...

holta, 2019-06-07:
Solved, many thanks to @tim-moody & @georgejhunt

closed 2019-06-07 by holta