iiab/iiab #1729: Use mysql native auth instead of unix socket auth

m-anish, 2019-06-12:
I'd like to implement mysql auth using the option #1 solution identified here, where mysql auth can happen remotely without invoking sudo. 

https://stackoverflow.com/questions/39281594/error-1698-28000-access-denied-for-user-rootlocalhost

This would allow installing apps over the web where mysql root credentials are required to create databases. 

This might be a security issue so need to be clear on it, AND that it wont break anything else. 

@holta @tim-moody @jvonau @georgejhunt

holta, 2019-06-12:
@tim-moody can you help here, if possible killing 2 birds (#1714) with 1 stone?

m-anish, 2019-06-13:
Okay, I am not in full favor of this as well. It might break something else. Let me think of alternate solutions.

closed 2019-06-13 by m-anish

holta, 2019-06-13:
Background: @m-anish was trying to make YouPHPTube work.