iiab/iiab #1737: Who can help test IIAB 7.0 on Debian 10 Buster?

holta, 2019-06-14:
Thanks to all who can help here, accelerating [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes)'s release in coming days/weeks!  Using http://d.iiab.io to install an IIAB 7.0 pre-release on the very latest build of Debian 10 Buster (generally several ISO's are published each day!) as can be found here:

https://cdimage.debian.org/cdimage/daily-builds/daily/arch-latest/amd64/iso-cd/debian-testing-amd64-netinst.iso

FYI "Debian 10's software package freeze takes place on Tuesday, June 25 at 13:00 UTC" and release date is expected to be Saturday July 6th.

WARNINGS/NOTES:

1) You might still need to set `VERSION_ID="10"` (or `VERSION_ID="10.0"`) in `/etc/os-release` as explained in #1387

2) If you're installing a BIG-sized IIAB (good idea during testing!) please remember to change your [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) to:

    sugarizer_install: False
    sugarizer_enabled: False

As explained in #1437

3) Consider setting http://d2.iiab.io within Line 23 of your [/opt/iiab/iiab/vars/default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L23) before running `sudo iiab` to test this IIAB's new download server, that should come online later in June.

Refs: #878 #1387 #1437

holta, 2019-06-14:
Please keep in touch with @floydianslips if you too can help here!

holta, 2019-06-15:
So far so good, http://d.iiab.io installed cleanly onto a 2019-06-15 pre-release snapshot of Debian 10 Buster, as documented at https://github.com/iiab/iiab/issues/1387#issuecomment-502345790

mitra42, 2019-06-21:
Adam - what are the instructions for installing (on Raspbian) over the top of an existing distro (v6.x). I'm guessing that just running the script will say its already installed. 
(I'll follow whatever instructions you give exactly to help test them)

holta, 2019-06-21:
> Adam - what are the instructions for installing (on Raspbian) over the top of an existing distro (v6.x).

Is someone trying to upgrade IIAB 6.6 or 6.7 to [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) for example?

1) For major upgrades, IIAB communities have not requested in-place upgrades generally, as bandwidth is completely lacking or way too pricey.  Safeguarding the most valuable local content and then rebuilding a school/clinic's IIAB every 2 years (or every couple semesters) seems to be the dominant model.  Whether the hardware and microSD are still good 2 years later or not...it depends.

   (An update every 2 years may seem off in the [FOMO](https://en.wikipedia.org/wiki/Fear_of_missing_out) hyperconnected ADHD West, but for perspective: these de facto every-year-or-two-updates are a *huge* step forward in impoverished communities where a 50-year-old mouldy encyclopedias were considered a luxury until recently...)

2) A few IIAB communities do however bring their IIAB microSD's "into the capital city for more bandwidth" a bit more often, e.g. for...
(a) content upgrades
(b) security upgrades
(c) "git pull" software upgrades as described in Item 4 here:

   https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack

Possibly I didn't answer your precise question?  But IIAB adoption patterns are quite clear after a decade &mdash; and of course growing in new directions at the same time, thanks to your work and others!

mitra42, 2019-06-21:
OK - so maybe its not a concern for the release then. 
Still a concern for me, as I have two boxes, one of which is on the Internet Archive's ethernet so I can't get to it physically and need to upgrade it (from IIAB 6.7 I believe) to IIAB 7.0 and current InternetArchive code.  I use it to point people at a working IIAB/RPI to prove it can be done, so would be good to be update to current :-)

holta, 2019-06-21:
Yes I agree that upgrades would be nice to have, in some future version of IIAB.

These are extremely hard to engineer however, as the Linux OS's are constantly changing underneath us, and dependency hell often means it's far more wise to install a clean/fresh system the next year.

For now you can consider the upgrade process mentioned earlier, in Step 4 here:
https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack

(At your own risk of course, as this is really only intended for upgrades a few weeks later, not a few months later...)

ASIDE: today many implementers buy another 128GB microSD card for $20 for backup just in case one microSD dies etc.  Allowing them to quickly/instantly recover if an upgrade or media/hardware goes south for any reason.

mitra42, 2019-06-22:
So its ...
```
sudo su -

cd /opt/iiab/iiab
git pull
./iiab-install --reinstall    # takes about 20+ min (or "./runrole osm-vector-maps")

cd /opt/iiab/iiab-admin-console
git pull
./install    
```
Will that pull get the v7, ok ?

holta, 2019-06-22:
> So its ...
> 
> ```
> sudo su -
> 
> cd /opt/iiab/iiab
> git pull
> ./iiab-install --reinstall    # takes about 20+ min (or "./runrole osm-vector-maps")
> 
> cd /opt/iiab/iiab-admin-console
> git pull
> ./install    
> ```
> 
> Will that pull get the v7, ok ?

Yes, that will get you a near-final IIAB 7.0

(Of course make sure you're on the master branch of both these repos, before doing "git pull")

mitra42, 2019-06-22:
Understood this is not a supported step , and that updating a remote installation might be rare. But here's a report back ....
First step `/opt/iiab/iiab` appeared to work - didn't give me a chance to edit /etc/iiab/local_vars.yml and I didn't find the `internetarchive_install` flags in it.  So I'm guessing it doesnt touch that file even to add new functions turned off - maybe it should ? 

Second step failed in `Create cmdsrv.conf file]`,  `FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'vector_map_path' is undefined"}`  - told me to rerun with `--limit @/opt/iiab/iiab-admin-console/iiab-admin.retry` but `./install --limit @/opt/iiab/iiab-admin-console/iiab-admin.retry` gave same result 

Stuck at this point ... and admin console is down at this point, (I can ssh into the box but have no physical access for next few weeks) anyone know how to get past this and complete the iiab-admin-console install ? 

Suggestions ... (maybe not good ones - highly likely I'm misunderstanding something) 
- [ ]  the lines in Ansible give the full path, it would make finding the files with problems easier ... 
- [ ] that the "Create cmdsrv.conf" file adds new lines to /opt/admin/cmdsrv/cmdsrv.conf if they are missing.

mitra42, 2019-06-23:
Trying manually to add "internetarchive" I cant do the "./runrole internetarchive" because that role isn't installed, and I can't figure out which variable to change to make it install it.   Adding to ' /etc/iiab/local_vars.yml ' and then a `./iiab-install --reinstall` didn't work (in stage 9 it didn't check for `internetarchive`). 

I'm more than happy to delete /opt/iiab and reinstall from scratch, but not sure that this wont make a bigger mess :-) so checking in before I try that.

holta, 2019-06-23:
@mitra42 it's very likely your `/etc/iiab/local_vars.yml` is many months out-of-date indeed, and you need to fix that as follows:
```
sudo su -

cd /opt/iiab/iiab
git pull
```

Then backup your local_vars.yml:
```
cp /etc/iiab/local_vars.yml /etc/iiab/local_vars.yml.old
```

Copy one of these 3 templates to create your new local_vars.yml:
```
#cp /opt/iiab/iiab/local_vars_min.yml /etc/iiab/local_vars.yml
cp /opt/iiab/iiab/local_vars_medium.yml /etc/iiab/local_vars.yml
#cp /opt/iiab/iiab/local_vars_big.yml /etc/iiab/local_vars.yml
```

Edit your new `/etc/iiab/local_vars.yml` as appropriate, setting both these variables:
```
internetarchive_install: True
internetarchive_enabled: True
```

Then complete the upgrade/install of both repos:
```
./iiab-install --reinstall

cd /opt/iiab/iiab-admin-console
git pull
./install
```

mitra42, 2019-06-24:
Note ... it should be 
```
cp /opt/iiab/iiab/vars/local_vars_medium.yml /etc/iiab/local_vars.yml
```

holta, 2019-06-24:
> Note ... it should be
> 
> ```
> cp /opt/iiab/iiab/vars/local_vars_medium.yml /etc/iiab/local_vars.yml
> ```

Fixed above, in case that helps others, Thx!

holta, 2019-06-27:
FYI here's a known issue we need to solve for Ubuntu 19.04, 19.10 later this year, and Ubuntu 20.04 arriving in just over 9 months:

#1714 MySQL database password-changing failure during fresh install of IIAB on Ubuntu 19.04 (& 18.04.2 ?)

holta, 2019-07-11:
Update: IIAB 7.0 pre-releases now run very well on Debian 10.0.0 Buster and the various flavors of Raspbian Buster running on RPi 3/4.

There is however at least 1 glitch (MySQL security issue!) when installing IIAB onto Ubuntu 19.04 and presumably also 19.10 daily/nightly pre-releases: #1714

holta, 2019-09-26:
@tim-moody tested IIAB 7.0's RC1 (or very similar) on Debian 10.1, thanks Tim:

#1951 "Test IIAB 7.0 on Debian 10.1 prior to release!"

holta, 2019-09-30:
As we release IIAB 7.0 later today, am moving this tkt to the 7.1 Milestone.

Thanks all for you continued assistance here, making "offline" learning infra shine for schools - clinics - all!

holta, 2019-10-15:
Please see PR #2002 and issue #2003 where Ubuntu 19.10 is rapidly coming together for this Thursday's OS release.

holta, 2019-12-27:
Progress here today: (thanks to #2082)

#1799 Who can test IIAB 7.x on Raspberry Pi Zero W?

holta, 2020-01-03:
FYI IIAB 7.1 pre-released (http://download.iiab.io) seem very solid on Ubuntu 19.10 (server of desktop version of the OS) which is extremely promising as we prepare to release IIAB 7.1 later this month hopefully!

ASIDE: testing IIAB's master branchers (http://download.iiab.io) on pre-release / daily builds of Ubuntu 20.04 LTS is a good idea, starting (around) late January 2020 if possible &mdash; using the links herein: https://github.com/iiab/iiab/wiki/IIAB-Platforms

holta, 2020-01-04:
ASIDE: @m-anish could not get Ubuntu 19.10 itself to start using its official Vagrant images at https://app.vagrantup.com/ubuntu/boxes/eoan64

Possible Reason: https://bugs.launchpad.net/cloud-images/+bug/1829625

holta, 2020-04-08:
Huge thanks to @georgejhunt, @jvonau, @tim-moody and @shanti-bhardwa who've done extremely productive Ubuntu 20.04 pre-release testing & development over the past 4 months now -- preparing us well for [IIAB 7.1's release](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) approaching quickly in coming weeks -- likely to occur sometime around April 30 / May 1.

jvonau, 2020-04-29:
Think this has been the most heavily tested pre-release we have ever had...

holta, 2020-09-11:
Related:

#2422 "Test IIAB 7.2 pre-releases on 64-bit Raspbian (renamed to Raspberry Pi OS / RaspiOS)"

holta, 2020-09-18:
A BIG-sized IIAB install (http://download.iiab.io) was smoke-tested on Linux Mint 20 overnight.

The install completed without a single hitch.

holta, 2020-09-20:
A BIG-sized IIAB install (http://download.iiab.io) of [IIAB 7.2](https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes) pre-release was successfully smoke-tested on the latest pre-release of [Ubuntu 20.10 Groovy Gorilla](http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/) overnight, and it worked great.  i.e. Node.js 12 needs to be installed manually using apt, super easy!

Details written up @ https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems (see section 2. for lesser-supported OS's).

I tested the "Hard-Float" version of the OS.

Is there somebody who can test their "64-bit ARM" version ?

fisherbryan, 2020-10-02:
Hi, I am installing IIAB option 3 BIG-sized, with the 1-line Installer on Ubuntu 20.04.01.
I had this issue with dnsmasq:

TASK [1-prep : dnsmasq (install now, configure LATER in 'network', after Stage 9)] ***
included: /opt/iiab/iiab/roles/network/tasks/dnsmasq.yml for 127.0.0.1

TASK [1-prep : Install dnsmasq] ************************************************
ok: [127.0.0.1]

TASK [1-prep : Install /etc/dnsmasq.d/dnsmasq-iiab, allowing systemd-resolved AND dnsmasq to work (#1306) and custom unit file] ***
ok: [127.0.0.1] => (item={'src': 'roles/network/templates/network/dnsmasq.service.u18', 'dest': '/etc/systemd/system/iiab-dnsmasq.service', 'mode': '0644'})
ok: [127.0.0.1] => (item={'src': 'roles/network/templates/network/dnsmasq-iiab', 'dest': '/etc/dnsmasq.d/dnsmasq-iiab', 'mode': '644'})

TASK [1-prep : Don't use stock dnsmasq systemd unit file during boot but start now] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=36   changed=8    unreachable=0    failed=1    skipped=8    rescued=0    ignored=0   

sys-admin@sib:~$ journalctl -xe
Oct 02 08:07:31 sib.content systemd[1]: dnsmasq.service: Control process exited>
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- An ExecStart= process belonging to unit dnsmasq.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 3.
Oct 02 08:07:31 sib.content systemd[1]: dnsmasq.service: Failed with result 'ex>
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit dnsmasq.service has entered the 'failed' state with result 'exit-co>
Oct 02 08:07:31 sib.content systemd[1]: Failed to start dnsmasq - A lightweight>
-- Subject: A start job for unit dnsmasq.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit dnsmasq.service has finished with a failure.
-- 
-- The job identifier is 1011 and the job result is failed.
Oct 02 08:07:31 sib.content sudo[1410]: pam_unix(sudo:session): session closed >

Solution for me was:
sudo rm /etc/dnsmasq.d/lxd

Then continued with 'sudo iiab' - still installing while i'm sending this message.

I'm would like to assist with testing further if there is anything specific you would like to see or know about? I'm setting up this server to use it for a Moodle instance and Monitoring devices with AWSTATS. 

It's my first time participating to this type of platform, But I'd love to learn more about how to contribute to the best of my ability :)

fisherbryan, 2020-10-02:
Install Complete:

[image]

holta, 2020-10-02:
Profound Thanks @fisherbryan

If possible can you run 'sudo iiab-diagnostics' and reply with the resulting URL, to help all understand?

(That's a sprunge.us URL, whose contents you can examine prior to the pastebin being generated!)

holta, 2020-10-02:
@fisherbryan

- What model hardware, Raspberry Pi or Virtual Machine are you using?
- Which Ubuntu 20.04.1 installer file/URL did you use?

Thanks!  FYI we have a live chat channel I can join you to if that helps you out &mdash; don't hesitate to write me if so: holt @ unleashkids [dot] org

fisherbryan, 2020-10-02:
@holta please see:
http://sprunge.us/g07tNW

fisherbryan, 2020-10-02:
@holta I'm using an;
Intel Nuc i3: MODEL: NUC8BEH.
Installer: curl d.iiab.io/install.txt | sudo bash

My final install in future will be on an MSI NUC, model: 3168NGW (FYI)

jvonau, 2020-10-02:
> Solution for me was:
sudo rm /etc/dnsmasq.d/lxd
Then continued with 'sudo iiab' - still installing while i'm sending this message.
I'm would like to assist with testing further if there is anything specific you would like to see or know about? I'm setting up this server to use it for a Moodle instance and Monitoring devices with AWSTATS.
It's my first time participating to this type of platform, But I'd love to learn more about how to contribute to the best of my ability :)

When you installed Ubuntu it appears that container support was added to your install at some point, did you reboot to initialize the containers' network layout prior to starting the '1-line Installer'?

holta, 2020-10-02:
@fisherbryan how exactly did you install Ubuntu 20.04.1?

(If possible also mention the OS/image URL you used, Thanks!)

fisherbryan, 2020-10-02:
Hi @holta,

I downloaded and Ubuntu 18 ISO from https://ubuntu.com/download/server
Option 3. Used a USB Drive to install. Choose all default installation, I
did not select ANY services on the services page. I had no idea about LXD
containers that neede to be initialized. After the OS installed I did an
apt ... update/upgrade and installed IIAB.

jvonau, 2020-10-02:
Ok that makes sense, /etc/dnsmasq.d/lxd is a remnant of the 18.04 -> 20.04 upgrade process that didn't do a good job cleaning up.
Refs:
https://www.reddit.com/r/pihole/comments/g86xo8/after_ubuntu_2004_upgrade_unable_to_start_dns/
https://discourse.pi-hole.net/t/pi-hole-breaks-after-upgrade-from-ubuntu-18-04-to-20-04-fixed/32356

fisherbryan, 2020-10-02:
Thanks so much for the refs. much appreciated!!

fisherbryan, 2020-10-02:
@jvonau <https://github.com/jvonau>
When I connect to *ip/awstats* it works fine. When I go to ip/moodle the
address changes to *hostname/moodle* (screenshot attached) and ip/kolibri I
get 502 Bad Gateway. Do I have to make any configuration changes? Can you
please guide me to a solution or necessary steps to take?
[image: Screenshot from 2020-10-02 23-01-03.png]
[image: Screenshot from 2020-10-02 23-00-44.png]
[image: Screenshot from 2020-10-02 23-00-38.png]
[image: Screenshot from 2020-10-02 23-07-28.png]

holta, 2020-10-02:
> When I go to ip/moodle the address changes to *hostname/moodle*

Please read http://FAQ.IIAB.IO > "36 WordPress & Moodle Administration: What tips & tricks exist?" > "How do I set the Moodle URL..."

> ip/kolibri I get 502 Bad Gateway

You're the 2nd person to experience a 502 with http://box/kolibri

We don't yet understand this!

They were apparently able to get Kolibri 0.14.3 working by reinstalling it as follow:

```
cd /opt/iiab/iiab
sudo ./runrole --reinstall kolibri
```

Does that work for you?

Please also reboot, and retry different things in Kolibri to confirm (we all need to understand this issue better, Thanks!)

fisherbryan, 2020-10-02:
@holta Thanks so much!
to get Kolibri 0.14.3 working by reinstalling it as follow:

cd /opt/iiab/iiab
sudo ./runrole --reinstall kolibri

Worked 100%

[image]

modifying /opt/iiab/moodle/config.php --> wwwroot = 'http://192.168.0.100/moodle' 
Worked 100%
[image]

Thanks very much for your efficient response, much appreciate it!

fisherbryan, 2020-10-02:
In case someone reporst this: not sure of its a bug or something missing in the script:
sys-admin@sib:/opt/iiab/iiab$ sudo ./runrole www_option && reboot
        "ansible_python": {
            "executable": "/usr/bin/python3",
            "type": "cpython",
        "ansible_python_version": "3.8.2",
        "ansible_selinux_python_present": false,
ERROR! the role 'www_option' was not found in /opt/iiab/iiab/roles:/root/.ansible/roles:/usr/share/ansible/roles:/etc/ansible/roles:/opt/iiab/iiab

The error appears to be in '/opt/iiab/iiab/run-one-role.yml': line 13, column 7, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

    - { role: 0-init, tags: ['0-init'] }
    - { role: "{{ role_to_run }}", tags: ['run'] }
      ^ here
We could be wrong, but this one looks like it might be an issue with
missing quotes. Always quote template expression brackets when they
start a value. For instance:

    with_items:
      - {{ foo }}

Should be written as:

    with_items:
      - "{{ foo }}"

holta, 2020-10-02:
> sudo ./runrole www_option

@fisherbryan don't you want an 's' on the end, as follows?

```
sudo ./runrole www_options

jvonau, 2020-10-02:
> sudo ./runrole www_option && reboot

>pi@box:/opt/iiab/iiab $ sudo ./runrole
Usage: ./runrole <name of role>
Usage: ./runrole --reinstall <name of role>

>Optional 2nd parameter is full PATH/FILENAME for logging.
If omitted, <current directory>/iiab-debug.log is used.

The '&&' might be interpreted as the log file name, suppose there could be a check for excessive variables on the command line for those who make assumptions on the usage.

jvonau, 2020-10-02:
> > When I go to ip/moodle the address changes to _hostname/moodle_
> > ip/kolibri I get 502 Bad Gateway
> 
> You're the 2nd person to experience a 502 with http://box/kolibri
> 

Where is the first reported issue recorded?

> They were apparently able to get Kolibri 0.14.3 working by reinstalling it as follow:
> 

There is more to the story that is being told, https://learningequality.org/download/ lists 14.3 as being released on Aug 21, 2020, IIAB would install 0.14.3 since then while 8fc732e1e4f24c012c4a8a73cefb8fe93f3fec90 pushed on Sept 2, 2020 dropped support for Ubuntu 18.04. Having to use `runrole --reinstall kolibri` points to a exisiting install on the filesystem. Putting my crystal rear view mirror to use think the original install was preformed on 18.04 prior to Aug 21, 2020 then was upgraded 20.04 to meet the current reduced supported OS criteria. If there was no tinkering with the installed files, /etc/iiab/iiab.ini would show the original install date within the [version] and the last ansible run in [runtime].

holta, 2020-10-02:
> > > ip/kolibri I get 502 Bad Gateway
> > 
> > You're the 2nd person to experience a 502 with http://box/kolibri
> 
> Where is the first reported issue recorded?.

FYI it definitely did not take place on Ubuntu 18.04

It took place on "RaspiOS with desktop"

Sorry there aren't more details.

fisherbryan, 2020-10-03:
> If omitted, /iiab-debug.log is used.

Thanks so much guys! Truly appreciate it!!

jvonau, 2020-10-03:
We are not chasing #2045 again, slow to update 'latest' link are we?

fisherbryan, 2020-10-03:
Does IIAB 7.2 support kolibri-server yet?

jvonau, 2020-10-03:
Not really, with all the other services to integrate under nginx we supply a suitable nginx config file for use with kolibri.

fisherbryan, 2020-10-03:
Understood👌 we use kolibri as our primary LMS on all our servers, with
kolibri-server and postgresql db. The one I'm building now is a "special"
one with IIAB😁 check out www.schoolinabox.co.za when you get a chance.

Thanks so much!

jvonau, 2020-10-03:
@fisherbryan Looks slick, other things I noticed from iiab-diagnosics:
>=IIAB==========================================================================
lrwxrwxrwx 1 root root 39 Aug  6 22:35 /etc/resolv.conf -> ../run/systemd/resolve/stub-resolv.conf

Sort of confirms my theory on the install date above.
>hostapd[727]: rfkill: WLAN hard blocked

The built in local hostspot is unavailable, is your WiFi hardware unavailable due to the 'button' for WiFi on the laptop not being enabled?  That is the most likely cause of
> Oct  2 10:19:27 sib vmunix: [   37.829323] ------------[ cut here ]------------
Oct  2 10:19:27 sib vmunix: [   37.829353] WARNING: CPU: 2 PID: 732 at net/wireless/nl80211.c:7029 nl80211_get_reg_do+0x1fc/0x230 [cfg80211]
Oct  2 10:19:27 sib vmunix: [   37.829353] Modules linked in: bridge stp llc ip6table_filter ip6_tables iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bpfilter dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua snd_sof_pci snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda snd_sof_intel_byt snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core snd_compress iwlmvm snd_hda_codec_hdmi ac97_bus snd_hda_codec_realtek btusb snd_hda_codec_generic snd_pcm_dmaengine mac80211 btrtl btbcm btintel ledtrig_audio bluetooth libarc4 intel_rapl_msr intel_rapl_common snd_hda_intel x86_pkg_temp_thermal snd_intel_dspcfg ecdh_generic ecc snd_hda_codec mei_hdcp intel_powerclamp coretemp kvm_intel kvm snd_hda_core rapl intel_cstate mei_me snd_hwdep mei snd_pcm iwlwifi snd_timer snd rtsx_pci_ms wmi_bmof intel_wmi_thunderbolt memstick soundcore cfg80211 intel_pch_thermal mac_hid acpi_tad acpi_pad sch_fq_codel parport_pc ppdev lp parport ip_tables
Oct  2 10:19:27 sib vmunix: [   37.829375]  x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear i915 i2c_algo_bit crct10dif_pclmul crc32_pclmul drm_kms_helper syscopyarea ghash_clmulni_intel sysfillrect sysimgblt rtsx_pci_sdmmc aesni_intel fb_sys_fops crypto_simd ahci cryptd glue_helper e1000e i2c_i801 rtsx_pci drm libahci wmi pinctrl_cannonlake video pinctrl_intel
Oct  2 10:19:27 sib vmunix: [   37.829389] CPU: 2 PID: 732 Comm: hostapd Not tainted 5.4.0-48-generic #52-Ubuntu
Oct  2 10:19:27 sib vmunix: [   37.829389] Hardware name: Intel(R) Client Systems NUC8i3BEH/NUC8BEB, BIOS BECFL357.86A.0064.2019.0213.1122 02/13/2019
Oct  2 10:19:27 sib vmunix: [   37.829402] RIP: 0010:nl80211_get_reg_do+0x1fc/0x230 [cfg80211]
Oct  2 10:19:27 sib vmunix: [   37.829403] Code: 45 cc 01 00 00 00 e8 b3 0c ce e2 85 c0 0f 84 fd fe ff ff eb a8 4c 89 e7 48 89 45 c0 e8 5d 8f 0a e3 48 8b 45 c0 e9 40 ff ff ff <0f> 0b 4c 89 e7 e8 4a 8f 0a e3 b8 ea ff ff ff e9 2c ff ff ff e9 7a
Oct  2 10:19:27 sib vmunix: [   37.829404] RSP: 0018:ffffb6e000b97a18 EFLAGS: 00010202
Oct  2 10:19:27 sib vmunix: [   37.829405] RAX: 0000000000000000 RBX: 0000000000000001 RCX: 0000000000000000
Oct  2 10:19:27 sib vmunix: [   37.829405] RDX: ffff99316a5e8008 RSI: 0000000000000000 RDI: ffff99316a5e8300
Oct  2 10:19:27 sib vmunix: [   37.829406] RBP: ffffb6e000b97a58 R08: ffff99316a5e8300 R09: ffff99317838a014
Oct  2 10:19:27 sib vmunix: [   37.829406] R10: 0000000000000000 R11: 000000000000001e R12: ffff993168ff6300
Oct  2 10:19:27 sib vmunix: [   37.829406] R13: ffffb6e000b97ab0 R14: ffff99317838a014 R15: 0000000000000000
Oct  2 10:19:27 sib vmunix: [   37.829407] FS:  00007f7be90b1180(0000) GS:ffff993179b00000(0000) knlGS:0000000000000000
Oct  2 10:19:27 sib vmunix: [   37.829408] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Oct  2 10:19:27 sib vmunix: [   37.829408] CR2: 000055b191789308 CR3: 0000000171046002 CR4: 00000000003606e0
Oct  2 10:19:27 sib vmunix: [   37.829409] Call Trace:
Oct  2 10:19:27 sib vmunix: [   37.829413]  ? rtnl_lock+0x15/0x20
Oct  2 10:19:27 sib vmunix: [   37.829415]  genl_family_rcv_msg+0x1b9/0x470
Oct  2 10:19:27 sib vmunix: [   37.829417]  genl_rcv_msg+0x4c/0xa0
Oct  2 10:19:27 sib vmunix: [   37.829419]  ? _cond_resched+0x19/0x30
Oct  2 10:19:27 sib vmunix: [   37.829420]  ? genl_family_rcv_msg+0x470/0x470
Oct  2 10:19:27 sib vmunix: [   37.829420]  netlink_rcv_skb+0x50/0x120
Oct  2 10:19:27 sib vmunix: [   37.829422]  genl_rcv+0x29/0x40
Oct  2 10:19:27 sib vmunix: [   37.829422]  netlink_unicast+0x187/0x220
Oct  2 10:19:27 sib vmunix: [   37.829423]  netlink_sendmsg+0x222/0x3e0
Oct  2 10:19:27 sib vmunix: [   37.829425]  sock_sendmsg+0x65/0x70
Oct  2 10:19:27 sib vmunix: [   37.829426]  ____sys_sendmsg+0x212/0x280
Oct  2 10:19:27 sib vmunix: [   37.829427]  ___sys_sendmsg+0x88/0xd0
Oct  2 10:19:27 sib vmunix: [   37.829429]  ? ___sys_recvmsg+0x88/0xc0
Oct  2 10:19:27 sib vmunix: [   37.829431]  ? __cgroup_bpf_run_filter_setsockopt+0xae/0x2d0
Oct  2 10:19:27 sib vmunix: [   37.829432]  ? _cond_resched+0x19/0x30
Oct  2 10:19:27 sib vmunix: [   37.829434]  ? aa_sk_perm+0x43/0x170
Oct  2 10:19:27 sib vmunix: [   37.829435]  __sys_sendmsg+0x5c/0xa0
Oct  2 10:19:27 sib vmunix: [   37.829437]  __x64_sys_sendmsg+0x1f/0x30
Oct  2 10:19:27 sib vmunix: [   37.829438]  do_syscall_64+0x57/0x190
Oct  2 10:19:27 sib vmunix: [   37.829440]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Oct  2 10:19:27 sib vmunix: [   37.829441] RIP: 0033:0x7f7be91f8747
[... cut ...]
Oct  2 10:19:27 sib vmunix: [   37.829443] RDX: 0000000000000000 RSI: 00007ffc57b46960 RDI: 0000000000000004
Oct  2 10:19:27 sib vmunix: [   37.829444] RBP: 000055b19177c460 R08: 0000000000000004 R09: 00007f7be92c0b80
Oct  2 10:19:27 sib vmunix: [   37.829444] R10: 00007ffc57b46a34 R11: 0000000000000246 R12: 000055b19177ac60
Oct  2 10:19:27 sib vmunix: [   37.829445] R13: 00007ffc57b46960 R14: 00007ffc57b46a34 R15: 000055b19177ef70
Oct  2 10:19:27 sib vmunix: [   37.829446] ---[ end trace 992a2a4b712decc0 ]---

Email me directly should you need hardware related guidance.

fisherbryan, 2020-10-03:
@jvonau great! I'm using a Mikrotik router to handle gateway and dhcp requests, ultimately this will be an offline system but updated with a router that has internet access plugged into the Mikrotik router, monthly or quarterly. Upon final completion on this server I'll change it to a static IP. The primary use of the server will be used to access moodle content and to monitor specific devices (android tablets), so that we can determine which tablets are not being used in the interim without the user telling us or informing us about an issue with the device (using AWSTATS). I'll create an ACL list in the Mikrotik router for the specific devices that will have access to the network. Our tablets have are locked down with a launcher that only allow apps, web apps etc we set available. I would appreciate any guidance on making this work and I'm completely new to this setup, it will be my first attempt to do this.  Please send me an email to bryan1fisher@gmail.com. thanks so much!

holta, 2020-10-03:
@fisherbryan I have a new theory that Kolibri fails when 0.14.3 is installed on top of 0.14.2 -- and possibly other situations :/

Can you confirm that you get `502 Bad Gateway` when visiting http://box/kolibri after reboot?  Thanks for checking!

jvonau, 2020-10-03:
502's are either dns related or the backend served by nginx failing to start before the timeout, if you see a 502 error please capture the output of `sudo systemctl status kolibri` and `sudo journalctl -b 0 -u kolibri` 

I suspect something like #1932 related, kolibri starts before the database or slow to start with channels present.

jvonau, 2020-10-03:
The question for @fisherbryan becomes did you install|enable any channels for use with Kolibri?

holta, 2020-10-03:
> The question for @fisherbryan becomes did you install|enable any channels for use with Kolibri?

@fisherbryan please also see patch #2556 which has now been merged.

We'd greatly appreciate your feedback, after you upgrade Kolibri as follows:

```
cd /opt/iiab/iiab
sudo git pull    # if you want to try the latest IIAB 7.2 pre-release
sudo ./runrole --reinstall kolibri    # (re)install Kolibri 0.14.3
```

(Hopefully that permanently gets rid of most of your "502 Bad Gateway" errors...)

jvonau, 2020-10-03:
Is that safe given kolibri_provision: True if the channels are already loaded?

holta, 2020-10-03:
> Is that safe given kolibri_provision: True if the channels are already loaded?

It worked for me.

Aside: that `kolibri_provision` flag should be likely refined or removed in future, per IIAB norms and Kolibri recommendations.

fisherbryan, 2020-10-05:
Hi @Jerry,

No, I didn't install any channels yet. Only that screen shot of Kolibri
login screen is what I saw, assumed it was working correctly and left it
there. However after the server reboots it then as you know stops working
without intervention.

fisherbryan, 2020-10-05:
I am currently installing a fresh install of IIAB with the 1 liner
script: curl d.iiab.io/install.txt | sudo bash on an MSI, Model:3168 NGW. A
clean installation without any changes to the .yaml file. SO far there were
no dns errors like before (with lxd container nagging).

I'll send you the resulting URL of the iiab diagnostic tool when it
has completed the installation.

[14 more replies]

closed 2021-07-27 by holta