iiab/iiab #1739: Could Map Packs auto-display (on http://box) a footnote with (1) size-in-GB (2) levels-of-map-zoom (3) levels-of-satellite-zoom (4) number-of-searchable-cities (5) date-of-map-pack publication?

holta, 2019-06-14:
This is a question for @georgejhunt and @tim-moody as to whether this is possible of not, such that every downloadable Map Pack would display these things on the IIAB home pages automatically:

1) size-in-GB
2) levels-of-map-zoom
3) levels-of-satellite-zoom
4) number-of-searchable-cities
5) date-of-map-pack publication

(Of course if it's not possible for IIAB 7.0, the question applies to 7.1 or 7.2!)

holta, 2019-07-06:
Should this be implemented in https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/templates/iiab-update-map presumably?

(Or is there a better place?)

holta, 2019-07-19:
> Should this be implemented in https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/templates/iiab-update-map presumably?
> 
> (Or is there a better place?)

Is this already partially implemented at https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/templates/iiab-update-map#L195-L206 ?

What's missing or needed if so?

georgejhunt, 2019-07-19:
It's a nice project to familiarize someone with admin console code. It
involves pulling the data of interest from regions.json, and populating
/library/www/html/common/assets/vector-idx.json with the information
pertinent to that particular region, and that particular release. The
information is not currently where it needs to be, to use the strategy we
applied to zims and oer2go.

In my opinion, this objective does not add enough value to risk the
disruption this late in the release cycle. And at this point, I'm very
discouraged in my efforts to fix shortcomings in the map part of the admin
console.

holta, 2019-07-19:
Many Thanks @georgejhunt for the explanation.

Someone can work with on this for the IIAB 7.1 release cycle.

tim-moody, 2019-07-19:
My understanding after the weekly call a couple of calls ago was that osm vector maps would have a major revision that would integrate all maps installed into a single access rather than one per region. If this is the case, then fussing with the admin interface should wait until that is done.

tim-moody, 2019-07-19:
Also, as I thought more about regions.json, I realized that it is not entirely analogous to the kiwix and oer2go catalogs. Those are a list of content items with a unique key for each. The key to regions.json, region, is more an attribute and the key, some portion of the unique file name, is an attribute rather than the key. The purpose of the 'idx' file is to relate generic name to current id, but regions.json is already organized like an idx.