iiab/iiab #1741: Admin links not working

mikerjensen, 2019-06-18:
on going to box.local/admin i get: (same problem after a reboot)

GET-WHLIST FAILED and reported IIAB-CMDSRV failed to initialize
clicking on the OK button, gives:
GET-VARS FAILED and reported IIAB-CMDSRV failed to initialize
clicking on the OK button, gives:
GET-ZIM-STAT FAILED and reported IIAB-CMDSRV failed to initialize
then:
GET-OER2GO-STAT FAILED and reported IIAB-CMDSRV failed to initialize
then:
GET-SPACE-AVAIL FAILED and reported IIAB-CMDSRV failed to initialize.. 
the page background has:
Starting Application
6/18/2019, 12:37:54 PM: initConfigVars found empty data
6/18/2019, 12:37:54 PM: procOneCatalog found empty data
6/18/2019, 12:37:54 PM: procOneCatalog found empty data
6/18/2019, 12:34:13 PM: GET-CONF FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-ANS FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-IIAB-INI FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-EXTDEV-INFO FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-SPACE-AVAIL FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-OER2GO-STAT FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-ZIM-STAT FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: Successfully connected to Server
6/18/2019, 12:34:12 PM: GET-VARS FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: GET-WHLIST FAILED and returned IIAB-CMDSRV failed to initialize
6/18/2019, 12:34:12 PM: Command GET-EXTDEV-INFO sent to server
6/18/2019, 12:34:12 PM: Command GET-SPACE-AVAIL sent to server
6/18/2019, 12:34:11 PM: Command GET-OER2GO-STAT sent to server
6/18/2019, 12:34:11 PM: Command GET-ZIM-STAT sent to server
6/18/2019, 12:34:11 PM: Command GET-IIAB-INI sent to server
6/18/2019, 12:34:11 PM: Command GET-CONF sent to server
6/18/2019, 12:34:11 PM: Command GET-ANS sent to server
6/18/2019, 12:34:11 PM: Command GET-VARS sent to server
6/18/2019, 12:34:11 PM: Command GET-WHLIST sent to server
6/18/2019, 12:34:11 PM: Checking Server Connection
6/18/2019, 12:34:11 PM: Starting init

holta, 2019-06-18:
Thanks @mikerjensen:

Can you give a bit more context so @tim-moody can dig into this more?

Please post the URL's resulting from these commands:
```
pastebinit < /etc/iiab/iiab.env
pastebinit < /etc/iiab/iiab.ini
pastebinit < /etc/iiab/local_vars.yml
```
PS have you verified that your disk (microSD) is working well (and not full!) using `df -h` or similar?

tim-moody, 2019-06-18:
When the cmdsrv fails to initialize, the app can't start, and it's links are not activated.

So the question is why does iiab-cmdsrv fail to initialize. Probably there is something unexpected in storage devices or networking devices.

@mikerjensen the easiest way to troubleshoot is to do the following after sshing into iiab:

sudo su -
systemctl stop iiab-cmdsrv
cd /opt/admin/cmdsrv
./iiab-cmdsrv

now go into the admin console and then go back to the ssh session. there should be an error on the screen.

alternatively, if you give me access I can do this.

mikerjensen, 2019-06-18:
Thanks Tim, I tried that and immediately upon starting
      iiab-cmdsrv (before going to the admin console) i got the below
      text. Going to the admin console doesn't result in any further
      output in the ssh session...
    Mike
    
    root@box:/opt/admin/cmdsrv# ./iiab-cmdsrv
      Traceback (most recent call last):
        File "./iiab-cmdsrv", line 2770, in <module>
          main()
        File "./iiab-cmdsrv", line 169, in main
          init()
        File "./iiab-cmdsrv", line 2240, in init
          read_iiab_vars()
        File "./iiab-cmdsrv", line 2337, in read_iiab_vars
          local_vars = yaml.load(stream)
        File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line
      71, in load
          return loader.get_single_data()
        File "/usr/lib/python2.7/dist-packages/yaml/constructor.py",
      line 37, in get_single_data
          node = self.get_single_node()
        File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line
      36, in get_single_node
          document = self.compose_document()
        File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line
      55, in compose_document
          node = self.compose_node(None, None)
        File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line
      84, in compose_node
          node = self.compose_mapping_node(anchor)
        File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line
      127, in compose_mapping_node
          while not self.check_event(MappingEndEvent):
        File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98,
      in check_event
          self.current_event = self.state()
        File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line
      428, in parse_block_mapping_key
          if self.check_token(KeyToken):
        File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line
      116, in check_token
          self.fetch_more_tokens()
        File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line
      156, in fetch_more_tokens
          self.scan_to_next_token()
        File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line
      777, in scan_to_next_token
          self.forward()
        File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line
      103, in forward
          self.update(length+1)
        File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line
      170, in update
          exc.encoding, exc.reason)
      yaml.reader.ReaderError: 'utf8' codec can't decode byte #xc2:
      invalid continuation byte
        in "/etc/iiab/local_vars.yml", position 2220
      root@box:/opt/admin/cmdsrv# 
      
    
    On 18/06/19 14:56, Tim Moody wrote:
    
    
      
      When the cmdsrv fails to initialize, the app can't start, and
        it's links are not activated.
      So the question is why does iiab-cmdsrv fail to initialize.
        Probably there is something unexpected in storage devices or
        networking devices.
      @mikerjensen
        the easiest way to troubleshoot is to do the following after
        sshing into iiab:
      sudo su -
        systemctl stop iiab-cmdsrv
        cd /opt/admin/cmdsrv
        ./iiab-cmdsrv
      now go into the admin console and then go back to the ssh
        session. there should be an error on the screen.
      alternatively, if you give me access I can do this.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56UEWOXXDXDNLN2Q4GTP3DSSHA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6WTJI#issuecomment-503146917",
"url": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56UEWOXXDXDNLN2Q4GTP3DSSHA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6WTJI#issuecomment-503146917",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

mikerjensen, 2019-06-18:
Thanks, 
    
    urls below....when i try to do the last one it results in the
      below error message..
    disk is not full....not sure how i check its integrity without
      taking the SD card out?
    Filesystem      Size  Used Avail Use% Mounted on
      /dev/root        30G   11G   18G  38% /
      devtmpfs        459M     0  459M   0% /dev
      tmpfs           464M  8.0K  464M   1% /dev/shm
      tmpfs           464M   13M  451M   3% /run
      tmpfs           5.0M  4.0K  5.0M   1% /run/lock
      tmpfs           464M     0  464M   0% /sys/fs/cgroup
      /dev/mmcblk0p1   43M   23M   21M  53% /boot
      tmpfs            93M  8.0K   93M   1% /run/user/1000
      /dev/sda1       1.8T  1.7T   43G  98% /media/usb0
    
    
    http://paste.debian.net/1088320/
    http://paste.debian.net/1088321/
      
    
     pastebinit < /etc/iiab/local_vars.yml
      Traceback (most recent call last):
        File "/usr/bin/pastebinit", line 424, in <module>
          params = urlencode(params)
        File "/usr/lib/python3.5/urllib/parse.py", line 801, in
      urlencode
          v = quote_via(str(v), safe, encoding, errors)
        File "/usr/lib/python3.5/urllib/parse.py", line 729, in
      quote_plus
          string = quote(string, safe + space, encoding, errors)
        File "/usr/lib/python3.5/urllib/parse.py", line 707, in quote
          string = string.encode(encoding, errors)
      UnicodeEncodeError: 'utf-8' codec can't encode character '\udcc2'
      in position 2220: surrogates not allowed
      
    
    On 18/06/19 13:30, A Holt wrote:
    
    
      
      Thanks @mikerjensen:
      Can you give a bit more context so @tim-moody
        can dig into this more?
      Please post the URL's resulting from these commands:
      pastebinit < /etc/iiab/iiab.env
pastebinit < /etc/iiab/iiab.ini
pastebinit < /etc/iiab/local_vars.yml

      PS have you verified that your disk (microSD) is working well
        (and not full!) using df -h or similar?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56RMGIC5XUAKLFOLOWDP3DIP3A5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6G66I#issuecomment-503082873",
"url": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56RMGIC5XUAKLFOLOWDP3DIP3A5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6G66I#issuecomment-503082873",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

tim-moody, 2019-06-18:
looks like /etc/iiab/local_vars.yml has a unicode string that can't be decoded either by pastebinit or iiab-cmdsrv

tim-moody, 2019-06-18:
the two pastebin links are iiab.env and iiab.ini. can you email me /etc/iiab/local_vars.yml

tim-moody, 2019-06-18:
what if you cat /etc/iiab/local_vars.yml

tim-moody, 2019-06-18:
or another question would be what changes have you made to local vars?

tim-moody, 2019-06-18:
https://www.charbase.com/dcc2-unicode-invalid-character seems to be the offending character.

mikerjensen, 2019-06-18:
No problem catting the file.... which is apparently: 
    
    file $_
      /etc/iiab/local_vars.yml: ISO-8859 text
    attached..
    i don't think i've ever touched this file manually...
    
    
    
    On 18/06/19 15:21, Tim Moody wrote:
    
    
      
      the two pastebin links are iiab.env and iiab.ini. can you email
        me /etc/iiab/local_vars.yml
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56WDH4MX5BDJ3RVRCQ3P3DVOLA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6ZHBY#issuecomment-503157639",
"url": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56WDH4MX5BDJ3RVRCQ3P3DVOLA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX6ZHBY#issuecomment-503157639",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

tim-moody, 2019-06-18:
I don't see the attachment

tim-moody, 2019-06-18:
how did you install iiab, script from download.iiab.io? If you have not customized it we could just copy over one of the sample local vars.

mikerjensen, 2019-06-18:
Strange, i just checked my sent folder - the file is there
      attached, but here it is again, and also pasted into the message
      body below.
    I installed IIAB according to the instructions here:
    https://github.com/internetarchive/dweb-mirror/blob/master/INSTALLATION-iiab-raspberry.md
    which reminds me i did edit the local_vars.yml file to enable
      internet archive..
    MIke
    
    
    
     
    # This is local_vars_min.yml -- copy it to /etc/iiab/local_vars.yml then...
# modify variables below, to override /opt/iiab/iiab/vars/default_vars.yml
wan_ip: 192.168.1.218
wan_netmask: 255.255.255.0
wan_gateway: 192.168.1.1
wan_nameserver: 8.8.8.8

# PLZ READ http://wiki.laptop.org/go/IIAB/local_vars.yml AND http://FAQ.IIAB.IO
# Orig Idea: branch github.com/xsce/xsce-local for your deployment/community

# Ansible's default timeout for "get_url:" downloads (10 seconds) often fails
download_timeout: 200

# Languages (for Apache)
default_language: en
language_priority: en es fr

# Set iiab_admin_user_install: False if you don't want iiab_admin_user & wheel
# group auto-created in roles/iiab-admin/tasks/main.yml (hence disabling sudo-
# checks/warnings of published passwds like pi/raspberry & iiab-admin/g0adm1n).
iiab_admin_user_install: True
# If iiab_admin_user_install: False, set iiab_admin_user (below) to an existing
# Linux user that has sudo access, for login to Admin Console http://box/admin
iiab_admin_user: iiab-admin
# Password hash to be used if Ansible creates the above user:
iiab_admin_pwd_hash: $6$xsce51$D.IrrEeLBYIuJkGDmi27pZUGOwPFp98qpl3hxMwWV4hXigFGmdSvy3s/j7tn6OnyTTLmlV7SsN0lCUAFzxSop.
# Obtain a password hash - NEW MORE SECURE WAY:
#    python3 -c 'import crypt; print(crypt.crypt("<plaintext>", crypt.mksalt(crypt.METHOD_SHA512)))'
# Obtain a password hash - OLD WAY:
#    python -c 'import crypt; print crypt.crypt("<plaintext>", "$6$<salt>")'

iiab_hostname: box
iiab_domain: lan

# Homepage: set to /home or /wordpress or /mediawiki or /wiki (for DokuWiki)
iiab_home_url: /home

# Raspbian requires Wi-Fi country since March 2018.  Please set it here:
host_country_code: US
host_ssid: "Internet in a Box"
host_wifi_mode: g
host_channel: 6
hostapd_secure: False
hostapd_password: changeme

# Enable "campus access" to ~10 common IIAB services like Kiwix (3000), KA Lite
# (8008) and Calibre (8010 or 8080) etc, on the WAN side of your IIAB server.
# Only 1 of the 6 lines below should be uncommented:
#
#ports_externally_visible: 0    # none
#ports_externally_visible: 1    # ssh only
#ports_externally_visible: 2    # ssh + http-or-https (for Admin Console's box.lan/admin too)
#�ports_externally_visible: 3     # ssh + http-or-https + common IIAB services
#ports_externally_visible: 4    # ssh + http-or-https + common IIAB services + Samba
ports_externally_visible: 5    # all but databases
#
# Or further customize your iptables firewall by editing:
# /opt/iiab/iiab/roles/network/templates/gateway/iiab-gen-iptables
[... 301 lines cut ...]
    
    On 18/06/19 15:53, Tim Moody wrote:
    
    
      
      how did you install iiab, script from download.iiab.io? If you
        have not customized it we could just copy over one of the sample
        local vars.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56SC7TMAU5AJ2M4LD5LP3DZGFA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX64ZUI#issuecomment-503172305",
"url": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56SC7TMAU5AJ2M4LD5LP3DZGFA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX64ZUI#issuecomment-503172305",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

tim-moody, 2019-06-18:
bingo : #�ports_externally_visible: 3

mikerjensen, 2019-06-18:
Phew! thanks! all working now...
    
    
    On 18/06/19 16:26, Tim Moody wrote:
    
    
      
      bingo : #�ports_externally_visible: 3
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56UWPOTFD3YXEU753ZLP3D5BDA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX7AHZA#issuecomment-503186404",
"url": "https://github.com/iiab/iiab/issues/1741?email_source=notifications\u0026email_token=ABOY56UWPOTFD3YXEU753ZLP3D5BDA5CNFSM4HY7DSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODX7AHZA#issuecomment-503186404",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-18:
> Phew! thanks! all working now... On 18/06/19 16:26, Tim Moody wrote: bingo : #�ports_externally_visible: 3

@mikerjensen how did that illegal char get in there...any idea what editor you were using etc...that might have caused it?

closed 2019-06-18 by holta

tim-moody, 2019-06-18:
my bet is a del or destructive backspace key stroke that the terminal emulator took as a literal and recorded between the # and the ports