iiab/iiab #1744: nameserver setting

mikerjensen, 2019-06-19:
IIAB 7.0's /etc/resolv.conf sets nameserver to 127.0.0.1 - this doesn't work for me, and I can't change the settings on the DHCP server to which the RPI is connected via Ethernet, so after an RPI reboot i always need to edit /etc/resolv.conf and change the nameserver setting to an external DNS, e.g 8.8.8.8 

Is there a way to set this permanently?

tim-moody, 2019-06-19:
@mikerjensen we need to know the network topology

mikerjensen, 2019-06-19:
RPI connected via Ethernet cable to broadband/wireless router.

holta, 2019-06-19:
@mikerjensen please run:

systemctl status dnsmasq | pastebinit

And reply with the URL, Thanks!

mikerjensen, 2019-06-19:
Seems there is an unacceptable character for pastebinit...
    
    
    $ systemctl status dnsmasq | pastebinit
      Traceback (most recent call last):
        File "/usr/bin/pastebinit", line 424, in <module>
          params = urlencode(params)
        File "/usr/lib/python3.5/urllib/parse.py", line 801, in
      urlencode
          v = quote_via(str(v), safe, encoding, errors)
        File "/usr/lib/python3.5/urllib/parse.py", line 729, in
      quote_plus
          string = quote(string, safe + space, encoding, errors)
        File "/usr/lib/python3.5/urllib/parse.py", line 707, in quote
          string = string.encode(encoding, errors)
      UnicodeEncodeError: 'utf-8' codec can't encode character '\udce2'
      in position 0: surrogates not allowed
    pi@box:~ $ systemctl status dnsmasq 
      ● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS
      server
         Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled;
      vendor preset:
         Active: inactive (dead)
      
    
    On 19/06/19 13:39, A Holt wrote:
    
    
      
      @mikerjensen
        please run:
      systemctl status dnsmasq | pastebinit
      And reply with the URL, Thanks!
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56WCJ54WY2YYWJZNNDLP3ISHTA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYBXJPI#issuecomment-503542973",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56WCJ54WY2YYWJZNNDLP3ISHTA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYBXJPI#issuecomment-503542973",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-19:
@mikerjensen what is the output of `brctl show` ?

jvonau, 2019-06-19:
Ok looks like dnsmasq is not running, is br0 present?

holta, 2019-06-19:
@mikerjensen please also include the output of `ip a` Thanks!

jvonau, 2019-06-19:
systemctl iiab-dnsmasq status

holta, 2019-06-19:
> systemctl iiab-dnsmasq status

@mikerjensen I think @jvonau means `systemctl status iiab-dnsmasq` if you can send the output of that too please.

jvonau, 2019-06-19:
systemctl status iiab-dnsmasq

jvonau, 2019-06-19:
How is the IP address configured static or dhcp?

holta, 2019-06-19:
Some Background from @jvonau:

- If dnsmasq were runnning properly, 8.8.8.8 would (generally) not be necessary, as it dnsmasq uses the Internet root servers to build up its own DNS cache, if the ISP is failing to provide DNS.
- So it's very important to discover the root cause as to why dnsmasq is failing to start for @mikerjensen
- Force 8.8.8.8 (or the more privacy-conscious 1.1.1.1) is also possible if absolutely necessary, but let's uncover the root cause of the failure first

mikerjensen, 2019-06-19:
Thanks folks, sorry, i should have mentioned the IP address is
      configured static, so i guess DHCP setup is irrelevant..
    br0 is present:
    br0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
              inet 169.254.152.202  netmask 255.255.0.0  broadcast
      169.254.255.255
              ether 5a:66:3f:77:a5:0b  txqueuelen 1000  (Ethernet)
              RX packets 0  bytes 0 (0.0 B)
    
    ip a:
    1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state
      UNKNOWN group default qlen 1000
          link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
          inet 127.0.0.1/8 scope host lo
             valid_lft forever preferred_lft forever
      2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc
      pfifo_fast state UP group default qlen 1000
          link/ether b8:27:eb:e1:74:dd brd ff:ff:ff:ff:ff:ff
          inet 192.168.1.218/24 brd 192.168.1.255 scope global eth0
             valid_lft forever preferred_lft forever
      3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc
      pfifo_fast state UP group default qlen 1000
          link/ether b8:27:eb:b4:21:88 brd ff:ff:ff:ff:ff:ff
      4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc
      noqueue state UNKNOWN group default qlen 1000
          link/ether 5a:66:3f:77:a5:0b brd ff:ff:ff:ff:ff:ff
          inet 169.254.152.202/16 brd 169.254.255.255 scope global br0
             valid_lft forever preferred_lft forever
    
    
    
    systemctl status iiab-dnsmasq:
    ● iiab-dnsmasq.service - Network iiab-dnsmasq
         Loaded: loaded (/etc/systemd/system/iiab-dnsmasq.service;
      disabled; vendor pr
         Active: inactive (dead)
    
    brctl show:
    bridge name    bridge id        STP enabled    interfaces
      br0        8000.000000000000    no        
    
    
    
    On 19/06/19 14:01, Jerry Vonau wrote:
    
    
      
      How is the IP address configured static or dhcp?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56RXHAVEPDW4XMPRTS3P3IU3LA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYBZFKA#issuecomment-503550632",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56RXHAVEPDW4XMPRTS3P3IU3LA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYBZFKA#issuecomment-503550632",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

jvonau, 2019-06-19:
Please check /etc/dhcpcd/dhcpcd.conf for your static IP address information. Should have your IP/netmask/gateway/nameserver info present and correct

mikerjensen, 2019-06-19:
Thanks Jerry,
    
    /etc/dhcpcd.conf shows the correct IP settings but no nameserver
      settings. I set the static IP address using the IIAB admin console
      but there doesn't seem to be a field there to set a DNS..
    
    
    On 19/06/19 14:24, Jerry Vonau wrote:
    
    
      
      Please check /etc/dhcpcd/dhcpcd.conf for your static IP address
        information. Should have your IP/netmask/gateway/nameserver info
        present and correct
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56QEC7ZTXGHY7T7YO5LP3IXPRA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYB3GUA#issuecomment-503558992",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56QEC7ZTXGHY7T7YO5LP3IXPRA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYB3GUA#issuecomment-503558992",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-19:
@mikerjensen please also reply the URL resulting from `pastebinit < /etc/iiab/config_vars.yml`

Thx!

jvonau, 2019-06-19:
Can you post the contents /etc/iiab/config_vars.yml

mikerjensen, 2019-06-19:
http://paste.debian.net/1088482/
    
    On 19/06/19 14:34, Jerry Vonau wrote:
    
    
      
      Can you post the contents /etc/iiab/config_vars.yml
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56STCNRWZ6DEPI2MWY3P3IYWVA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYB4F2A#issuecomment-503562984",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56STCNRWZ6DEPI2MWY3P3IYWVA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYB4F2A#issuecomment-503562984",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-19:
@mikerjensen we'd really like to solve this properly, and permanently for you and all.

Any chance you can call me to schedule a remote session on your RPi machine?

Please write holt @ unleashkids.org if at all possible to arrange, Thanks!

holta, 2019-06-19:
Also @mikerjensen can you reply with these 3 URL's so we know for sure what we're dealing with?
```
pastebinit < /etc/iiab/iiab.env
pastebinit < /etc/iiab/iiab.init
pastebinit < /etc/iiab/local_vars.yml
```
Thanks!

mikerjensen, 2019-06-19:
Seems iiab.init doesn't exist...
    
     pastebinit < /etc/iiab/iiab.env
      http://paste.debian.net/1088495/
      pi@box:~ $ pastebinit < /etc/iiab/iiab.init
      -bash: /etc/iiab/iiab.init: No such file or directory
      pi@box:~ $ pastebinit < /etc/iiab/local_vars.yml
      http://paste.debian.net/1088496/
      pi@box:~ $ locate iiab.init
      
    
    On 19/06/19 15:47, A Holt wrote:
    
    
      
      Also @mikerjensen
        can you reply with these 3 URL's so we know for sure what we're
        dealing with?
      pastebinit < /etc/iiab/iiab.env
pastebinit < /etc/iiab/iiab.init
pastebinit < /etc/iiab/local_vars.yml

      Thanks!
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56TGRUANLIEMV6IUURTP3JBIVA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYCDRJY#issuecomment-503593127",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56TGRUANLIEMV6IUURTP3JBIVA5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYCDRJY#issuecomment-503593127",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-19:
> Seems iiab.init doesn't exist...

I meant /etc/iiab/iiab.ini but we've got live access to your machine now, making good progress towards a proper diagnosis/fix, such that you will be able to put your `wan_nameserver: 1.1.1.1` (similar to Google's 8.8.8.8) on future such machines to avoid all DNS hassles.

Noting that @jvonau has a day job, so this might take overnight possibly to wrap up, if that's ok?  Note that we're halfway there, with PR #1745 now merged and your IIAB (RPi) has brought up to latest IIAB 7.0 master fixes since you installed it May ~25th, so we're close but not entirely done yet, as:

1) dnsmasq is still not starting properly when running `cd /opt/iiab/iiab; ./iiab-network`
2) We're concerned that your br0 (bridge, as appears when running `ip a`) is still showing a bogus 169.* link-local IP address.

Thanks @mikerjensen for your patience!

holta, 2019-06-19:
Thank you @jvonau for your assistance here today!

When you have time, let's try to explain & resolve those 2 quirky issues mentioned in my note to @mikerjensen  just above, and also perhaps make general recommendations to Mike and other like him, around when-and-why he should choose [gateway vs. appliance vs. LAN Controller](https://github.com/iiab/iiab/wiki/IIAB-Networking#internet-in-a-box-iiab-networking) as his IIAB network topology.

holta, 2019-06-19:
CLARIFs:

> 1. dnsmasq is still not starting properly when running cd /opt/iiab/iiab; ./iiab-network

@jvonau confirms and I've tested that this was solved by today's PR #1745

> 2. We're concerned that your br0 (bridge, as appears when running ip a) is still showing a bogus 169.* link-local IP address.

@jvonau is no longer concerned: he suspects that appliance-mode was manually enabled within Admin Console earlier, sometime in the past month, perhaps during an attempt by @mikerjensen to test the internetarchive playbook &mdash; possibly even something we might have advised!

closed 2019-06-19 by holta

mikerjensen, 2019-06-20:
Thanks for figuring all this out, and you're quite right I did
      enable appliance mode manually, sorry i forgot to mention that -
      it seemed that 'appliance' was the appropriate option, as the
      'gateway mode' says you need 'two or more network adaptors' and as
      the RPI only has one, i assumed i needed the appliance mode.
    mike
    
    On 20/06/19 00:55, A Holt wrote:
    
    
      
      CLARIFs:
      
        
          dnsmasq is still not starting properly when running cd
            /opt/iiab/iiab; ./iiab-network
        
      
      @jvonau
        confirms and I've tested that this was solved by today's PR #1745
      
        
          We're concerned that your br0 (bridge, as appears when
            running ip a) is still showing a bogus 169.* link-local IP
            address.
        
      
      @jvonau
        is no longer concerned: he suspects that appliance-mode was
        manually enabled within Admin Console earlier, sometime in the
        past month, perhaps during an attempt by @mikerjensen
        to test the internetarchive playbook — possibly even something
        we might have advised!
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56QSUJVW4ERXVCVI543P3LBP3A5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYDTEUA#issuecomment-503788112",
"url": "https://github.com/iiab/iiab/issues/1744?email_source=notifications\u0026email_token=ABOY56QSUJVW4ERXVCVI543P3LBP3A5CNFSM4HZH3U52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYDTEUA#issuecomment-503788112",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2019-06-20:
> it seemed that 'appliance' was the appropriate option, as the 'gateway mode' says you need 'two or more network adaptors' and as the RPI only has one, i assumed i needed the appliance mode.

Either can work with RPi 3 (B+) here, and are not that different in the end,  no worries!