iiab/iiab #1747: dnsmasq can take upto 100% cpu in some cases

m-anish, 2019-06-19:
I did a IIAB install with docker as part of the initial install (azuracast role), and that caused dnsmasq to not startup at system boot

output of brctl show is this
```
bridge name	bridge id		STP enabled	interfaces
br-dc6d86a97ce0		8000.0242053d1c42	no		veth56337fa
							veth75efaf8
							vethc29d6ab
							vethd306b78
							vethf09e42c
br0		8000.02e5a74cf78f	no		wlp58s0
docker0		8000.02421c6334fb	no	
```

Then when I start dnsmasq as `service dnsmasq start`, after a little while, it takes 100% cpu
```

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                            
  773 systemd+  20   0   71884   7448   5596 R  98.7  0.1   4:14.61 systemd-resolve                                                    
 4422 dnsmasq   20   0   51580   2692   2300 R  49.5  0.0   2:05.83 dnsmasq 
```

m-anish, 2019-06-19:
Hardware: Intel NUC
OS: Ubuntu 18.04.2 LTS

holta, 2019-06-19:
Thanks @m-anish.

It turns out dnsmasq was more issues than we ever imagined.

Thanks everyone for your patience as @jvonau & others are helping to work thru these issues one at a time, e.g. dnsmasq sometimes doesn't start at all in static IP address situations like this one, that we're working on at this moment:

#1744 #1745

m-anish, 2019-06-20:
While this gets resolved, can you suggest how I should go about installing IIAB in LAN-controller mode?

My current plan is:
* Install IIAB in gateway
* Install azuracast
* Remove USB-ethernet
* Rerun install OR rerun network setup

Thoughts?

tim-moody, 2019-06-20:
I would skip the last step. For me there is no difference between Lan controller and Gateway without a WAN and it means the USB-ethernet can be reconnected at any time.

holta, 2019-06-23:
@m-anish please run `/usr/bin/iiab-diagnostics` to post more detailed diagnostics when possible / where possible towards solving this in coming days?

As explained here:
https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md

Per PR #1763

m-anish, 2019-06-24:
Dnsmasq and systemd-resolved are taking 100% cpu again

/etc/resolv.conf
http://paste.ubuntu.com/p/PH4MSBG9CK/ 

/etc/systemd/resolved.conf
http://paste.ubuntu.com/p/NPWzVRR7mp/

m-anish, 2019-06-24:
service systemd-resolved status
http://paste.ubuntu.com/p/yZZ5FPGp7K/

m-anish, 2019-06-24:
systemd-resolve --status
http://paste.ubuntu.com/p/dwf9nJnYbw/

m-anish, 2019-06-24:
this is the tcpdump output when the system's going crazy 

https://paste.ubuntu.com/p/QhNfQF5h6Y/

jvonau, 2019-06-24:
Upstream reference https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1670959

m-anish, 2019-06-25:
Checkout #1773 which fixes this!

commit 4b4d3498 (2019-06-25): Merge pull request #1773 from m-anish/dnsmasq-systemd-resolved-race

Fix #1747 | dnsmasq and systemd-resolved race issues
closed 2019-06-25 by holta