iiab/iiab #1749: OpenVPN is very commonly needed when DNS is failing

holta, 2019-06-19:
@georgejhunt can OpenVPN be made more resilient, so that it works when DNS is not available?

This would make OpenVPN an order of magnitude more useful, and allow IIAB to much more rapidbly work through networking bugs, as are ever-present across OS's and evolving Internet standards, evolving OS's/protocols etc.

As OpenVPN's most important use case is troubleshooting complex situations when DNS is inevitably failing or absent for a wide assortment of reasons.

TeamViewer, for example, works when DNS is absent, for much the same reasons.

Do you know if this might be possible?

holta, 2019-06-19:
Good news @georgejhunt believes it is likely possible to run OpenVPN when DNS is not working.

There is a concern than the IP address of IIAB's "OpenVPN cloud" might occasionally change however.

I've asked George if he has time to investigate various strategies to make OpenVPN substantially more resilient across many networking situations, and ideally several years into the future.  In Short: there might be contractual issues with our upstream "OpenVPN cloud" provider, whereby they might possibly change the IP address(es) from time to time &mdash; or possible another better/cheaper cloud provider becomes necessary in the future.

e.g. in any case, it might be possible to make OpenVPN into a substantially more robust diagnosis/troubleshooting tool, by checking **both** the "most recently known IP address" **and** also using DNS ***if*** indeed DNS happens to be working at that moment in time.

Noting that  OpenVPN-like tools are already a liefsaver and a lifeline for so many of these distant machines, where their background/context is so often extremely unclear, until a live connection is established...  Thank you to George and others who've made these OpenVPN tools possible!

ASIDE: #1596 is another key tool the IIAB community should add to its "arsenal" in 2019, to strengthen all countries' collective wisdom here, building out a more rock solid community product.

georgejhunt, 2019-06-20:
> possibly change the IP address(es) from time to time — or possible another

Looks like we could have ip and dns references to the same server
https://unix.stackexchange.com/questions/183425/can-you-set-openvpn-to-auto-connect-to-fallback-ovpns-in-case-the-current-one

holta, 2019-06-20:
> Looks like we could have ip and dns references to the same server https://unix.stackexchange.com/questions/183425/can-you-set-openvpn-to-auto-connect-to-fallback-ovpns-in-case-the-current-one

Interesting, as I'd only been thinking about establishing the initial OpenVPN connection using IP and/or DNS name, one as a fallback after the other, depending on which is available at that time.

But OpenVPN **also** includes handling of IP's & DNS names during connections and/or after connections fail, that's more powerful yet _(and why not avoid building out this kind of logic using Ansible, if some/all of it's already done, so we don't have to!)_

holta, 2019-06-27:
We hope this is dramatically improved by PR #1752 

Please re-open as nec, if further improvement is nec.

closed 2019-06-27 by holta