iiab/iiab #1755: Tracking tkt for "Download if filename is changed [OSM maps]" iiab/iiab-admin-console#211

holta, 2019-06-20:
@georgejhunt if https://github.com/iiab/iiab-admin-console/pull/211 is revised and/or merged in coming days, what do I then need to change within https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack ?

Thanks!

holta, 2019-06-24:
Solving these 4 tkts before releasing IIAB 7.0 Preview 2 "Thursday in 3 days from now" would be tremendous if we're so lucky:

- PR #1752 Add client config that works when DNS does not [OpenVPN]
- #1755 Download if filename is changed [OSM maps]
- #1771 Sugarizer fails on latest master [only when freshly installed!]
- #1747 dnsmasq can take up to 100% cpu in some cases

holta, 2019-06-26:
> @georgejhunt if [iiab/iiab-admin-console#211](https://github.com/iiab/iiab-admin-console/pull/211) [ -> https://github.com/[iiab/iiab-admin-console/pull/224](https://github.com/iiab/iiab-admin-console/pull/224) ] is revised and/or merged in coming days, what do I then need to change within https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack ?

@georgejhunt when https://github.com/iiab/iiab-admin-console/pull/224 is complete, plz tell me how to revise the user-facing doc above, Thanks!

georgejhunt, 2019-06-26:
> @georgejhunt <https://github.com/georgejhunt> if
> iiab/iiab-admin-console#211
> <https://github.com/iiab/iiab-admin-console/pull/211> [ ->
> https://github.com/iiab/iiab-admin-console/pull/224
> <https://github.com/iiab/iiab-admin-console/pull/224> ] is revised and/or
> merged in coming days, what do I then need to change within
> https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack
> ?
>
> @georgejhunt <https://github.com/georgejhunt> when
> iiab/iiab-admin-console#224
> <https://github.com/iiab/iiab-admin-console/pull/224> is complete, plz
> tell me how to revise the user-facing doc above, Thanks!

Changed, and current I think

holta, 2019-06-27:
> Changed, and current I think

Thanks much & thanks to @tim-moody for merging @georgejhunt's iiab/iiab-admin-console#224

Hope it's Ok I clarified some of the historical stuff in context, in the spirit of open source / transparency / full disclosure.

closed 2019-06-27 by holta