iiab/iiab #1756: Admin Console unable to remove "Boundless (Static) Textbooks" from IIAB home page

holta, 2019-06-21:
@randytate writes:

Boundless Textbooks
Dropped the "static" option from menu
Tried to use /admin to remove the module, _but it didn't work ...._

_While trying to remove the boundless-static module, /admin would timeout on calls to cmd-service.
After multiple tries, I decided to reboot the Pi.
It came back up fine, but now there are errors just loading /admin ..._

```
GET-WHLIST FAILED and reported IIAB-CMDSRV failed to initialize
```

[image]

It is running:

[image]

Rebooted again ... same result.  Appreciate any pointers you might have.

tim-moody, 2019-06-21:
Most likely one of the files read during init is corrupt. What does journalctl -b -t IIAB-CMDSRV say, anything useful. The actions described should not affect any init files.

Not sure I understand 

> Boundless Textbooks
Dropped the "static" option from menu
Tried to use /admin to remove the module, but it didn't work ....

What do 'dropped' and 'didn't work mean'?

tim-moody, 2019-06-21:
To diagnose further as root type:
systemctl stop iiab-cmdsrv
cd /opt/admin/cmdsrv/
./iiab-cmdsrv

report any errors

tim-moody, 2019-06-21:
I will make the try except on init more granular as it currently only traps the entire process.

holta, 2019-06-21:
> > Boundless Textbooks
> > Dropped the "static" option from menu
> > Tried to use /admin to remove the module, but it didn't work ....
> 
> What do 'dropped' and 'didn't work mean'?

Sounds like @randytate wants to use **Admin Console** (http://box.lan/admin) > **Content Menus** > **Content Item List** to remove http://oer2go.org/viewmod/en-boundless-static from his IIAB home page, using **Drag & Drop** there.

Possibly to swap that out in favor of http://oer2go.org/viewmod/en-boundless

When he says 'didn't work' probably he's just saying he can't login to Admin Console?  Hopefully he can confirm more details later today.

holta, 2019-06-21:
Background / FYI I'm pretty sure @randytate is running a very recent IIAB 7.0 pre-release (master branch of both repos, which he's updated sometime in the past few days) on Raspbian With Desktop, on RPi 3 B+.

randytate, 2019-06-23:
@holta is correct -- recent version of IIAB 7.0 from within the last week, on RPi 3 B+.

----
I see a few of these in the output of `sudo journalctl -b -t IIAB-CMDSRV`
```
Jun 24 00:36:15 box.lan IIAB-CMDSRV[1411]: IIAB-CMDSRV : Error: INST-OSM-VECT-SET {osm_vect_id:africa} duplicates an Active Command.
```

----
> What do 'dropped' and 'didn't work mean'?

Previously when trying to remove the boundless-static module (http://box.lan/admin, then Install Content, Manage Content, select "boundless-static", click Remove Selected Content) I got a browser alert with some error.  I tried again today, got another alert about osm_vect_id:africa duplicates (as above).  But when I went back into the Manage Content list, it was gone, and also gone from `/library/www/html/modules/`

----
Output of running iiab-cmdsrv in shell:

```
root@box:/library# systemctl stop iiab-cmdsrv
root@box:/library# cd /opt/admin/cmdsrv/
root@box:/opt/admin/cmdsrv# ./iiab-cmdsrv
in job_minder_thread
sending message server received from client to worker GET-ZIM-STAT id k�Eh
sending data message server received from client to worker GET-ZIM-STAT id k�Eh
sending message server received from client to worker GET-ZIM-STAT id k�Eh
sending message server received from client to worker GET-SPACE-AVAIL id k�Ei
sending data message server received from client to worker GET-SPACE-AVAIL id k�Ei
sending message server received from client to worker GET-SPACE-AVAIL id k�Ei
sending message server received from client to worker GET-SPACE-AVAIL id k�Ej
sending data message server received from client to worker GET-SPACE-AVAIL id k�Ej
sending message server received from client to worker GET-SPACE-AVAIL id k�Ej
sending message server received from client to worker GET-ZIM-STAT id k�Ek
sending data message server received from client to worker GET-ZIM-STAT id k�Ek
Sending worker message to client {"INSTALLED": {"c2b8c685-6a6f-edf7-c8d8-a5005144c8bf": {"pub id k�Eh
sending message server received from client to worker GET-ZIM-STAT id k�Ek
Sending worker message to client {"library_on_root": true, "root": {"avail_in_k": "47361992", id k�Ei
Sending worker message to client {"library_on_root": true, "root": {"avail_in_k": "47361992", id k�Ej
Sending worker message to client {"INSTALLED": {"c2b8c685-6a6f-edf7-c8d8-a5005144c8bf": {"pub id k�Ek
```

With iiab-cmdsrv running in a shell, I navigated to http://box.lan/admin, then to Install Content, Manage Content, and got a browser popup: `LIST-LIBR FAILED and reported CMDSRV has started but is not ready.`

tim-moody, 2019-06-24:
it is possible that if running cmdsrv in a shell it never clears out the running flags, but then all these other message would have been from a different session.

holta, 2019-06-27:
> it is possible that if running cmdsrv in a shell it never clears out the running flags, but then all these other message would have been from a different session.

Any news here @randytate ? No worries but keep in touch if this should be closed and/or progress can be made!

holta, 2019-07-03:
@randytate please re-open and/or let us know if this problem recurs for you or anybody in Uganda for IIAB 7.0 or 7.1, Thanks!

closed 2019-07-03 by holta