iiab/iiab #1758: Admin Console > Install Content > Manage Content freezes screen (Ubuntu & Debian)

holta, 2019-06-21:
The Admin Console GUI (in browser) fades (screen becomes somewhat dark but still visible) when you click "Manage Content", even after reboot on:
- Ubuntu 18.04 http://10.8.0.26/admin
- "Debian 10" http://10.8.0.22/admin

This doesn't happen on Raspbian.

(If this is hard to fix, perhaps the "Manage Content" button should just be grayed out so Ubuntu/Debian folks don't click on it?  Or perhaps a popup should say "this functionality is supported on Raspbian only for now" ?  Or if this turns out to be fixable across Ubuntu/Debian, great!)

tim-moody, 2019-06-21:
I am looking at this on a ub 18 vm and not seeing this. I pulled the latest version and don't see it there either.

tim-moody, 2019-06-21:
this is probably a javascript problem. use f12 to view console

holta, 2019-06-21:
> this is probably a javascript problem. use f12 to view console

I'm away from computer most of today. Can you check these 2 URL's directly?

> Ubuntu 18.04 http://10.8.0.26/admin
> "Debian 10" http://10.8.0.22/admin
This doesn't happen on Raspbian.

tim-moody, 2019-06-21:
likely a race condition on modal

holta, 2019-06-21:
Solved by https://github.com/iiab/iiab-admin-console/pull/215 ?

holta, 2019-06-24:
> Solved by [iiab/iiab-admin-console#215](https://github.com/iiab/iiab-admin-console/pull/215) ?

Is this likely resolved?

holta, 2019-06-25:
Is this being solved by ?

https://github.com/iiab/iiab-admin-console/pull/223 "remove fade that causes race condition on spinner modal"

tim-moody, 2019-06-25:
thinks so, but testing appreciated. not totally happy with the solution, but seems to work.

holta, 2019-06-25:
Thanks @tim-moody -- clicking http://box.lan/admin -> `Install Content` > `Manage Content` now works on both!
- Ubuntu 18.04 http://10.8.0.26/admin
- "Debian 10" http://10.8.0.22/admin

closed 2019-06-25 by holta