iiab/iiab #1762: IIAB Analytics above & beyond AWStats

holta, 2019-06-22:
Some IIAB field projects want deeper community insights, to understand important/subtle usage patterns in ways that AWStats is not able to offer.

One of the most common question is "how many unique cellphones used this IIAB over the past day/week/month?"

However we seek to go about answering these kinds of questions, there are huge/tricky privacy-versus-surveillance questions if HW GUID's or cookies are tracked to make this happen.

Let's discuss what's possible in 2019 &mdash; to give IIAB project leaders stronger insights into the patterns of how communities are using IIAB &mdash; while taking unintended antiprivacy consequences very seriously, so that disadvantaged populations are never abused based on their library reading/viewing/listening habits.

tim-moody, 2019-06-22:
what awstats offers is substantially more sophisticated than this request. if we assume that dnsmasq routinely gives the same device the same ip, then it even support to the count the devices request.

holta, 2019-06-22:
> what awstats offers is substantially more sophisticated than this request. if we assume that dnsmasq routinely gives the same device the same ip, then it even supports the count the devices request.

What is there are more than 256 users/devices?  (Even if the above assumption were to be validated.)

holta, 2019-06-28:
It's very likely that IIAB should drive to include [Matomo Analytics](https://matomo.org/) for IIAB 7.1 — for local operators, teachers, implementers,  content developers — and yes also kids/patients/citizens wanting to better understand their own community.

A hosted version should also be offered, offering a modern dashboard for NGO's, content partners and similar.

Difficult privacy and connectivity issues are rife, at every level.  But this does NOT mean we and the communities depending on Internet-in-a-Box can afford to bury our heads in the sand.  Taking on these extremely challenging but extremely important questions WILL take time, and won't be at all easy, however...

https://en.wikipedia.org/wiki/The_unexamined_life_is_not_worth_living

m-anish, 2019-07-08:
related/duplicate #1846
related #1845

holta, 2019-07-09:
We need an action plan too, but starting with @mikkokotila's thought-provoking philosophical summary presentation yesterday was a good start towards answering critical question in front of many IIAB communities: _(whether this developing world work is supported by Small, Medium or Large NGO's etc!)_

- Thank You for Ad Blocking: The Greatest Boycott in Human History
  https://www.slideshare.net/MikkoKotila/thank-you-for-ad-blocking-the-greatest-boycott-in-human-history
- Privacy as 1 of 4 mindsets/modes about personal data...
  1) Anonymous (activists, journalists, etc)
  2) Secret (with a few/chosen 3rd parties)
  3) Privacy (control over your PERSONAL data)
  4) Public
  5) Absence of above 4? ("i don't care")
- That was the individual/personal side.  Now 5 Questions for Organizations...
  1) Possible/desirable/saleable?
  2) Practical/feasible?
  3) Legal e.g. HIPPA compliance of medic data? GDPR? HIDDEN Liability: India's "privacy" law requiring all ISP's-or-similar-netwok-providers to provide nearly all records to authorities?
  4) Is it moral?
  5) _Do I want this for my own kids?_
- "Let's talk about people not users..." &mdash; 3 C's of records:
  - Collection (is record created?)
  - Control (of people vs. by people themselves)
  - Comprehension (informed consent!)
- System Design should preserve power in the hands of the grassroots communities & individuals...at the same time how to we empower communities / NGO's / content designers with very real "fieldback" loops evolving/iterating responsively to grassroots communities actual needs?
- 2 Technical Means to collect web-like stats:
  - Log-based e.g. Apache / AWStats / etc (for graphics/charts)
  - Tag-based e.g. JavaScript (tracker tags)
- Even "Ad Tech" people misunderstand "Cookie Sink" e.g. Princeton Study: avg news sites have more than 30, each containing more tags, ad inifinitum.  AKA network-wide propagation -> unmanageable personal data.
  - Risk: we (IIAB) need to avoid such approaches.
- What is "JUST ENOUGH" data to be collected over the coming 5-10 years?
  - Cross-device tracking
  - "Householding" attempts to monitor ppl across devices?
- Traditionally variables are derived from:
  - user-agent
  - IP addresses / subnets
  - Models created from above using crypto hashes to abandon original data ... pseudo-anonymization?
  - https://en.wikipedia.org/wiki/Pseudonymization (and associated risks, when de-identified personal data gets RE-identified)
- Data Minimization / Unintended Consequences regardless:
  - https://medium.com/@mikkokotila/five-essential-points-on-data-visualization-2856b80730b3
- Anthropological/ethnographic/sociological approaches are orthogonal but always relevant (during those very rare times they actually happen)
  - AI "learning models"

holta, 2019-11-15:
Tangentially-related discussion of field community survey software tools...e.g. LimeSurvey vs KoboToolbox:

#1355

holta, 2020-02-02:
Just FYI IIAB's AWStats installer was just reworked for NGINX & significantly cleaned up earlier today:

#2227 "Cleaner AWStats (basic analytics) playbook for IIAB"

_(Whether or not Internet-in-a-Box communities & offline digital communities in general find a way to embrace [Matomo](https://matomo.org/) - on - NGINX or similar this year!)_

PS Matomo's current slogan is "Ethical Analytics, Powerful Insights" which is quite unusual in the data science business (:

holta, 2020-02-14:
Human-driven surveys are not at all unrelated: _(and occasionally can even deliver was more insights!)_

#2041 Survey software: KoboToolbox server / ODK Collect

holta, 2020-11-03:
Thanks @tim-moody for noting:

> Matomo will issue you with a JavaScript tag. This code must appear on every page that you want Matomo to analyze.

"Add Google Analytics To Your Pages With Nginx's sub_filter Directive"
> https://adarrohn.com/blog/nginx-google-analytics

holta, 2020-11-23:
FYI Matomo 4.0.0 was just [released](https://github.com/matomo-org/matomo/releases/tag/4.0.0), almost 4 years after Matomo 3.0.0

As they roll out this announcement in coming days, let's look out for (hopefully!) clear explanations + illustrations &mdash; as to what kinds of insights this new Matomo might be capable of helping to assist offline communities with~

holta, 2020-12-03:
"Announcing Matomo 4 - More security, privacy and a new feature"
https://matomo.org/blog/2020/11/announcing-matomo-4/

holta, 2022-03-24:
See also:

- PR #2511
- #2661
- https://github.com/cwivagg/iiab_analytics/blob/main/notebooks/notebook_matomo-setup.ipynb

holta, 2022-08-15:
FYI Matomo 4.11+ now works with IIAB, for those who need it:

- README: https://github.com/iiab/iiab/tree/master/roles/matomo#readme
- PR #3304