iiab/iiab #1766: AzuraCast fails to install on BIG-sized IIAB on "Debian 10" VM

holta, 2019-06-23:
@m-anish & @jvonau please take a quick look at the error below.

(VM/OS settings & configuration detailed within http://paste.debian.net/1088960/ thanks to/usr/bin/iiab-diagnostics.)

> TASK [azuracast : AzuraCast - Run the installer] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/bin/bash docker.sh install", "delta": "0:00:25.723063", "end": "2019-06-23 07:19:55.421552", "msg": "non-zero return code", "rc": 1, "start": "2019-06-23 07:19:29.698489", "stderr": "+ sh -c apt-get update -qq >/dev/null\n+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null\n+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null\nWarning: apt-key output should not be parsed (stdout is not a terminal)\n+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list\n+ sh -c apt-get update -qq >/dev/null\nE: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.\nsudo: unable to resolve host box.lan: Name or service not known\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100   617    0   617    0     0   3247      0 --:--:-- --:--:-- --:--:--  3247\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r 45 15.4M   45 7139k    0     0  5729k      0  0:00:02  0:00:01  0:00:01 7175k\r100 15.4M  100 15.4M    0     0  7899k      0  0:00:01  0:00:01 --:--:-- 9040k\nsudo: unable to resolve host box.lan: Name or service not known\nsudo: unable to resolve host box.lan: Name or service not known\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100 13258  100 13258    0     0   114k      0 --:--:-- --:--:-- --:--:--  113k\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100  2586  100  2586    0     0  16576      0 --:--:-- --:--:-- --:--:-- 16576\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/", "stderr_lines": ["+ sh -c apt-get update -qq >/dev/null", "+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null", "+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null", "Warning: apt-key output should not be parsed (stdout is not a terminal)", "+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list", "+ sh -c apt-get update -qq >/dev/null", "E: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.", "sudo: unable to resolve host box.lan: Name or service not known", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100   617    0   617    0     0   3247      0 --:--:-- --:--:-- --:--:--  3247", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", " 45 15.4M   45 7139k    0     0  5729k      0  0:00:02  0:00:01  0:00:01 7175k", "100 15.4M  100 15.4M    0     0  7899k      0  0:00:01  0:00:01 --:--:-- 9040k", "sudo: unable to resolve host box.lan: Name or service not known", "sudo: unable to resolve host box.lan: Name or service not known", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100 13258  100 13258    0     0   114k      0 --:--:-- --:--:-- --:--:--  113k", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100  2586  100  2586    0     0  16576      0 --:--:-- --:--:-- --:--:-- 16576", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/"], "stdout": "Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48\nDocker Compose does not appear to be installed. Install Docker Compose now? [Y/n] Creating default AzuraCast settings file...", "stdout_lines": ["Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48", "Docker Compose does not appear to be installed. Install Docker Compose now? [Y/n] Creating default AzuraCast settings file..."]}
> 
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=216  changed=54   unreachable=0    failed=1    skipped=79   rescued=0    ignored=0

m-anish, 2019-06-23:
I guess its a question for the azuracast community. Will check docs.

holta, 2019-06-23:
When trying again using `cd /opt/iiab/iiab` then `./runrole azuracast`, the error is similar but shorter:

> TASK [azuracast : AzuraCast - Run the installer] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/bin/bash docker.sh install", "delta": "0:00:18.764472", "end": "2019-06-23 07:41:52.304876", "msg": "non-zero return code", "rc": 1, "start": "2019-06-23 07:41:33.540404", "stderr": "+ sh -c apt-get update -qq >/dev/null\nE: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/", "stderr_lines": ["+ sh -c apt-get update -qq >/dev/null", "E: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/"], "stdout": "Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48\nDocker Compose is already installed! Continuing...", "stdout_lines": ["Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48", "Docker Compose is already installed! Continuing..."]}

holta, 2019-06-23:
On @m-anish's suggestion, this was also tried &mdash; no luck so far:

> root@box:/opt/azuracast# ./docker.sh install
Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48
> + sh -c apt-get update -qq >/dev/null
E: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.
Docker Compose is already installed! Continuing...
ERROR: Couldn't connect to Docker daemon. You might need to install Docker:
>
> https://docs.docker.com/engine/installation/
ERROR: Couldn't connect to Docker daemon. You might need to install Docker:
>
> https://docs.docker.com/engine/installation/
ERROR: Couldn't connect to Docker daemon. You might need to install Docker:
>
> https://docs.docker.com/engine/installation/

m-anish, 2019-06-23:
Cool. This needs to be handled upstream. I've already marked the ticket in Azuracast to link here...

holta, 2019-06-23:
> Cool. This needs to be handled upstream. I've already marked the ticket in Azuracast to link here...

Thanks for explaining.  Aside/FYI: note that `./docker.sh install` never actually prompts for `[Yn]` above, even though it claims to...there is time (or place) to type anything, as it continues on its own, and crashed within less than 10 seconds, returning you to the prompt.

m-anish, 2019-06-23:
Yes, thats because in my playbook, i tinker with the script to not ask for y/n, and just assume y everytime

holta, 2019-06-23:
Ah.

FYI for @jvonau who requested it, here's the 347-line /opt/azuracast/docker.sh: http://paste.debian.net/1088962/

holta, 2019-06-23:
@jvonau asks if the [AzuraCast playbook](https://github.com/iiab/iiab/tree/master/roles/azuracast) should perhaps be disabled for Debian 9 / Debian 10 both, and restricted to Ubuntu 18+ only for now / possibly

m-anish, 2019-06-23:
https://github.com/AzuraCast/AzuraCast/issues/1606#issuecomment-504747865

Once Debian 10 is released, this should fix itself.

holta, 2019-07-04:
> [AzuraCast/AzuraCast#1606 (comment)](https://github.com/AzuraCast/AzuraCast/issues/1606#issuecomment-504747865)
> 
> Once Debian 10 is released, this should fix itself.

Hope so!  Still fails in a new test today, 2 days prior to Debian 10's "Saturday 2019-07-06" release, just FYI (:

The Debian 10 VM in question was a BIG-sized IIAB (10.8.0.30) with 2GB RAM, and plenty of disk space, that failed to install AzuraCast as follows:

> TASK [azuracast : AzuraCast - Run the installer] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/bin/bash docker.sh install", "delta": "0:00:12.859991", "end": "2019-07-04 00:08:48.784623", "msg": "non-zero return code", "rc": 1, "start": "2019-07-04 00:08:35.924632", "stderr": "+ sh -c apt-get update -qq >/dev/null\n+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null\n+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null\nWarning: apt-key output should not be parsed (stdout is not a terminal)\n+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list\n+ sh -c apt-get update -qq >/dev/null\nE: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/", "stderr_lines": ["+ sh -c apt-get update -qq >/dev/null", "+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null", "+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null", "Warning: apt-key output should not be parsed (stdout is not a terminal)", "+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list", "+ sh -c apt-get update -qq >/dev/null", "E: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/"], "stdout": "Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48\nDocker Compose is already installed! Continuing...", "stdout_lines": ["Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48", "Docker Compose is already installed! Continuing..."]}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=30   changed=7    unreachable=0    failed=1    skipped=14   rescued=0    ignored=0

Refs PRs #1733 #1817

holta, 2019-07-07:
No luck, anyway not yet at this time: (24h after Debian 10's release)

> TASK [azuracast : AzuraCast - Run the installer] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/bin/bash docker.sh install", "delta": "0:00:19.571178", "end": "2019-07-07 19:01:47.003419", "msg": "non-zero return code", "rc": 1, "start": "2019-07-07 19:01:27.432241", "stderr": "+ sh -c apt-get update -qq >/dev/null\n+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null\n+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null\nWarning: apt-key output should not be parsed (stdout is not a terminal)\n+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list\n+ sh -c apt-get update -qq >/dev/null\nE: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/\nCouldn't connect to Docker daemon. You might need to install Docker:\n\nhttps://docs.docker.com/engine/installation/", "stderr_lines": ["+ sh -c apt-get update -qq >/dev/null", "+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null", "+ sh -c curl -fsSL \"https://download.docker.com/linux/debian/gpg\" | apt-key add -qq - >/dev/null", "Warning: apt-key output should not be parsed (stdout is not a terminal)", "+ sh -c echo \"deb [arch=amd64] https://download.docker.com/linux/debian 10 stable\" > /etc/apt/sources.list.d/docker.list", "+ sh -c apt-get update -qq >/dev/null", "E: The repository 'https://download.docker.com/linux/debian 10 Release' does not have a Release file.", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/", "Couldn't connect to Docker daemon. You might need to install Docker:", "", "https://docs.docker.com/engine/installation/"], "stdout": "Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48\nDocker Compose is already installed! Continuing...", "stdout_lines": ["Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48", "Docker Compose is already installed! Continuing..."]}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=35   changed=9    unreachable=0    failed=1    skipped=14   rescued=0    ignored=0

m-anish, 2019-07-08:
Lets wait a week :)

holta, 2019-08-04:
Appears to have resolved itself on Debian 10.0.0 now, in any case the 'AzuraCast First-Time Setup' screen appears at http://box.lan:10080/setup/register

Getting AzuraCast to work on RPi is the outstanding challenge here (at #1772).

closed 2019-08-04 by holta