iiab/iiab #1767: Rpi Prob: "Copy script to restart dnsmasq whenever br0 comes up" (Destination directory /etc/networkd-dispatcher/routable.d does not exist)

holta, 2019-06-23:
Error is:

> TASK [network : Copy script to restart dnsmasq whenever br0 comes up] **********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "checksum": "50de730947aa4cccc1586ff91d9f6bfc984b5127", "msg": "Destination directory /etc/networkd-dispatcher/routable.d does not exist"}
>
> 
PLAY RECAP *********************************************************************
127.0.0.1                  : ok=551  changed=143  unreachable=0    failed=1    skipped=241  rescued=0    ignored=0

Machine/OS background details: http://paste.debian.net/1088961/

m-anish, 2019-06-23:
okay, there should be a check somewhere. ill take a look

holta, 2019-06-23:
SOlved by PR #1768 it appears, Thanks @m-anish !

closed 2019-06-23 by holta