iiab/iiab #1771: Sugarizer fails on latest master

m-anish, 2019-06-24:
### Expected behavior and actual behavior.
```
Jun 24 05:00:00 box sugarizer[9962]: FATAL ERROR, uncaught exception 'expressValidator is not a function'
Jun 24 05:00:00 box sugarizer[9962]: TypeError: expressValidator is not a function
Jun 24 05:00:00 box sugarizer[9962]:     at module.exports (/opt/iiab/sugarizer-server-1.1.0/config/main.js:46:10)
Jun 24 05:00:00 box sugarizer[9962]:     at /opt/iiab/sugarizer-server-1.1.0/sugarizer.js:31:26
Jun 24 05:00:00 box sugarizer[9962]:     at /opt/iiab/sugarizer-server-1.1.0/config/wait4db.js:20:6
Jun 24 05:00:00 box sugarizer[9962]:     at result (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/utils.js:410:17)
Jun 24 05:00:00 box sugarizer[9962]:     at executeCallback (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/utils.js:402:9)
Jun 24 05:00:00 box sugarizer[9962]:     at err (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo_client_ops.
Jun 24 05:00:00 box sugarizer[9962]:     at connectCallback (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo
Jun 24 05:00:00 box sugarizer[9962]:     at server.connect (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo_
Jun 24 05:00:00 box sugarizer[9962]:     at Server.connectHandler (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/topologies
Jun 24 05:00:00 box sugarizer[9962]:     at Object.onceWrapper (events.js:286:20)
Jun 24 05:00:00 box systemd[1]: sugarizer.service: Main process exited, code=exited, status=255/n/a
Jun 24 05:00:00 box systemd[1]: sugarizer.service: Failed with result 'exit-code'.
```

### Steps to reproduce the problem.
Install from scratch on 18.04

### Specifications like the version of the IIAB, operating system version, or hardware details.
OS: 18.04
H/w: Intel NUC
IIAB in gateway mode

holta, 2019-06-24:
@m-anish 's above log/output is apparently from `journalctl -xe` or `journalctl -u sugarizer`

holta, 2019-06-24:
@llaske do you know why Sugarizer's failing to start when freshly installed on Ubuntu 18.04 and Raspbian?  (Strangely Sugarizer i.e. http://box/sugarizer works on pre-existing installs on these same OS's, even when their IIAB code is updated to the latest master!)

MongoDB is running fine, but Sugarizer's systemd service fails almost immediately after starting (within about 1 second) every time we try, on both OS's.

e.g. a fresh install of IIAB 7.0's near-final pre-release that I did on Raspbian Lite shows essentially the exact same thing that @m-anish is experiencing above:

```
root@box:~# systemctl is-active mongodb
active

root@box:~# systemctl is-active sugarizer
activating

root@box:~# journalctl -xe
...
-- Subject: Unit sugarizer.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit sugarizer.service has finished shutting down.
Jun 24 03:09:43 box.lan systemd[1]: Started Node.js Sugarizer Server.
-- Subject: Unit sugarizer.service has finished start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit sugarizer.service has finished starting up.
--
-- The start-up result is done.
Jun 24 03:09:46 box.lan sugarizer[7865]: FATAL ERROR, uncaught exception 'expressValidator is not a function'
Jun 24 03:09:46 box.lan sugarizer[7865]: TypeError: expressValidator is not a function
Jun 24 03:09:46 box.lan sugarizer[7865]:     at module.exports (/opt/iiab/sugarizer-server-1.1.0/config/main.js:46:10)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at /opt/iiab/sugarizer-server-1.1.0/sugarizer.js:31:26
Jun 24 03:09:46 box.lan sugarizer[7865]:     at /opt/iiab/sugarizer-server-1.1.0/config/wait4db.js:20:6
Jun 24 03:09:46 box.lan sugarizer[7865]:     at result (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/utils.js:410:17)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at executeCallback (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/utils.js:402:9)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at err (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo_client_ops.js:286:5)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at connectCallback (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo_client_ops.js:265:5)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at server.connect (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/operations/mongo_client_ops.js:379:5)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at Server.connectHandler (/opt/iiab/sugarizer-server-1.1.0/node_modules/mongodb/lib/topologies/server.js:298:9)
Jun 24 03:09:46 box.lan sugarizer[7865]:     at Object.onceWrapper (events.js:286:20)
Jun 24 03:09:46 box.lan systemd[1]: sugarizer.service: Main process exited, code=exited, status=255/n/a
Jun 24 03:09:46 box.lan systemd[1]: sugarizer.service: Unit entered failed state.
Jun 24 03:09:46 box.lan systemd[1]: sugarizer.service: Failed with result 'exit-code'.
```

llaske, 2019-06-24:
Hmmm. Strange. Could you check the node.js version installed and ensure that « npm install » command has been launched ?

     Lionel

holta, 2019-06-24:
> Hmmm. Strange. Could you check the node.js version installed and ensure that « npm install » command has been launched ? Lionel

On IIAB 7.0 pre-release on Raspbian Lite on RPi 3 B+:
```
root@box:~# nodejs -v
v10.16.0
```

`npm install --allow-root --unsafe-perm=true` was run successfully (as we've always done over the last year) as can be seen by this log file (/opt/iiab/iiab/iiab-install.log) below.  Quite confusing that (initially!) Sugarizer (and Sugarizer Server) 1.1.0 appear to install correctly:
```
2019-06-23 21:20:50,893 p=11418 u=root |  TASK [sugarizer : Run 'npm install --allow-root --unsafe-perm=true' to create /opt/iiab/sugarizer-server/node_modules (CAN TAKE ~5 MINUTES)] ***
2019-06-23 21:24:35,409 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:24:35,672 p=11418 u=root |  TASK [sugarizer : For http://box/sugarizer -- run 'npm install --allow-root --unsafe-perm=true path-prefix-proxy' to create /opt/iiab/sugarizer-server/node_modules/path-prefix-proxy] ***
2019-06-23 21:25:20,611 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:20,848 p=11418 u=root |  TASK [sugarizer : Install from templates: sugarizer.service (systemd), sugarizer.conf (Apache)] ***
2019-06-23 21:25:22,306 p=11418 u=root |  changed: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/sugarizer.service', u'src': u'sugarizer.service'})
2019-06-23 21:25:23,687 p=11418 u=root |  changed: [127.0.0.1] => (item={u'dest': u'/etc/apache2/sites-available/sugarizer.conf', u'src': u'sugarizer.conf.j2'})
2019-06-23 21:25:23,941 p=11418 u=root |  TASK [sugarizer : Set Sugarizer port to 8089 in /opt/iiab/sugarizer-server/env/sugarizer.ini] ***
2019-06-23 21:25:24,658 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:24,910 p=11418 u=root |  TASK [sugarizer : Set MongoDB port to 27018 in /opt/iiab/sugarizer-server/env/sugarizer.ini] ***
2019-06-23 21:25:25,667 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:25,906 p=11418 u=root |  TASK [sugarizer : For http://box/sugarizer -- add pathPrefix lines in /opt/iiab/sugarizer-server/sugarizer.js] ***
2019-06-23 21:25:26,617 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:26,858 p=11418 u=root |  TASK [sugarizer : Create symlink sugarizer.conf from sites-enabled to sites-available, for short URLs http://box/sugar & http://box/sugarizer (if sugarizer_enabled)] ***
2019-06-23 21:25:27,700 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:27,940 p=11418 u=root |  TASK [sugarizer : Remove symlink /etc/apache2/sites-enabled/sugarizer.conf (if not sugarizer_enabled)] ***
2019-06-23 21:25:28,021 p=11418 u=root |  skipping: [127.0.0.1]
2019-06-23 21:25:28,261 p=11418 u=root |  TASK [sugarizer : Enable & Restart 'sugarizer' systemd service (if sugarizer_enabled)] ***
2019-06-23 21:25:32,464 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:32,714 p=11418 u=root |  TASK [sugarizer : Disable & Stop 'sugarizer' systemd service (if not sugarizer_enabled)] ***
2019-06-23 21:25:32,802 p=11418 u=root |  skipping: [127.0.0.1]
2019-06-23 21:25:33,069 p=11418 u=root |  TASK [sugarizer : Restart Apache service (apache2) to enable/disable http://box/sugarizer (not just http://box:8089)] ***
2019-06-23 21:25:49,334 p=11418 u=root |  changed: [127.0.0.1]
2019-06-23 21:25:49,589 p=11418 u=root |  TASK [sugarizer : Add 'sugarizer' variable values to /etc/iiab/iiab.ini] *******
```

llaske, 2019-06-24:
I would like to see the log of `npm install` because the error means that one of package installed by `npm install` - specifically [express-validator](https://github.com/llaske/sugarizer-server/blob/master/package.json#L16) - is not correctly install.

holta, 2019-06-24:
> I would like to see the log of `npm install` because the error means that one of package installed by `npm install` - specifically [express-validator](https://github.com/llaske/sugarizer-server/blob/master/package.json#L16) - is not correctly install.

Can you help tell me where I find that log file, since `npm install --allow-root --unsafe-perm=true` ran and completed?

llaske, 2019-06-24:
A first step could be to ensure that you've got a `node_modules/express-validator/` directory in the Sugarizer Server directory since it's where node.js will try to find the module.

holta, 2019-06-24:
> A first step could be to ensure that you've got a `node_modules/express-validator/` directory in the Sugarizer Server directory since it's where node.js will try to find the module.

```
root@box:/opt/iiab/sugarizer-server-1.1.0# ls -l node_modules/express-validator/
total 44
drwxr-xr-x 2 pi pi 4096 Jun 23 21:21 check
-rw-r--r-- 1 pi pi 1490 Oct 26  1985 CONTRIBUTING.md
drwxr-xr-x 2 pi pi 4096 Jun 23 21:21 declarations
drwxr-xr-x 2 pi pi 4096 Jun 23 21:22 docs
drwxr-xr-x 2 pi pi 4096 Jun 23 21:22 filter
-rw-r--r-- 1 pi pi 1100 Oct 26  1985 LICENSE
-rw-r--r-- 1 pi pi 2881 Jun 23 21:23 package.json
-rw-r--r-- 1 pi pi 1575 Oct 26  1985 README.md
drwxr-xr-x 5 pi pi 4096 Jun 23 21:22 src
-rw-r--r-- 1 pi pi  857 Oct 26  1985 tst.js
-rw-r--r-- 1 pi pi 2087 Oct 26  1985 UPGRADE_GUIDE.md
```

holta, 2019-06-24:
root@box:/opt/iiab/sugarizer-server-1.1.0# ls -ld node_modules/express-validator/
drwxr-xr-x 8 root root 4096 Jun 23 21:23 node_modules/express-validator/

holta, 2019-06-24:
> Curious that the parent directory is owned by root:root, whereas its content [child files/directories] are all owned by pi:pi ?

FWIW this is Not true on machines that were installed earlier (child files/directories are owned by root there, i.e. on these earlier machines where Sugarizer works...)

holta, 2019-06-24:
@m-anish also confirms below that ownership of stuff within `node_modules/express-validator/` is not owned by root:root on his fresh install on Ubuntu 18.04 &mdash; whereas this stuff was owned by root:root in prior installations: (seems to be the same pattern as Raspbian)
```
root@box:/etc/systemd/system# ls -l /opt/iiab/sugarizer-server-1.1.0/node_modules/express-validator/
total 44
drwxr-xr-x 2 anish anish 4096 Jun 23 21:46 check
-rw-r--r-- 1 anish anish 1490 Oct 26  1985 CONTRIBUTING.md
drwxr-xr-x 2 anish anish 4096 Jun 23 21:46 declarations
drwxr-xr-x 2 anish anish 4096 Jun 23 21:46 docs
drwxr-xr-x 2 anish anish 4096 Jun 23 21:46 filter
-rw-r--r-- 1 anish anish 1100 Oct 26  1985 LICENSE
-rw-r--r-- 1 anish anish 2881 Jun 23 21:46 package.json
-rw-r--r-- 1 anish anish 1575 Oct 26  1985 README.md
drwxr-xr-x 5 anish anish 4096 Jun 23 21:46 src
-rw-r--r-- 1 anish anish  857 Oct 26  1985 tst.js
-rw-r--r-- 1 anish anish 2087 Oct 26  1985 UPGRADE_GUIDE.md
```

llaske, 2019-06-24:
Hmmm. I'm not familiar with options `--allow-root --unsafe-perm=true`. Do you really need it? Could you try to relaunch on a clean install (node_modules) without these options?

holta, 2019-06-24:
> Hmmm. I'm not familiar with options `--allow-root --unsafe-perm=true`. Do you really need it?

Sugarizer would not install without those options, during our extensive testing in 2018 (and every early 2019?)

> Could you try to relaunch on a clean install (node_modules) without these options?

@m-anish can you try?

holta, 2019-06-25:
@llaske here is why `npm install --allow-root --unsafe-perm=true` has always been necessary over the past year &mdash; in short this is because all of Internet-in-a-Box is installed using `sudo` :

https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml#L83-L111

But if you can help us solve this in another way, that'd be great too!

holta, 2019-06-25:
I've confirmed npm was suddenly in recent weeks upgraded to 6.9.0 on both OS's (Raspbian & Ubuntu 18.04) as a result of both running Node.js 10.16.0, as seen here:

https://nodejs.org/en/download/releases/

Theory: it's quite likely that 6.9.0 implements `--allow-root --unsafe-perm=true` a bit differently than prior versions of npm.

holta, 2019-06-25:
As an experiment, I removed `--allow-root` from both lines here:
```
https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml#L114
https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml#L133
```
Results:
- Contents of `/opt/iiab/sugarizer-server-1.1.0/node_modules/validator/` are now owned by root:root (as apparently is desired) which looks much more promising, but...
- `systemctl status sugarizer` still shows this service failing, within a couple short seconds, every time...

holta, 2019-06-25:
`npm install` alone (with both flags `--allow-root --unsafe-perm=true` removed, from both stanzas) also failed to improve the situation.

This is the same exact result ([as described above](https://github.com/iiab/iiab/issues/1771#issuecomment-505286365)) when just `--allow-root` was removed from both stanzas.

holta, 2019-06-25:
This is presumably similar to "security" issues like https://github.com/iiab/iiab/issues/1714 when something runs fine & works perfectly as root, but annoyingly fails completely when run as sudo :/

#1714 "MySQL database password-changing failure during fresh install of IIAB on Ubuntu 19.04 (& 18.04.2 ?)"

holta, 2019-06-27:
Regardless whether running `npm install --unsafe-perm=true` or `npm install --allow-root` as part of both stanzas in https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml#L114-L136 (on Ubuntu 18.04), Sugarizer does not work.  FYI these tests were done exhaustively as part of a new install of IIAB &mdash; just like the original `npm install --allow-root --unsafe-perm=true` &mdash; and just like the several other above experiments on Raspbian.

This is not looking promising, with npm 6.9.0 having change the way one or both of these flag(s) work :/

If others have suggestions, that would be helpful, as IIAB 7.0 needs to be released very shortly in coming days.

holta, 2019-06-27:
@llaske we cannot get Sugarizer to work at all on new OS's (like Ubuntu 18.04.2 & Raspbian Buster) that bring npm 6.9.0 as part of node 10.16.0

Can you?

Do you want live access to this Ubuntu Server 18.04.2 VM via ssh to try?  I can send you credentials privately if so.

Even when running the following at the command line (not just through `sudo`, as IIAB must do via Ansible) it turns out it still fails as follows:

```
root@box:/opt/iiab/sugarizer-server-1.1.0# ls -ld node_modules
ls: cannot access 'node_modules': No such file or directory

root@box:/opt/iiab/sugarizer-server-1.1.0# npm install

> websocket@1.0.28 install /opt/iiab/sugarizer-server-1.1.0/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/opt/iiab/sugarizer-server-1.1.0/node_modules/websocket/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/opt/iiab/sugarizer-server-1.1.0/node_modules/websocket/build'

> nodemon@1.19.1 postinstall /opt/iiab/sugarizer-server-1.1.0/node_modules/nodemon
> node bin/postinstall || exit 0

npm WARN The package nodemon is included as both a dev and production dependency.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 499 packages from 834 contributors and audited 2922 packages in 23.353s
found 0 vulnerabilities

root@box:/opt/iiab/sugarizer-server-1.1.0# npm install path-prefix-proxy
npm WARN The package nodemon is included as both a dev and production dependency.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

+ path-prefix-proxy@1.1.0
updated 1 package and audited 2922 packages in 8.711s
found 0 vulnerabilities

root@box:/opt/iiab/sugarizer-server-1.1.0# systemctl restart sugarizer

root@box:/opt/iiab/sugarizer-server-1.1.0# systemctl status sugarizer
● sugarizer.service - Node.js Sugarizer Server
   Loaded: loaded (/etc/systemd/system/sugarizer.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-06-27 09:14:10 EDT; 1s ago
 Main PID: 12180 (node)
    Tasks: 7 (limit: 2340)
   CGroup: /system.slice/sugarizer.service
           └─12180 /usr/bin/node /opt/iiab/sugarizer-server/sugarizer.js

Jun 27 09:14:10 box.lan systemd[1]: Started Node.js Sugarizer Server.

root@box:/opt/iiab/sugarizer-server-1.1.0# systemctl status sugarizer
● sugarizer.service - Node.js Sugarizer Server
   Loaded: loaded (/etc/systemd/system/sugarizer.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-06-27 09:14:11 EDT; 2s ago
  Process: 12180 ExecStart=/usr/bin/node /opt/iiab/sugarizer-server/sugarizer.js (code=exited, status=255)
 Main PID: 12180 (code=exited, status=255)

Jun 27 09:14:11 box.lan systemd[1]: sugarizer.service: Main process exited, code=exited, status=255/n/a
Jun 27 09:14:11 box.lan systemd[1]: sugarizer.service: Failed with result 'exit-code'.

root@box:/opt/iiab/sugarizer-server-1.1.0# ls -l node_modules/validator/
total 172
-rw-r--r-- 1 root root 22332 Oct 26  1985 CHANGELOG.md
-rw-r--r-- 1 root root  8535 Oct 26  1985 index.js
drwxr-xr-x 3 root root  4096 Jun 27 09:11 lib
-rw-r--r-- 1 root root  1077 Oct 26  1985 LICENSE
-rw-r--r-- 1 root root  2754 Jun 27 09:11 package.json
-rw-r--r-- 1 root root 24133 Oct 26  1985 README.md
-rw-r--r-- 1 root root 62971 Oct 26  1985 validator.js
-rw-r--r-- 1 root root 33680 Oct 26  1985 validator.min.js
```

llaske, 2019-06-27:
@holta yes please send me credentials privately, I will have a look on it.

llaske, 2019-06-27:
The issue is related to permission on the `node_modules` directory. `npm install` fail because it can't update the content of the directory.

```
iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$ npm install --allow-root --unsafe-perm=true
npm WARN checkPermissions Missing write access to /opt/iiab/sugarizer-server-1.1.0/node_modules
npm WARN The package nodemon is included as both a dev and production dependency.

npm ERR! path /opt/iiab/sugarizer-server-1.1.0/node_modules
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/opt/iiab/sugarizer-server-1.1.0/node_modules'
npm ERR!  { [Error: EACCES: permission denied, access '/opt/iiab/sugarizer-server-1.1.0/node_modules']
npm ERR!   stack:
npm ERR!    'Error: EACCES: permission denied, access \'/opt/iiab/sugarizer-server-1.1.0/node_modules\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/opt/iiab/sugarizer-server-1.1.0/node_modules' }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/iiab-admin/.npm/_logs/2019-06-27T19_25_29_276Z-debug.log
iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$
```

I've solved the issue with a ` sudo rm -rf node_modules` but then there is another issue related this time to the `sugarizer-server` directory itself:

```
iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$ npm install --allow-root --unsafe-perm=true
npm WARN checkPermissions Missing write access to /opt/iiab/sugarizer-server-1.1.0
npm WARN The package nodemon is included as both a dev and production dependency.

npm ERR! path /opt/iiab/sugarizer-server-1.1.0
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/opt/iiab/sugarizer-server-1.1.0'
npm ERR!  { [Error: EACCES: permission denied, access '/opt/iiab/sugarizer-server-1.1.0']
npm ERR!   stack:
npm ERR!    'Error: EACCES: permission denied, access \'/opt/iiab/sugarizer-server-1.1.0\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/opt/iiab/sugarizer-server-1.1.0' }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/iiab-admin/.npm/_logs/2019-06-27T19_29_11_677Z-debug.log
iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$ 
```

So it look like the `iiab-admin` account has no right on the directory: only `root` can update the directory. To install and run Sugarizer you must a write access on the directory.

llaske, 2019-06-27:
It works when running `npm install` with `sudo`:

```
iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$ sudo npm install --allow-root --unsafe-perm=true

> websocket@1.0.28 install /opt/iiab/sugarizer-server-1.1.0/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

make: Entering directory '/opt/iiab/sugarizer-server-1.1.0/node_modules/websocket/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/opt/iiab/sugarizer-server-1.1.0/node_modules/websocket/build'

> nodemon@1.19.1 postinstall /opt/iiab/sugarizer-server-1.1.0/node_modules/nodemon
> node bin/postinstall || exit 0

npm WARN The package nodemon is included as both a dev and production dependency.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 499 packages from 834 contributors and audited 2922 packages in 39.892s
found 0 vulnerabilities

iiab-admin@box:/opt/iiab/sugarizer-server-1.1.0$ 
```

llaske, 2019-06-27:
Finally reproduced the issue on my side.

The issue is related to Sugarizer Server version 1.1.0.

A work around is explain here: https://github.com/llaske/sugarizer-server/issues/187

holta, 2019-06-30:
@anish & All: sugarizer-server v1.1.1 is now part of IIAB 7.0 pre-releases (i.e. master branch). per PR #1777.

Please if you can test this on different versions of Ubuntu, Raspbian Buster, etc &mdash; ideally with a fresh install of IIAB (from http://download.iiab.io) if possible!

holta, 2019-06-30:
Re-opening this issue just as a tracking tkt, while folks help to beef up IIAB 7.0's testing in coming days!

anish, 2019-06-30:
@holta I think you meant @m-anish, not me

holta, 2019-06-30:
> @holta I think you meant @m-anish, not me

Apologies!  And thank you for correcting my error.

holta, 2019-07-01:
@llaske and his team's fix is confirmed to work as part of a fresh install of BIG-sized IIAB 7.0 pre-release on Ubuntu Server 18.04 @ http://10.8.0.30/sugarizer

holta, 2019-07-01:
Also smoke-tested on Raspbian Buster With Desktop on RPi 3 B+, looks great!

closed 2019-07-01 by holta

m-anish, 2019-07-01:
Shall test!

PR #1777 (2019-06-30): sugarizer-server-1.1.0 -> sugarizer-server-1.1.1
Fixes #1771

Thanks to @llaske & All

The hotfix (llaske/sugarizer-server#187) is expected to be published as part of v1.1.1 in coming days here: https://github.com/llaske/sugarizer-server/releases