iiab/iiab #1772: Can we make AzuraCast work on RPi?

holta, 2019-06-24:
@m-anish is there hope here?  So far `cd /opt/iiab/iiab; ./runrole azuracast` fails as follows on Raspbian Lite on RPi 3 at 10.8.0.62:

> TASK [azuracast : AzuraCast - Run the installer] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/bin/bash docker.sh install", "delta": "0:03:27.664500", "end": "2019-06-24 04:11:58.452522", "msg": "non-zero return code", "rc": 127, "start": "2019-06-24 04:08:30.788022", "stderr": "+ sh -c apt-get update -qq >/dev/null\n+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null\n+ sh -c curl -fsSL \"https://download.docker.com/linux/raspbian/gpg\" | apt-key add -qq - >/dev/null\nWarning: apt-key output should not be parsed (stdout is not a terminal)\n+ sh -c echo \"deb [arch=armhf] https://download.docker.com/linux/raspbian stretch stable\" > /etc/apt/sources.list.d/docker.list\n+ sh -c apt-get update -qq >/dev/null\n+ [ -n  ]\n+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null\n+ sh -c docker version\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100     9    0     9    0     0     32      0 --:--:-- --:--:-- --:--:--    32\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100 13258  100 13258    0     0  72958      0 --:--:-- --:--:-- --:--:-- 73248\n  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100  2586  100  2586    0     0  12119      0 --:--:-- --:--:-- --:--:-- 12084\n/usr/local/bin/docker-compose: line 1: Not: command not found\n/usr/local/bin/docker-compose: line 1: Not: command not found\n/usr/local/bin/docker-compose: line 1: Not: command not found", "stderr_lines": ["+ sh -c apt-get update -qq >/dev/null", "+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null", "+ sh -c curl -fsSL \"https://download.docker.com/linux/raspbian/gpg\" | apt-key add -qq - >/dev/null", "Warning: apt-key output should not be parsed (stdout is not a terminal)", "+ sh -c echo \"deb [arch=armhf] https://download.docker.com/linux/raspbian stretch stable\" > /etc/apt/sources.list.d/docker.list", "+ sh -c apt-get update -qq >/dev/null", "+ [ -n  ]", "+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null", "+ sh -c docker version", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100     9    0     9    0     0     32      0 --:--:-- --:--:-- --:--:--    32", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100 13258  100 13258    0     0  72958      0 --:--:-- --:--:-- --:--:-- 73248", "  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current", "                                 Dload  Upload   Total   Spent    Left  Speed", "", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0", "100  2586  100  2586    0     0  12119      0 --:--:-- --:--:-- --:--:-- 12084", "/usr/local/bin/docker-compose: line 1: Not: command not found", "/usr/local/bin/docker-compose: line 1: Not: command not found", "/usr/local/bin/docker-compose: line 1: Not: command not found"], "stdout": "Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48\nClient:\n Version:           18.09.0\n API version:       1.39\n Go version:        go1.10.4\n Git commit:        4d60db4\n Built:             Wed Nov  7 00:57:21 2018\n OS/Arch:           linux/arm\n Experimental:      false\n\nServer: Docker Engine - Community\n Engine:\n  Version:          18.09.0\n  API version:      1.39 (minimum version 1.12)\n  Go version:       go1.10.4\n  Git commit:       4d60db4\n  Built:            Wed Nov  7 00:17:57 2018\n  OS/Arch:          linux/arm\n  Experimental:     false\nIf you would like to use Docker as a non-root user, you should now consider\nadding your user to the \"docker\" group with something like:\n\n  sudo usermod -aG docker your-user\n\nRemember that you will have to log out and back in for this to take effect!\n\nWARNING: Adding a user to the \"docker\" group will grant the ability to run\n         containers which can be used to obtain root privileges on the\n         docker host.\n         Refer to https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface\n         for more information.\nDocker Compose does not appear to be installed. Install Docker Compose now? [Y/n] Creating default AzuraCast settings file...", "stdout_lines": ["Docker does not appear to be installed. Install Docker now? [Y/n] # Executing docker install script, commit: 2f4ae48", "Client:", " Version:           18.09.0", " API version:       1.39", " Go version:        go1.10.4", " Git commit:        4d60db4", " Built:             Wed Nov  7 00:57:21 2018", " OS/Arch:           linux/arm", " Experimental:      false", "", "Server: Docker Engine - Community", " Engine:", "  Version:          18.09.0", "  API version:      1.39 (minimum version 1.12)", "  Go version:       go1.10.4", "  Git commit:       4d60db4", "  Built:            Wed Nov  7 00:17:57 2018", "  OS/Arch:          linux/arm", "  Experimental:     false", "If you would like to use Docker as a non-root user, you should now consider", "adding your user to the \"docker\" group with something like:", "", "  sudo usermod -aG docker your-user", "", "Remember that you will have to log out and back in for this to take effect!", "", "WARNING: Adding a user to the \"docker\" group will grant the ability to run", "         containers which can be used to obtain root privileges on the", "         docker host.", "         Refer to https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface", "         for more information.", "Docker Compose does not appear to be installed. Install Docker Compose now? [Y/n] Creating default AzuraCast settings file..."]}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=31   changed=14   unreachable=0    failed=1    skipped=13   rescued=0    ignored=0

Refs:
AzuraCast/AzuraCast#332 Raspberry Pi 3 (Raspbian) Docker Support
AzuraCast/AzuraCast#1606 Questions about AzuraCast intergration in the internet-in-a-box project

m-anish, 2019-06-24:
Okay I tried and tried and tried. 

first, docker-compose wasn't installing. Next pip2 install docker-compose seem to have worked. After that the install completes, but when Azuracast containers run, they come up, and then go down. I suspect the rpi just doesn't have the juice to drive Azuracast. It didn't help that it was a big sized install.

holta, 2019-06-24:
Should we consider [Libretime](https://libretime.org) which is a fork of [Airtime](https://en.wikipedia.org/wiki/Airtime_(software))?  Keeping in mind:

- "[AzuraCast](https://www.azuracast.com) also does the job of not only streaming, but what is also typically done by a third party radio s/w running on a client...like mixxx"
- "AzuraCast has other features which quickly become important one you start doing serious radio things...like keep a log/recording of the last x days of content streamed...etc."

m-anish, 2019-06-24:
If anyone has the bandwidth to try it, please take it for a spin. I quite like Azuracast, and how easy it has been to incorporate, and how helpful their community has been to us so far.

holta, 2019-07-12:
Related to this week's notion that AzuraCast might run on Debian 10.0.0 Buster:

#1766 AzuraCast fails to install on BIG-sized IIAB on "Debian 10" VM

m-anish, 2019-07-13:
Work on this will happen at my end post my Zanskar visit unfortunately. I want to try out their ansible install method but only after returning. That might be more cumbersome to get working but might be easier on the hardware.

holta, 2022-08-30:
I don't know if this means AzuraCast now fully works on Raspberry Pi 4?

- https://github.com/iiab/iiab/pull/2946#issuecomment-1231778099

holta, 2022-08-31:
This would appear to now work on 64-bit Raspberry Pi OS, when there's sufficient RAM in the Raspberry Pi 4 or similar!

Instructions here: https://github.com/iiab/iiab/tree/master/roles/azuracast#readme

Thanks to:

- PR #2946

closed 2022-08-31 by holta