iiab/iiab #1775: "Mobile/Desktop" button used to allow teachers to rapidly see Content Pack details (on their own Android phones)

holta, 2019-06-26:
Many implementers & teachers really loved & depended on this functionality in 2017 and 2018, but it is now gone.

Can this be restored for IIAB 7.1 — e.g. if it's not possible for IIAB 7.0 ?

Related:
- "Implementers overwhelmed by MenuDefs — there's an urgent need to make these understandable & upstreamable w/o GitHub bureaucracy" @ http://minutes.iiab.io -> 2. a. v.
- #1723 What's the best way to restore Content Pack metadata in small font / human-readable Spanish?

holta, 2019-06-26:
@tim-moody [wrote](https://github.com/iiab/iiab/issues/1723#issuecomment-505934220):
> > "Mobile/Desktop" button
> 
> what does this mean, whatever the implementer set for Mobile and Desktop?

The best Mobile interfaces on the public web have a `Desktop` button for those who are tired of being handicapped and prevented from accessing things they need.

After interviewing several Spanish teachers in 2017 and 2018, they agreed this button should be called:

`Vista Móvil/PC`

I don't know how this should be implemented, but now that there are many more mobile users than desktop users worldwide, it's become even more important to give such teachers & implementers rapid access to the metadata details, footnotes & all extra stuff.

A reasonably large button would greatly help (the original button in the center / on top of sites like http://iiab.me/peru worked extremely well) to make sure teachers & implementers always have rapid access to the full information they regularly need.

tim-moody, 2019-06-26:
that was not my question. before the fields for mobile and desktop were hard coded to title or all. now they are configurable. I would like the toggle to be between the two configured settings, not between all and title only.

holta, 2019-06-26:
> that was not my question. before the fields for mobile and desktop were hard coded to title or all. now they are configurable. I would like the toggle to be between the two configured settings, not between all and title only.

From my perspective: "Desktop View when a Desktop" and "Desktop View when using Mobile" should show all details possible, at all times, so that folks always have what they need instead of feeling censored.

But possibly there are other use cases / scenarios I'm not familiar with...

tim-moody, 2019-06-26:
are you saying toggle between all and configured for both devices?

holta, 2019-06-26:
> are you saying toggle between all and configured for both devices?

Something like that's sounds sufficient and reasonable.

A few "Real Desktop" users might want to lighten up their page by changing to Mobile View, but I really haven't heard of anyone asking for that so far, so let's not worry about it.

holta, 2019-06-30:
@georgejhunt: @tim-moody has posted a great-sounding proposal at https://github.com/iiab/iiab-admin-console/issues/227 to get this important missing functionality back into IIAB.

Do you have time to work on this critical feature?

holta, 2019-07-17:
Ref to 8 languages that might all be included in these metadata footnotes some day: (in an increasingly more automated way?)
https://github.com/iiab/iiab-admin-console/pull/210#issuecomment-501790051

holta, 2019-09-30:
FYI IIAB 7.0 now includes a Mobile/Desktop button, but it's a bit hard to find, so here are basic instructions:

- look in the top-left corner of your IIAB home page (e.g. http://box)
- then click on the 3 dots (vertical ellipsis)
  - a white popup/dropdown menu will appear
- then click on its middle icon (which looks like 2 squares, one square on top of the other)

holta, 2019-09-30:
Documented @ http://FAQ.IIAB.IO #16:

"How do I change to Mobile view or Desktop view?"

closed 2019-09-30 by holta