iiab/iiab #1778: ./iiab-network fails on Ubuntu 16.04

holta, 2019-06-27:
@jvonau & @m-anish you help refine PR #1768 if this is indeed the same #1767 issue, presumably caused by PR #1765?

> TASK [network : Copy script to restart dnsmasq whenever br0 comes up] **********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "checksum": "50de730947aa4cccc1586ff91d9f6bfc984b5127", "msg": "Destination directory /etc/networkd-dispatcher/routable.d does not exist"}

m-anish, 2019-06-28:
Are we going the dnsmasq route on 16.04 as well? I thought back in the day, we used bind?

Perhaps we should think which DNS protocol/server we should use on 16.04 before we fix this. Thoughts?

This is easy enough to patch, but we should address the larger question.

holta, 2019-06-28:
dnsmasq doesn't work perfectly on 16.04, but it's been mainline IIAB for 6 months now, so highly unlikely to be reverted at this point?

holta, 2019-06-29:
Solved by PR #1779

closed 2019-06-29 by holta