iiab/iiab #1782: Kolibri 0.12.5 reinstall failed on Raspbian Desktop on RPi 3 B+

holta, 2019-06-29:
FYI Kolibri 0.12.5 is already installed.  I'll try `./iiab-install` and/or `./runrole kolibri` to try again.

The error is:

> TASK [kolibri : chown -R kolibri:www-data /library/kolibri] ********************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: OSError: [Errno 2] No such file or directory: '/library/kolibri/job_storage.sqlite3-wal'
fatal: [127.0.0.1]: FAILED! => {"changed": false, "module_stderr": "Traceback (most recent call last):\n  File \"/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py\", line 114, in <module>\n    _ansiballz_main()\n  File \"/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py\", line 106, in _ansiballz_main\n    invoke_module(zipped_mod, temp_path, ANSIBALLZ_PARAMS)\n  File \"/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py\", line 49, in invoke_module\n    imp.load_module('__main__', mod, module, MOD_DESC)\n  File \"/tmp/ansible_file_payload_nT1qpg/__main__.py\", line 921, in <module>\n  File \"/tmp/ansible_file_payload_nT1qpg/__main__.py\", line 907, in main\n  File \"/tmp/ansible_file_payload_nT1qpg/__main__.py\", line 629, in ensure_directory\n  File \"/tmp/ansible_file_payload_nT1qpg/__main__.py\", line 328, in recursive_set_attributes\n  File \"/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py\", line 1333, in set_fs_attributes_if_different\n  File \"/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py\", line 985, in set_owner_if_different\n  File \"/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py\", line 880, in user_and_group\nOSError: [Errno 2] No such file or directory: '/library/kolibri/job_storage.sqlite3-wal'\n", "module_stdout": "", "msg": "MODULE FAILURE\nSee stdout/stderr for the exact error", "rc": 1}
> 
> 
PLAY RECAP *********************************************************************
127.0.0.1                  : ok=354  changed=88   unreachable=0    failed=1    skipped=127  rescued=0    ignored=0

holta, 2019-06-30:
> FYI Kolibri 0.12.5 is already installed. I'll try `./iiab-install` and/or `./runrole kolibri` to try again.

All cleared up on the 2nd pass...Mysterious!

CC'ing @benjaoming @jredrejo just in case they're interested in the overall reliability of Kolibri.

benjaoming, 2019-07-01:
@holta I added a few line breaks in that traceback ;)

But it's not a Kolibri Traceback, as you can see the file might belong to Kolibri, but the issue that is raised comes from some behavior in an Ansible module. What is it doing and why? The file `<KOLIBRI_HOME>/job_storage.sqlite3-wal` is auto-generated by SQLite, and as such may also disappear. @jredrejo can maybe add a few more words here.

Do you have an ansible rule that sets permissions of these files? And did it set them AFTER Kolibri was activated? Because in this case, a running instance of Kolibri has the right to change the files in its data directory and not have external processes interfering :)

```
OSError: [Errno 2] No such file or directory: '/library/kolibri/job_storage.sqlite3-wal'
Traceback (most recent call last):
 File "/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py", line 114, in 
 _ansiballz_main()
 File "/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py", line 106, in _ansiballz_main
 invoke_module(zipped_mod, temp_path, ANSIBALLZ_PARAMS)
 File "/root/.ansible/tmp/ansible-tmp-1561852310.47-8796729553891/AnsiballZ_file.py", line 49, in invoke_module
 imp.load_module('main', mod, module, MOD_DESC)
 File "/tmp/ansible_file_payload_nT1qpg/main.py", line 921, in 
 File "/tmp/ansible_file_payload_nT1qpg/main.py", line 907, in main
 File "/tmp/ansible_file_payload_nT1qpg/main.py", line 629, in ensure_directory
 File "/tmp/ansible_file_payload_nT1qpg/main.py", line 328, in recursive_set_attributes
 File "/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py", line 1333, in set_fs_attributes_if_different
 File "/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py", line 985, in set_owner_if_different
 File "/tmp/ansible_file_payload_nT1qpg/ansible_file_payload.zip/ansible/module_utils/basic.py", line 880, in user_and_group
OSError: [Errno 2] No such file or directory: '/library/kolibri/job_storage.sqlite3-wal'
```

holta, 2019-07-01:
> Do you have an ansible rule that sets permissions of these files? And did it set them AFTER Kolibri was activated?  Because in this case, a running instance of Kolibri has the right to change the files in its data directory and not have external processes interfering :)

Thanks @benjaoming and remember that this is a reinstall.

Can you clarify if the command itself (`chown kolibri:www-data /library/kolibri`) is not correct in any way?

What are Kolibri's prereqs, in a nutshell?

(Also if the above chown is indeed correct to implement Kolibri's prereqs, is it now necessary for `systemctl stop kolibri` to be run a priori?)

I did not write this Ansible playbook (@arky did).  But it appears extremely straightforward and simple as you can see here:
https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L60-L65

benjaoming, 2019-07-01:
> Thanks @benjaoming and remember that this is a reinstall.

Yes, exactly that... so Kolibri could be running already during the execution of the playbook?

> Can you clarify if the command itself 

I can't really, since I don't know of the internals of this setup. But the process owner of the `kolibri` process/system service has to have read/write access to the files. If the process owner is indeed `kolibri`, then this `chown` operation is fine.

> I did not write this Ansible playbook (@arky did).

I think you should ensure that Kolibri isn't running when you are doing reinstalls. This could perhaps be added in the playbook. You can check if `kolibri` is in the path and always stop it before running operations that affect its runtime environment. Chown on the data directory is an example of such; Running migrations, too.

holta, 2019-07-01:
@arky can you consider revising your https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L60-L65 in context with @benjaoming's 2 recommendations below?

In short, `chown kolibri:www-data /library/kolibri` as you've implemented it appears fine, but kolibri's systemd service needs to be stopped a priori &mdash; and likewise it should be stopped before doing "Create Kolibri default facility name, admin account and language" etc:

> the process owner of the `kolibri` process/system service has to have read/write access to the files. If the process owner is indeed `kolibri`, then this `chown` operation is fine.

> I think you should ensure that Kolibri isn't running when you are doing reinstalls (running the playbook in general). This could perhaps be added in the playbook. You can check if `kolibri` is in the path and always stop it before running operations that affect its runtime environment. Chown on the data directory is an example of such; Running migrations, too.

holta, 2019-07-05:
I'd like to work on this with someone else over the wkd if possible?

@arky can you call or advise?

holta, 2019-07-08:
Solved by PR #1828

closed 2019-07-08 by holta