iiab/iiab #1783: San Jose map pack does show additional detail (when zooming in)

holta, 2019-06-30:
@georgejhunt: @avni observed this problem occurs on both machines:

- http://10.8.0.70/osm-vector-maps/en-osm-omt_san_jose_2017-07-03_v0.23/ 2GB RAM RPi 4 Raspbian Buster With Desktop

- http://10.8.0.74/osm-vector-maps/en-osm-omt_san_jose_2017-07-03_v0.23/ 2GB RAM RPi 4 Raspbian Buster Lite

Can you help us figure this out?

holta, 2019-06-30:
@georgejhunt do you know if this problem will affect the other 10 Map Packs as well?

avni, 2019-06-30:
I downloaded the Central America problem and do not have the same problem! I can zoom in just fine on that map.

avni, 2019-06-30:
See attached images showing the difference.

San Jose: 
[image]

Central America: 
[image]

holta, 2019-07-01:
@georgejhunt if this is not possible to fix in the coming week, consider removing "San Jose" from the live catalog?

(As it does not serve much practical purpose, aside from testing.)

georgejhunt, 2019-07-03:
I tried to remove san_jose from regions.json. I cannot do it via my private
ssh key directly. And I cannot copy it from my home directory, using the
password that worked on the older instance.

holta, 2019-07-03:
> I tried to remove san_jose from regions.json.

I've fixed the group write bit was set wrong on the new server.

Please try again, thanks!

georgejhunt, 2019-07-04:
I was able to update via shared key, but I'm not able to change my
password, or to copy a file from home to /var/www/d.iiab.io via sudo,
because what I remember and what it remembers,
 for my password, are different.

Can you just set it for me using root, and then I'll be able to manage from
here.

holta, 2019-07-04:
My understanding is that San Jose's Map Pack has been removed, resolving this issue.

While a tiny Map Pack is really great for testing, and might once again be appropriate at some point in time in the future, removing this appears to be the most expeditious solution for now, thanks George for making that happen.

Separately, @georgejhunt & I are discussing fileserver security practices/choices around d.iiab.io

closed 2019-07-04 by holta

holta, 2019-07-04:
Confirmed, works great during testing per https://github.com/iiab/iiab/wiki/IIAB-Maps

Thanks @avni & @georgejhunt for getting us closer to [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) final release-