iiab/iiab #1784: UX: Captive Portal: The title/text of the page is not centered on the screen

avni, 2019-06-30:
### Expected behavior and actual behavior.
The title/text of the page is not centered on the screen. See attached image. 

### Steps to reproduce the problem.
Connect to IIAB wifi. View captive portal popup

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169

[image]

tim-moody, 2019-06-30:
In most cases captive portal should go directly to the home page rather than interspersing a stale version of that page, which must be maintained along side the home page.

holta, 2019-06-30:
Does this only happen an Mac computers?

@georgejhunt @avni please clarify if you can, such that others can solve this, with a precise understanding of where this arises.

tim-moody, 2019-06-30:
it will happen for any device routed to this intermediate page. there are two solutions: change the page; don't route to it. the issue is that some devices have a dummied down browser for captive portal and the intermediate page is an attempt to get to the real browser. Not necessary for windows (which goes direct to home page) or for the latest android (which unfortunately is not yet captured)

holta, 2019-12-28:
@georgejhunt do you happen to know if this is still an issue 6 months later?

avni, 2020-01-04:
Just tested on the latest IIAB 7.1 pre-release on iOS 13.3, Android 8.1, and MacOS 10.2.6. 

1. Android 8.1 - improved, but still cannot see or bookmark the full URL. Connect to RPI's Wifi. Immediately the user gets a notification at the top of the screen. In the notification window, click "Sign in to Wi-Fi network <SSID name>". A browser-like "sign-in to network" app immediately appears with the expected home page. If you then select the 3 dots in the upper right, you get a dropdown. From the dropdown, select "Use this network as is". The "sign-in to network" app and IIAB content page closes.  If user disconnects from network and reconnects, user gets the same experience. 
2. iOS 13.3 - First time connecting, captive portal shows up as expected. User can redirect to open IIAB in Safari. If user disconnects and reconnects, the captive portal does not show up. 
3. MacOs 10.2.6 - Captive portal does not show up at all when connecting to network.

avni, 2020-01-05:
Update: tested again on MacOS X 10.2.6 today and if I forget the Wifi SSID, Captive Portal does show up! the main body text is now centered! It would be great if the header text could be centered as well or left-aligned and closer to the icon.