iiab/iiab #1785: Bug: Admin Console: The text on the page implies that I can only select one region to install, while the section where I select the regions are checkboxes, allowing me to select multiple regions. I would expect them to be radio buttons.

avni, 2019-06-30:
### Expected behavior and actual behavior.
Bug: Admin Console: The text on the page implies that I can only select one region to install, while the section where I select the regions are checkboxes, allowing me to select multiple regions. I would expect them to be radio buttons.

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Install Content", Select "Get Map Region", View page.

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169

tim-moody, 2019-06-30:
I thought Adam removed the text, which was never the case.

tim-moody, 2019-06-30:
please file admin console bugs against iiab-admin-console

holta, 2019-06-30:
@georgejhunt please clarify your intentions as to what's best here going forward.

@avni would like radio buttons to make the user experience much more clear.

But you may have other plans, so just let us know, Thanks!

tim-moody, 2019-06-30:
radio buttons are typically used to select on of set of options. further, the look and feel is that of zims and oer2go, which I am not prepared to change

tim-moody, 2019-06-30:
we have discussed changes to this design, but not for release 7. It should work more like zims, where color coding shows state of install and installed are checked, but can't be unchecked.

georgejhunt, 2019-06-30:
It was never my intention to limit the number of regions that could be installed on IIAB to one. That was a comment that Adam put into the code without discussing it with me. 

When we discussed  the issue in at a recent call, where I asked him about the reason for limiting downloads to one, I did not understand his answer, and we left the subject without agreement.

For 7.1, it is my intention to separate the code from the data. There will be a single viewer for all maps. During the download process any new map data will be merged with one of the two sqlite databases (osm or satellite) -- hopefully with a tag added so that each addition can later be deleted separately (though I'm not sure that deletion is important enough to justify the complexity).

holta, 2019-06-30:
> It was never my intention to limit the number of regions that could be installed on IIAB to one.

Certainly there is strong disagreement here.  Many implementers and most attendees at https://linuxfestnorthwest.org/conferences/2019 who discussed this in late April, found the above to be a very confusing UX, that's not really helping schools.

If we cannot solve this fundamental disagreement for IIAB 7.0, we'll all do our best to work on our community listening skills for IIAB 7.1

Related:

#1726 It would be easier if map content packs did not bundle both code and data [OSM]

holta, 2019-06-30:
We are helping nobody when we merge changes in direct opposition to IIAB implementer community discussions:

https://github.com/iiab/iiab-admin-console/pull/231

In any case, a group of IIAB implementers and volunteers went out of their way to meet in Washington State in April of 2019 at their own expense, and their input (that more explanation is needed by new implementers, in context within a confusing UX) is being directly ignored.

avni, 2020-09-08:
These are now radio buttons in 7.2 pre-release: http://192.168.0.172/osm-vector-maps/installer/

closed 2020-09-08 by avni