iiab/iiab #1788: Feature Request: Admin Console: One alert when deleting multiple jobs

avni, 2019-06-30:
### Expected behavior and actual behavior.
If I select multiple jobs to cancel, I would like one alert that summarizes  the cancelation success/failure of each job. Currently a new alert popups up for each job.

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Utilities", Select "Display Job Status"

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169

holta, 2019-06-30:
Related:

#1786 Feature Request: Admin Console: Grouping related jobs
#1787 Feature: Admin Console: Make it easier to delete multiple jobs

avni, 2020-01-04:
Just tested on the latest IIAB 7.1 pre-release. This now works.

closed 2020-01-04 by avni