iiab/iiab #1789: Bug: Admin Console: Region stays checked after installation

avni, 2019-06-30:
### Expected behavior and actual behavior.
Installed map region does not become unchecked after clicking install button

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Install Content", Select "Get Map Region", Select one region, Click "Install Selected region", When the page refreshes the region selected is still checked.

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169

holta, 2019-06-30:
Somewhat related:

#1785 Bug: Admin Console: Selecting one vs multiple regions - checkboxes/radio buttons [OSM]

holta, 2019-07-01:
Resolution may depend on iiab/iiab-admin-console#229 ("Admin Console reports osm region as installed when it isn't") whose tracking tkt is #1793.

tim-moody, 2019-07-01:
it should stay checked. that's what zims do. later there will be colors to tell you it is installed, but it doesn't install twice.

holta, 2019-07-01:
@avni we could use your independent set eyes to make sure the next design iteration is much less confusing.

Am moving this to the IIAB 7.1 Milestone in the meantime, so we circle back to take UAT (user acceptance testing) seriously once again at that point.

avni, 2019-07-02:
OK, happy to review again then.

avni, 2020-01-04:
Just tested on the latest IIAB 7.1 pre-release. This is still an issue and in my opinion, a confusing admin UX experience.

avni, 2020-09-08:
The 7.2 pre-release installer: http://192.168.0.172/osm-vector-maps/installer/ now shows already installed map content packs highlighted in green text. 

1. The end-user can still select this map content pack for reinstall, is this okay? 
2. Accessibility: Color blind users may not be able to decipher the green from the black text - text-based info to let the user know the content pack is installed will be helpful.