iiab/iiab #1791: Bug: Admin Console: Estimated Space Required incorrect after first region is installed

avni, 2019-06-30:
### Expected behavior and actual behavior.
Estimated space required does not update after first region is installed.

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Install Content", Select "Get Map Region", Select one region, Click "Install Selected region", Uncheck the region that was first selected (see row #16), select another regision, "Estimated space required" does not update

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169

holta, 2019-06-30:
Somewhat related:

#1785 Bug: Admin Console: Selecting one vs multiple regions - checkboxes/radio buttons [OSM]
#1789 Bug: Admin Console: Region stays checked after installation [OSM]

tim-moody, 2020-02-05:
there is a bug in admin_console.js sumMapWip() that causes map space estimate to fail.

holta, 2020-02-13:
@tim-moody summarizes this as 2 bugs:

- wrong variable name he just fixed
- OSM (maps) don't report size / how much space is required as other content types do report

Conclusion: when new [OSM] maps appear later this year, Tim will get back to solving this properly (very complicated, but the short version is that the back-end reports these size estimates... ongoing installs / queue / etc need this!)

holta, 2020-05-06:
> there is a bug in admin_console.js sumMapWip() that causes map space estimate to fail.

@tim-moody is this fixed now, do you know?

tim-moody, 2020-05-06:
>Conclusion: when new [OSM] maps appear later this year, Tim will get back to solving this properly (very complicated, but the short version is that the back-end reports these size estimates... ongoing installs / queue / etc need this!)

No plans to address this until all changes to maps are complete. Users should download maps first, so that space has been occupied and will be correct for other content.

holta, 2020-09-03:
@avni can you confirm this is closeable, with IIAB's new OSM Map Pack install process?

Outlined @ https://github.com/iiab/iiab/wiki/IIAB-Maps > "For 7.2 and later Maps, How do I Select and Install Tiles for my region?"

avni, 2020-09-08:
This issue is not relevant anymore on http://192.168.0.172/osm-vector-maps/installer/ as the estimated space required text is no longer there (and does not need to be given the UI has moved to having a radio button, which limits the user to one selection).

closed 2020-09-08 by avni