iiab/iiab #1793: Tracking Tkt for "Admin Concole reports osm region as installed when it isn't" iiab/iiab-admin-console#229

holta, 2019-06-30:
Many thanks @tim-moody, @avni & @georgejhunt:

https://github.com/iiab/iiab-admin-console/issues/229

PS I'm still hopeful that IIAB 7.0 might be released next wkd similar to Debian 10 Buster itself, around July 6th or 7th &mdash; but we should also take stock later this week, to (re)assess whether a bit of a delay is in fact wise and appropriate &mdash; as our community UX's (at every level, from docs/vids to graphical UX's) keep getting stronger, in service of critical global needs.

holta, 2019-07-06:
@georgejhunt is away for a few days but FYI this remains a very high-priority bug that needs to be fixed in the coming week:

1) Another report arrived from someone following the official instructions https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack to upgrade `en-osm-omt_africa_2017-07-03_v0.2`

   I've confirmed on their machine, that the process fails at Step 5 (http://box.lan/admin > `Install Content` > `Get Map Region`) with a popup announcing:

   ```
   [actual IP address] says
   Selected Map Region is already installed.
   ```

2) No jobs begin, i.e. nothing appears in http://box.lan > `Utilities` > `Display Job Status`

   i.e. this is more than just a confusing alert, as seems to be suggested by #1800 ~= PR https://github.com/iiab/iiab-admin-console/pull/232

tim-moody, 2019-07-08:
I have recommended before and continue to recommend going back to the previous code.

georgejhunt, 2019-07-08:
I think we need to move forwards on something.  I started working on this
problem because the previous code did not work for me. But I'm happy to let
Tim revert, and let him deal with the complaints. I'm done.

holta, 2019-07-09:
Hey just a reminder there is no mad rush to solve this by any particular deadline.

IIAB 7.0 can be released anytime if necessary, preferably in July?

As hard as we each and all try, we're not going to solve all 27 of these residual UX bugs/issues before release: https://github.com/iiab/iiab/milestone/5

So be it if we happen to release around mid-month coincident with @m-anish's departure to North India, sure that's a bonus why not, if it happens.

_But Conversely_ if it turns out some difficult UX issues (or whatever) need to be sorted out, so be it if this already incredibly impressive and ambitious release has to happen somewhat later!

holta, 2020-01-04:
@georgejhunt is this (iiab/iiab-admin-console#229) still an issue?

(Or can I close it now?)

georgejhunt, 2020-01-04:
I think this issue is moot. Now we will let the user add any additional
regions to the one he first installs.
Unknown at this point is how long the sqlite copy/merge will take -- for
example adding the Middle East to Africa. Based upon my experience creating
the regions it could take more than 24 hours (and I have no experience on
the rpi hardware).

I don't think we need to prevent a user from downloading the same region
twice. So any such checks can be deleted.

Close it

closed 2020-01-04 by holta