iiab/iiab #1794: Feature Request: Admin Console: Modify select lists in Manage Content Menus to display font

avni, 2019-06-30:
### Expected behavior and actual behavior.
In the "Manage Content Menus" section, it is difficult for the user to know what font they are trying to change the Mobile and Desktop fonts to with just the name. Ideally, the drop down (select list) would display the options in the font being selected. See here for a stack overflow question: https://stackoverflow.com/questions/10696004/how-to-change-font-family-of-drop-downs-list-item

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Content Menus", Open "Select Mobile Header Font" dropdown. See images.

[image]

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169
Server: RPI 4B Buster Lite
IIAB: 7.0 pre-release

tim-moody, 2019-06-30:
works for me on win client with deb 10 vm server
probably the fonts did not get installed
what's in /library/www/html/common/? should be fonts and webfonts

tim-moody, 2019-06-30:
also what browser?

avni, 2019-06-30:
Testing in Chrome Version 74.0.3729.169 (also listed in specifications section above)
See screenshot of contents of fonts and web fonts folder: 

[image]

tim-moody, 2019-06-30:
can you try iiab.me the fonts your browser is trying to load are no longer used. makes me think it is a caching issue.

avni, 2019-06-30:
Same behavior on user-facing page: http://iiab.me - I don't have admin access on this instance, so cannot test changing font and refreshing, etc.

tim-moody, 2019-06-30:
hm. and shows the 404 errors?

tim-moody, 2019-06-30:
I just changed the font with admin console. what do you see?

tim-moody, 2019-06-30:
[image]

avni, 2019-06-30:
[image]

avni, 2019-06-30:
[image]

tim-moody, 2019-06-30:
and now I changed it back

[image]

avni, 2019-06-30:
I see - so this is only supposed to change the header font? Why is this an important admin function? It is fun, but not something I necessarily see admins (TLCs) doing frequently. I guess I now even more strongly believe flipping menu properties and content item list should be done: https://github.com/iiab/iiab/issues/1796.

holta, 2020-01-04:
Suggestion: let's talk to newbie implementers in 2020 to see if the findability of drag & drop menus can be improved here?