iiab/iiab #1795: Bug: Admin Console: Font selection in Manage Content Menus does not seem to be doing anything?

avni, 2019-06-30:
### Expected behavior and actual behavior.
When selecting a new font from the "Manage Content Menus" admin function, I expect the font to change real-time. It does not. Not clear how to make the selection stick/have the desired effect.

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Content Menus", Go to "Select Desktop Header Font" dropdown, Select a new font. No fonts change.

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169
Server: RPI 4B Buster Lite
IIAB: 7.0 pre-release

holta, 2019-06-30:
Related:

#1794 "Feature Request: Admin Console: Modify select lists in Manage Content Menus to display font"

tim-moody, 2019-06-30:
Took "Manage Content Menus" Menu Properties changed Select Desktop Header Font from finger paint to freckle face. saved the menu. reloaded the menu in browser. the font changed.

did the same for mobile. font changed

avni, 2019-06-30:
Tested on Chrome and Safari in OS X and no change. Font family when I go into the inspector is: font-family: 'Open Sans', sans-serif; In safari I do see the failures that are showing in the screenshot 

[image]

tim-moody, 2019-06-30:
[image]

tim-moody, 2019-07-08:
works as designed. only header font is affected as the prompt says.

closed 2019-07-08 by tim-moody