iiab/iiab #1796: Feature Request: Admin Console: Flip Menu Properties and Content Item List

avni, 2019-06-30:
### Expected behavior and actual behavior.
As an admin user, I would prefer to have the "Content Item List" more readily available. Suggesting to flip the order of "Menu Properties" tab and the "Content Item List" and have "Content Item List" as the default. See screenshot.

### Steps to reproduce the problem.
Go to http://box.lan/admin/, Select "Content Menus", View Page.

### Specifications like the version of the IIAB, operating system version, or hardware details.
Macbook Pro 2.7 Ghz 8 GB Ram
Mac OS Sierra 10.12.6
Chrome Version 74.0.3729.169
Server: RPI 4B Buster Lite
IIAB: 7.0 pre-release

[image]

holta, 2019-07-01:
Related challenges:

#828 [screencast/videos to showcase mainline Community UX's like?] "Copy Content Packs to/from USB sticks/storage"
#1002 "Update [what was iiab-menu] with all recent Content Pack descriptions/logos"
#1601 "Which IIAB screencasts / videos are most urgently needed for implementers & teachers?"
#1708 "Understanding IIAB's Admin Console (box.lan/admin) > Content Menus (tab)"

avni, 2020-01-03:
Just tested on the latest IIAB 7.1 pre-release. This is important for a good admin UX experience, in my opinion.