iiab/iiab #1799: Who can test IIAB 7.0 on Raspberry Pi Zero W?

holta, 2019-06-30:
Guideline Instructions:

- Install the latest IIAB 7.0 pre-release (http://download.iiab.io) onto an ARMv7 RPi 3 or 3 B+ &mdash; or possibly the ARMv8 RPi 4 might work as well, even despite its architecture (??)
- Select MIN-sized, MEDIUM-sized...or BIG-sized if you dare?
- Remove the microSD card and insert into an ARMv6 RPi Zero W
- After booting it in your RPi Zero W, test a diversity IIAB Apps from http://FAQ.IIAB.IO Items 4 & 9:
  - "4 What can I do with E-books and Internet-in-a-Box?"
  - "9 What services (IIAB apps) are suggested during installation?"
- Report any glitches that you find unacceptable, _Many Thanks !_

dirkcuys, 2019-07-04:
Hey, I was in the process of doing this, but I only have a 4GB SD on hand and the MIN-sized install didn't fit.

What I can say is that the ansible playbooks seems to have completed fine while installing using a Pi 3b

holta, 2019-07-04:
> Hey, I was in the process of doing this, but I only have a 4GB SD on hand and the MIN-sized install didn't fit.
> 
> What I can say is that the ansible playbooks seems to have completed fine while installing using a Pi 3b

Many thanks @dirkcuys !

There are IIAB volunteers who would be willing to donate an microSD card, depending where you're located?

Feel free to email if this or similar helps (holt @ unleashkids . org)

dirkcuys, 2019-07-05:
@holta thanks for the offer, but I can source a SD if need be :) I'm based in Cape Town, South Africa, btw.

I did have a slightly larger USB thumb drive available, so I installed on there and it worked, except for the tune2fs bit. Below are my notes:

1. Install debian buster on a USB thumb drive
1. Enable SSH (`sudo touch /boot/ssh`)
1. Run `curl d.iiab.io/install.txt | sudo bash`
1. Update password: busterf13ndly
1. Choose edit /etc/iiab/local_vars.yml
1. Change `kalite_install` and `kalite_enabled` to `False`
1. Run `sudo iiab`
1. System rebooted after updating OS
1. Run `sudo iiab` again

> Detected RPi: Lower reserve disk (SD card) space from ~5% to 1%
> 
> tune2fs 1.44.5 (15-Dec-2018)
> tune2fs: No such file or directory while trying to open /dev/mmcblk0p2
> Couldn't find valid filesystem superblock.

10. Change mmcblk0p2 to sda2 `sudo sed -i "s/mmcblk0p2/sda2/g" /usr/sbin/iiab`
1. Run `sudo iiab` again
1. Run until it got to

> INTERNET-IN-A-BOX (IIAB) SOFTWARE INSTALL IS COMPLETE
> 
> (1A) A couple minutes after you reboot (below) try to connect any laptop to
> Wi-Fi hotspot "Internet in a Box".  If this works, verify that you can browse
> to http://box or http://box.lan or http://172.18.96.1
>
> (1B) IF THOSE 3 DON'T WORK, try http://box.local from any device connected to
> your usual network.  Or try http://localhost from your IIAB itself!
>
> (1C) IF ALL 5 ABOVE DON'T WORK, ask the person who set up the network/router
> in your building for the IP address of your IIAB, so you can browse to it
> using (something like) http://192.168.0.100
>
> (2) ADD CONTENT using http://box.lan/admin (changing "box.lan" to be as above!)
> PLEASE READ "What are the default passwords?" and "How do I customize my
> Internet-in-a-Box home page?" at http://FAQ.IIAB.IO
>
> (3) If you're installing IIAB over Wi-Fi (instead of Ethernet) remember to run
> 'iiab-hotspot-on' at the VERY END, when you're ready to ACTIVATE YOUR IIAB's
> INTERNAL WI-FI HOTSPOT.  CAUTION: this permanently kills your IIAB's Internet
> (over Wi-Fi anyway) until you later run 'iiab-hotspot-off'.  CAVEAT: these
> two commands only work with Raspberry Pi as of 2019-06-27.
>
> HIT [Enter] TO REBOOT, CONFIRMING YOU'VE "photographed" THE ABOVE PARAGRAPHS: 

13. reboot on the pi 3
1. Connect to the IIAB access point - working!
1. Kill second partition of 4GB SD card I have
1. Delete all content from first partition (/boot)
1. Copy [bootcode.bin](https://raspberrypi.stackexchange.com/questions/62965/boot-zero-w-from-usb-device) to first partition to boot from USB
1. Boot Pi Zero W 1.3 from USB!!
1. Connect to IIAB access point
1. go to http://box.lan
1. http://box.lan/kiwix/ works! Only test content
1. http://box.lan/munin/ works!
1. http://box.lan/osm-vector-maps/maplist/ works! Only world map though
1. http://box.lan/usb/ shows actual USB content
1. http://box.lan/awstats/awstats.pl works. Take some time to load.
1. http://box.lan/admin/ works!
1. I didn't install kalite

holta, 2019-07-06:
> 15. Kill second partition of 4GB SD card I have
> 16. Delete all content from first partition (/boot)

Question: Instead of the above 2 steps, could the microSD cars simply have been removed from the RPi Zero W?  (So long as Step 17 ["Copy bootcode.bin to first partition to boot from USB"] was completed?!)

> 23. http://box.lan/osm-vector-maps/maplist/ works! Only world map though

Presumably zooming in is limited by the fact you haven't yet downloaded one of the 10 "continents" (Map Packs) from http://box.lan/admin (username: iiab-admin) > `Install Content` > `Get Map Region` ?

More details here if nec!
https://github.com/iiab/iiab/wiki/IIAB-Maps

@dirkcuys Thanks so much for your above USB stick install recipe.  Perhaps this should be in http://FAQ.IIAB.IO ?

Others greatly appreciate knowing what works (and what doesn't!) using Raspberry Pi Zero W tricks you've uncovered...whether or not an "RPi 4 Zero W" ever materializes ;)

Ref: #1821

dirkcuys, 2019-07-07:
> Question: Instead of the above 2 steps, could the microSD cars simply have been removed from the RPi Zero W? (So long as Step 17 ["Copy bootcode.bin to first partition to boot from USB"] was completed?!)

According to [what I read](https://raspberrypi.stackexchange.com/questions/62965/boot-zero-w-from-usb-device) no, but I didn't test it, so maybe?

> Presumably zooming in is limited by the fact you haven't yet downloaded one of the 10 "continents" (Map Packs) from http://box.lan/admin (username: iiab-admin) > Install Content > Get Map Region ?
>
> More details here if nec! https://github.com/iiab/iiab/wiki/IIAB-Maps

Thanks, I assumed that much, but didn't have to space to download any of the maps

holta, 2019-07-10:
Data Point: building up a BIG-sized IIAB 7.0 pre-release microSD card in RPi 4 (using http://download.iiab.io) worked.

I simply moved the microSD card from the Raspberry Pi 4 into the Raspberry Pi Zero W.

Of course: I did not test every single IIAB App as itemized in http://FAQ.IIAB.IO ITEM 9 ("What services (IIAB apps) are suggested during installation?")

Not is anybody actually suggesting running ~30 apps/services on a 512MB RAM single-core RPi Zero W, which would be the height of foolishness (!)

But as a proof-of-concept, I was quite pleasantly astonished that this worked at all, with the Zero W booting quite reliably in about 90 seconds...

holta, 2019-07-11:
Am moving this tkt to the IIAB 7.1 Milestone.

(Raspberry Pi Zero W support is really-nice-to-have, but testing all its subtleties is not a blocker for this IIAB 7.0 release!)

holta, 2020-04-08:
@tim-moody let's raise the priority here looking for testers to do quick spot check(s) here in anticipation of IIAB 7.1's final release[*].

[*] Sometime around 2020-04-30 / 2020-05-01, i.e. ideally about 1 week after Ubuntu 20.04 LTS's own release on Thursday 2020-04-23.

holta, 2020-04-08:
Thanks to @tim-moody who is trying to test a MEDIUM-sized IIAB install in RPi Zero W here: #2287

tim-moody, 2020-04-08:
sugarizer service doesn't start
nodejs --version gives 'Illegal instruction'; signal=ILL

holta, 2020-04-08:
> sugarizer service doesn't start
> nodejs --version gives 'Illegal instruction'; signal=ILL

Seems likely this is a result of [Known Issue](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues) around CPU architecture:

> Node.js applications like Asterisk/FreePBX, Node-RED and Sugarizer won't work on Raspberry Pi Zero W (ARMv6) if you installed Node.js while on RPi 3, 3 B+ (ARMv7) or RPi 4 (ARMv8). If necessary, run `apt remove nodejs` then ([attempt!](https://nodered.org/docs/getting-started/raspberrypi)) things like `cd /opt/iiab/iiab; ./runrole nodejs to install Node.js` on the Raspberry Pi Zero W itself — before proceeding to install Asterisk/FreePBX, Node-RED and/or Sugarizer. #2082

holta, 2020-04-08:
Are there any other critical pieces that need to run on RPi Zero W?

Or is this good enough for IIAB 7.1?

tim-moody, 2020-04-08:
I typically put mediawiki on there

holta, 2020-04-08:
> I typically put mediawiki on there

Ok.  Please test that on Zero W if possible in coming days.

tim-moody, 2020-04-08:
./runrole nodejs
nodejs_enabled is False by default

holta, 2020-04-08:
> nodejs_enabled is False by default

Yes.  Variable `nodejs_enabled` is changed from False to True by apps that require that.  See for example:

https://github.com/holta/iiab/blob/iiab.ini-descriptions/roles/sugarizer/tasks/install.yml#L10-L13 (all 27 lines at the top of this file are relevant)

holta, 2020-04-08:
Oops, correct URL is: https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/install.yml#L7-L26

tim-moody, 2020-04-08:
not getting an install and not worth the effort

tim-moody, 2020-04-08:
ansible is really slow on the W

holta, 2020-04-08:
Aside/FYI: `nodejs_enabled` is a variable that does not serve a functional purpose.

holta, 2020-04-08:
> not getting an install and not worth the effort

MediaWiki definitely installs on RPi 4, so I'm presuming it runs fine when the microSD is then/later physically booted in a Zero W.

(Or do you have evidence to the contrary?)

tim-moody, 2020-04-08:
just runrolling now

jvonau, 2020-04-08:
> Seems likely this is a result of [Known Issue](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues) around CPU architecture:

Complete bogus instructions, it's something like
apt remove nodejs    maybe add apt purge in for good measure 
remove /etc/apt/sources.list.d/nodesoure.list
apt update
unpack tarfile noted in 2082

tim-moody, 2020-04-08:
it's all probably too heavy for a W anyway, so I don't think it's worth pursuing

holta, 2020-04-08:
> it's all probably too heavy for a W anyway, so I don't think it's worth pursuing

Agreed.  But we have several obsessed Zero W hackers out there.  So I've revised the tips around this [Node.js Known Issue](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues) as follows:

> Node.js applications like Asterisk/FreePBX, Node-RED and Sugarizer won't work on Raspberry Pi Zero W (ARMv6) if you installed Node.js while on RPi 3, 3 B+ (ARMv7) or RPi 4 (ARMv8). If necessary, run `apt remove nodejs` or `apt purge nodejs` then ([attempt!](https://nodered.org/docs/getting-started/raspberrypi)) things like `rm /etc/apt/sources.list.d/nodesoure.list; apt update; cd /opt/iiab/iiab; ./runrole nodejs` to install Node.js on the Raspberry Pi Zero W itself (a better approach than "./runrole nodejs" is to try installing the tar file mentioned at #2082). Finally, proceed to install Asterisk/FreePBX, Node-RED and/or Sugarizer.

tim-moody, 2020-04-08:
I think you will need to set nodejs_enabled: True in local vars.
Likely ./runrole nodejs should have --reinstall.
Or I guess your're saying to turn the listed node apps off for first run, then do node, then do any of the listed apps you want.

holta, 2020-04-08:
Another pass at documenting the [Node.js Known Issue](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues):

> Node.js applications like Asterisk/FreePBX, Node-RED and Sugarizer won't work on Raspberry Pi Zero W (ARMv6) if you installed Node.js while on RPi 3, 3 B+ (ARMv7) or RPi 4 (ARMv8). If necessary, run `apt remove nodejs` or `apt purge nodejs` then `rm /etc/apt/sources.list.d/nodesoure.list; apt update` then (attempt!) to install Node.js on the Raspberry Pi Zero W itself (a better approach than "cd /opt/iiab/iiab; ./runrole nodejs" is to try installing the tar file mentioned at #2082). Finally, proceed to install Asterisk/FreePBX, Node-RED and/or Sugarizer.

jvonau, 2020-04-08:
Wow smh... `runrole nodejs` installs nodesource, that is what breaks the W, after `apt update` it should be `apt install nodejs npm` then edit /etc/iiab/local_vars.yml adding nodejs_installed: True.
The last 2 lines could be done prior to the start of the install saving a bunch of unnecessary steps.

holta, 2020-04-08:
> Or I guess your're saying to turn the listed node apps off for first run, then do node, then do any of the listed apps you want.

That's the intended workflow, yes.

> I think you will need to set nodejs_enabled: True in local vars.
> Likely ./runrole nodejs should have --reinstall.

Good point that the --reinstall flag within things like `./runrole --reinstall nodejs` is often forgotten.

We should try hard to improve `./runrole` documention prior to release (runtime doc tips, etc) if at all possible here:

#2096 "NGINX & ./runrole need approachable docs for newbie/legacy implementers"

tim-moody, 2020-04-08:
apt install nodejs npm is tricky because in later version of nodejs, npm is included and apt can install an earlier version.

tim-moody, 2020-04-08:
/etc/apt/sources.list.d/nodesoure.list typo /etc/apt/sources.list.d/nodesource.list
in the node play version is set to 12, but on w apt installs 10.15.2 (doesn't include npm)

tim-moody, 2020-04-08:
root@box:~# apt install npm
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  gyp javascript-common libjs-inherits libjs-is-typedarray libnode-dev libssl-dev libuv1-dev node-abbrev node-ajv node-ansi node-ansi-align node-ansi-regex node-ansi-styles node-ansistyles
  node-aproba node-archy node-are-we-there-yet node-asn1 node-assert-plus node-asynckit node-aws-sign2 node-aws4 node-balanced-match node-bcrypt-pbkdf node-bluebird node-boxen
  node-brace-expansion node-builtin-modules node-builtins node-cacache node-call-limit node-camelcase node-caseless node-chalk node-chownr node-cli-boxes node-cliui node-clone node-co
  node-color-convert node-color-name node-combined-stream node-concat-map node-concat-stream node-config-chain node-console-control-strings node-copy-concurrently node-core-util-is
  node-cross-spawn node-cyclist node-dashdash node-decamelize node-decompress-response node-deep-extend node-defaults node-delayed-stream node-delegates node-detect-indent
  node-detect-newline node-duplexer3 node-duplexify node-ecc-jsbn node-editor node-encoding node-end-of-stream node-errno node-escape-string-regexp node-execa node-extend node-extsprintf
  node-find-up node-flush-write-stream node-forever-agent node-form-data node-from2 node-fs-vacuum node-fs-write-stream-atomic node-fs.realpath node-gauge node-get-caller-file
  node-get-stream node-getpass node-glob node-got node-graceful-fs node-gyp node-har-schema node-har-validator node-has-flag node-has-symbol-support-x node-has-to-string-tag-x
  node-has-unicode node-hosted-git-info node-http-signature node-iconv-lite node-iferr node-import-lazy node-imurmurhash node-inflight node-inherits node-ini node-invert-kv
  node-is-builtin-module node-is-npm node-is-object node-is-plain-obj node-is-retry-allowed node-is-stream node-is-typedarray node-isarray node-isexe node-isstream node-isurl node-jsbn
  node-json-parse-better-errors node-json-schema node-json-stable-stringify node-json-stringify-safe node-jsonify node-jsonparse node-jsonstream node-jsprim node-latest-version
  node-lazy-property node-lcid node-libnpx node-locate-path node-lockfile node-lowercase-keys node-lru-cache node-mem node-mime-types node-mimic-fn node-mimic-response node-minimatch
  node-minimist node-mississippi node-mkdirp node-move-concurrently node-mute-stream node-node-uuid node-nopt node-normalize-package-data node-npm-package-arg node-npm-run-path node-npmlog
  node-oauth-sign node-object-assign node-once node-opener node-os-locale node-osenv node-p-cancelable node-p-finally node-p-limit node-p-locate node-p-timeout node-package-json
  node-parallel-transform node-path-exists node-path-is-absolute node-path-is-inside node-performance-now node-prepend-http node-process-nextick-args node-promise-inflight node-promzard
  node-proto-list node-prr node-pump node-pumpify node-punycode node-qs node-qw node-rc node-read node-read-package-json node-readable-stream node-registry-auth-token node-registry-url
  node-request node-require-directory node-require-main-filename node-resolve-from node-retry node-rimraf node-run-queue node-safe-buffer node-semver node-semver-diff node-set-blocking
  node-sha node-shebang-command node-shebang-regex node-signal-exit node-slash node-slide node-sorted-object node-spdx-correct node-spdx-expression-parse node-spdx-license-ids node-sshpk
  node-ssri node-stream-each node-stream-iterate node-stream-shift node-string-decoder node-string-width node-strip-ansi node-strip-eof node-strip-json-comments node-supports-color
  node-tar node-term-size node-text-table node-through node-through2 node-timed-out node-tough-cookie node-tunnel-agent node-tweetnacl node-typedarray node-uid-number node-unique-filename
  node-unpipe node-url-parse-lax node-url-to-options node-util-deprecate node-uuid node-validate-npm-package-license node-validate-npm-package-name node-verror node-wcwidth.js node-which
  node-which-module node-wide-align node-widest-line node-wrap-ansi node-wrappy node-write-file-atomic node-xdg-basedir node-xtend node-y18n node-yallist node-yargs node-yargs-parser
Suggested packages:
  libssl-doc
^[[BThe following NEW packages will be installed:
  gyp javascript-common libjs-inherits libjs-is-typedarray libnode-dev libssl-dev libuv1-dev node-abbrev node-ajv node-ansi node-ansi-align node-ansi-regex node-ansi-styles node-ansistyles
  node-aproba node-archy node-are-we-there-yet node-asn1 node-assert-plus node-asynckit node-aws-sign2 node-aws4 node-balanced-match node-bcrypt-pbkdf node-bluebird node-boxen
  node-brace-expansion node-builtin-modules node-builtins node-cacache node-call-limit node-camelcase node-caseless node-chalk node-chownr node-cli-boxes node-cliui node-clone node-co
  node-color-convert node-color-name node-combined-stream node-concat-map node-concat-stream node-config-chain node-console-control-strings node-copy-concurrently node-core-util-is
  node-cross-spawn node-cyclist node-dashdash node-decamelize node-decompress-response node-deep-extend node-defaults node-delayed-stream node-delegates node-detect-indent
  node-detect-newline node-duplexer3 node-duplexify node-ecc-jsbn node-editor node-encoding node-end-of-stream node-errno node-escape-string-regexp node-execa node-extend node-extsprintf
[... cut ...]
  node-tar node-term-size node-text-table node-through node-through2 node-timed-out node-tough-cookie node-tunnel-agent node-tweetnacl node-typedarray node-uid-number node-unique-filename
  node-unpipe node-url-parse-lax node-url-to-options node-util-deprecate node-uuid node-validate-npm-package-license node-validate-npm-package-name node-verror node-wcwidth.js node-which
  node-which-module node-wide-align node-widest-line node-wrap-ansi node-wrappy node-write-file-atomic node-xdg-basedir node-xtend node-y18n node-yallist node-yargs node-yargs-parser npm
0 upgraded, 247 newly installed, 0 to remove and 0 not upgraded.

holta, 2020-04-08:
For now [IIAB 7.1 Release Notes -> Known Issues](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues) contains:

> Node.js applications like Asterisk/FreePBX, Node-RED and Sugarizer won't work on Raspberry Pi Zero W (ARMv6) if you installed Node.js while on RPi 3, 3 B+ (ARMv7) or RPi 4 (ARMv8). If necessary, run `apt remove nodejs` or `apt purge nodejs` then `rm /etc/apt/sources.list.d/nodesoure.list; apt update` then (attempt!) to install Node.js _on the Raspberry Pi Zero W itself_ (a better approach than "cd /opt/iiab/iiab; ./runrole nodejs" is to try `apt install nodejs` or try installing the tar file mentioned at #2082). You might also need `apt install npm`. Whatever versions of Node.js and npm you install, make sure `/etc/iiab/iiab_state.yml` contains the line `nodejs_installed: True` (add it if nec!) Finally, proceed to install Asterisk/FreePBX, Node-RED and/or Sugarizer.

jvonau, 2020-04-08:
Not tricky at all. Nodesource's nodejs bundles npm with it and does not offer a standalone apt package known as npm, npm is a dependency supplied by nodejs. Now out side of nodesouce, npm as a dependency is satisfied with an npm apt package that only exists in ubuntu/debian's apt world.

holta, 2020-04-08:
In other news, @tim-moody got MediaWiki 1.34.1 (http://box/wiki) to work on Zero W, so we're in really good shape overall here.

tim-moody, 2020-04-08:
npm -v 5.8.0 (not the latest)

jvonau, 2020-04-08:
yes supplied by the distro if you want some nodesouce version adapt the tar file notes from 2082.

holta, 2020-04-08:
RPi Zero W coverage for IIAB 7.1 seems good enough.

Thanks to everyone who helped make this happen over the past 9 months.

holta, 2020-04-09:
@tim-moody did you test the hotspot in Zero W?

Can you if you haven't yet?

Ref: #2265 "Raspbian blocks hostapd when IIAB microSD is built in RPi 4 and then moved to RPi 3 or 3 B+ (or Zero W?) [until you run "rfkill unblock wifi"]"

tim-moody, 2020-04-09:
I used the hotspot to access iiab from my laptop with no issues (cap portal worked too)

holta, 2020-04-09:
> I used the hotspot to access iiab from my laptop with no issues (cap portal worked too)

Great..and .you definitely did not have to run "rfkill unblock wifi" ??

(That's counterintuitive if so, given the February-in-Mexico result @ #2265.)

tim-moody, 2020-04-09:
correct

holta, 2020-04-09:
> correct. because it's lite, not desktop?

Hypothetically also possible that Raspbian has evolved to partly/fully solve the issue, during the~2 months since #2265 flared up?

holta, 2020-04-18:
Good enough for the [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) release cycle!

(Re-open as nec...)

closed 2020-04-18 by holta