iiab/iiab #1801: 'yarn | GPG' currently blocking some IIAB installs? [internetarchive]

holta, 2019-07-01:
@mitra42 do you know the recommended way to solve this?

When trying to install a BIG-sized IIAB 7.0 pre-release (http://download.iiab.io) onto Ubuntu Server 18.04 (including the `internetarchive` Ansible playbook) it failed as follows:

> TASK [yarn : Yarn | GPG] *******************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/bin/apt-key add -", "msg": "Warning: apt-key output should not be parsed (stdout is not a terminal)\ngpg: can't connect to the agent: IPC connect call failed", "rc": 2, "stderr": "Warning: apt-key output should not be parsed (stdout is not a terminal)\ngpg: can't connect to the agent: IPC connect call failed\n", "stderr_lines": ["Warning: apt-key output should not be parsed (stdout is not a terminal)", "gpg: can't connect to the agent: IPC connect call failed"], "stdout": "", "stdout_lines": []}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=534  changed=426  unreachable=0    failed=1    skipped=170  rescued=0    ignored=1

When trying my luck running `sudo iiab` a 2nd time, I get this error:

> IF YOU FACE ERROR "signatures couldn't be verified because the public key is not available" THEN REPEATEDLY RE-RUN "sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367"
>
> Executing: /tmp/apt-key-gpghome.EavCgH8G3s/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.EavCgH8G3s/S.dirmngr' failed: IPC connect call failed
gpg: keyserver receive failed: No dirmngr

> iiab-admin@u18-server:~$ which dirmngr
/usr/bin/dirmngr
iiab-admin@u18-server:~$ dirmngr --version
dirmngr (GnuPG) 2.2.4
...

holta, 2019-07-01:
On my 3rd try running `sudo iiab` both above problems went away fyi.

So we should prob close this issue, but I'll leave it open for a bit so that @mitra42 can weigh in, just in case there are better workarounds to consider the next time this happens!

mitra42, 2019-07-01:
This looks like its during the install of yarn, i.e. running the yarn playbook  rather than the internetarchive playbook, can you confirm ? 

If that is the case then it makes sense, ....

Yarn is tricky to install - there is an older package called "yarn" that is in the debian repo, in `iiab/roles/yarn/tasks/main.yml` it has to add the apt-key and then add the yarn source from its own repo, update apt's cache and then install yarn.  The ansible role for `yarn` was a standard one I grabbed from somewhere (not sure where). 

It looks to me like ubuntu is complaining about ansible's "apt-key" process. I see there are two versions of ansible in iiab/scripts. (2.7 n 2.8) and don't know why (this was the first time I'd used ansible). 

Who is the ansible expert among your team, maybe its a standard fix to make ansible do `apt-key` correctly ? 

Note - my diagnosis could be wrong, since I'm not an ansible or apt expert.

holta, 2019-07-02:
This seems to be a failure of apt-key to download https://dl.yarnpkg.com/debian/pubkey.gpg which is certainly annoying, but seems intermittent and (largely) out of our control.

Whereas the apparent extreme slowdown of `yarn add @internetarchive/dweb-archive @internetarchive` in recent days, might be something we can actually improve upon — to keep the base installation reasonably lean — where possible?

https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/main.yml#L21-L25

Ref PRs #1780 #1806

mitra42, 2019-07-02:
Ok - I'm not replicating the pubkey issue - it downloads quickly for me - if that turns into a significant issue then we could probably just include the key in the IIAB repo, though would need an ansible-artist to figure out how to make yarn/tasks/main.yml read a local key

holta, 2019-07-02:
> Ok - I'm not replicating the pubkey issue - it downloads quickly for me - if that turns into a significant issue then we could probably just include the key in the IIAB repo, though would need an ansible-artist to figure out how to make yarn/tasks/main.yml read a local key

Agreed.  Let's close this for now, as it's hopefully a very rare condition &mdash; re-opening it later to solve it methodically with Ansible later, if truly necessary.

closed 2019-07-02 by holta