iiab/iiab #1802: http://box/mediawiki "Cannot access the database" on fresh IIAB install

holta, 2019-07-01:
This is a fresh install of BIG-sized IIAB 7.0 pre-release (http://download.iiab.io) on Ubuntu Server 18.04 (VM) that was rebooted per the install instructions.

Even 10 min after booting, http://10.8.0.30/mediawiki/ and http://192.168.0.x/mediawiki/ keep showing:
```
Sorry! This site is experiencing technical difficulties.
Try waiting a few minutes and reloading.

(Cannot access the database)
```

@tim-moody do you have suggestions?

holta, 2019-07-01:
@tim-moody can you tell what might be causing these underlying MariaDB failures on 10.8.0.30, by logging in?  (Ubuntu Server 18.04 VM, with BIG-sized IIAB 7.0 pre-release freshly installed)

```
root@box:~# systemctl status mysql
● mariadb.service - MariaDB 10.1.40 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Sun 2019-06-30 23:22:49 EDT; 2min 39s ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 639 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 450 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP
  Process: 447 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 430 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 639 (code=exited, status=0/SUCCESS)
   Status: "MariaDB server is down"

Jun 30 23:19:27 box.lan systemd[1]: Starting MariaDB 10.1.40 database server...
Jun 30 23:19:50 box.lan mysqld[639]: 2019-06-30 23:19:49 140426294455424 [Note] /usr/sbin/mysqld (mysqld 10.1.40-MariaDB-0ubuntu0.18.04.1) starting as process 639 ...
Jun 30 23:21:09 box.lan systemd[1]: mariadb.service: Start operation timed out. Terminating.
Jun 30 23:22:39 box.lan systemd[1]: mariadb.service: State 'stop-sigterm' timed out. Skipping SIGKILL.
Jun 30 23:22:49 box.lan systemd[1]: mariadb.service: Failed with result 'timeout'.
Jun 30 23:22:49 box.lan systemd[1]: Failed to start MariaDB 10.1.40 database server.
```

holta, 2019-07-02:
1) FYI manually forcing `systemctl start mysql` acts as a temporary workaround.  (Whereupon things like `systemctl status mysql` and http://box/mediawiki start to work...)

   But then mysql fails during each subsequent reboots (with the same message pasted onto the top of this #1802 ticket) so manual overrides are by no means a solution...

2) Does anyone have ideas to look into this 10.8.0.30 properly (rebooting if nec) to solve this?

holta, 2019-07-02:
@jvonau do you see any kind of flaw in MySQL's 152-line systemd unit file (`/lib/systemd/system/mariadb.service`) in this pastebin?

### http://paste.ubuntu.com/p/zMjY6qMfQm/

Background...showing that both `mysql.service` and `mysqld.service` link to the above file:

> root@box:/opt/iiab/iiab# ls -l /etc/systemd/system/mysql.service
lrwxrwxrwx 1 root root 35 Jun 30 20:55 /etc/systemd/system/mysql.service -> /lib/systemd/system/mariadb.service

root@box:/opt/iiab/iiab# ls -l /etc/systemd/system/mysqld.service
lrwxrwxrwx 1 root root 35 Jun 30 20:55 /etc/systemd/system/mysqld.service -> /lib/systemd/system/mariadb.service

tim-moody, 2019-07-02:
from the unit file:
`
# Kernels like killing mysqld when out of memory because its big.
# Lets temper that preference a little.
# OOMScoreAdjust=-600
`

how much memory does your vm have?

tim-moody, 2019-07-02:
the error message is from mediawiki, not mysql

tim-moody, 2019-07-02:
https://stackoverflow.com/questions/40997257/mysql-service-fails-to-start-hangs-up-timeout-ubuntu-mariadb

holta, 2019-07-02:
> how much memory does your vm have?

2 GB as seen here, which FWIW is much more than 512MB and 1GB on Raspberry Pis:

> root@box:~# free -h
              total        used        free      shared  buff/cache   available
Mem:           1.9G        802M        160M        8.9M        1.0G        1.0G
Swap:          947M          0B        947M

Those particular AppArmor suggestions don't apply:

> https://stackoverflow.com/questions/40997257/mysql-service-fails-to-start-hangs-up-timeout-ubuntu-mariadb

...as AppArmor is off:

> root@box:~# systemctl status apparmor
● apparmor.service - AppArmor initialization
   Loaded: loaded (/lib/systemd/system/apparmor.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:apparmor(7)
           http://wiki.apparmor.net/

Uncommenting the following (in `/lib/systemd/system/mariadb.service`) and rebooting did not work:

> OOMScoreAdjust=-600

Specifically, `systemctl status mysql` shows the same timeout failure as pasted near the very top of this tkt :/

holta, 2019-07-03:
@jvonau's suggestion worked &mdash; look at log files in `/var/log/mysql` and insert the following into `/lib/systemd/system/mariadb.service`, directly under the `[Service]` heading:

```
# 2019-07-03: @jvonau @holta doubled the default boot timeout from 90s to 180s
# for slow machines like this Ubuntu 18.04.2 VM:
# https://github.com/iiab/iiab/issues/1802
# https://mariadb.com/kb/en/library/what-to-do-if-mariadb-doesnt-start/#systemd
TimeoutStartSec=180
```

I'll work on a PR to encode this into https://github.com/iiab/iiab/blob/master/roles/mysql

holta, 2019-07-03:
Aside: we should also ask ourselves why BIG-sized Ubuntu 18.04.2 VM's are much slower than in the past.

Is it possible the `-c` (crawl) flag recent [recently added](https://github.com/iiab/iiab/pull/1780#issuecomment-507451425) to our `internetarchive` playbook might be related?

PR #1813 (2019-07-03): Fix for MySQL timing out on boot: put TimeoutStartSec=180 in mariadb.service
Fixes #1802 "MySQL (MariaDB 10.1.40) fails on every boot, Ubuntu Server 18.04 VM, fresh IIAB BIG-sized install [so http://box/mediawiki "Cannot access the database" etc]"
closed 2019-07-03 by holta