iiab/iiab #1805: Lokole 0.4.3 transient failure during install (on Raspberry Pi 3 and other OS's too?)

holta, 2019-07-01:
@c-w just fyi: is there something that can/should be made more reliable during Lokole installation, so as to avoid the need to re-run the installation twice as I did below?

FWIW this was a reinstall/upgrade from 0.4.2 to 0.4.3 on RPi 3 @ 10.8.0.26 using `cd /opt/iiab/iiab; ./iiab-install --reinstall`

It succeeded the 2nd time I tried it.  Aside: I also could have run `cd /opt/iiab/iiab; ./runrole lokole` but running complete IIAB reinstalls are useful for general QA (:

During the 1st attempt, the error was:

> TASK [lokole : pip install opwen_email_client (Lokole) 0.4.3 from PyPI to /library/lokole/venv] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/library/lokole/venv/bin/pip", "install", "--no-cache-dir", "opwen_email_client==0.4.3"], "msg": "stdout: Collecting opwen_email_client==0.4.3\n  Downloading https://files.pythonhosted.org/packages/3f/3a/1ecb56ab5dc0b29f95283ec6bfd166f8b607e7f6dd20601455abedc7b915/opwen_email_client-0.4.3.tar.gz (602kB)\nCollecting Babel==2.7.0 (from opwen_email_client==0.4.3)\n  Downloading https://files.pythonhosted.org/packages/2c/60/f2af68eb046c5de5b1fe6dd4743bf42c074f7141fe7b2737d3061533b093/Babel-2.7.0-py2.py3-none-any.whl (8.4MB)\nRequirement already satisfied: Flask-BabelEx==0.9.3 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nCollecting Flask-Caching==1.7.2 (from opwen_email_client==0.4.3)\n  Downloading https://files.pythonhosted.org/packages/bd/4e/c701bdcd566f26187398cf058ad2ca14c8b2f50b17a17f9076794ae2f8db/Flask_Caching-1.7.2-py2.py3-none-any.whl\nCollecting Flask-Migrate==2.5.2 (from opwen_email_client==0.4.3)\n  Downloading https://files.pythonhosted.org/packages/4b/22/d20a105f13c58fe590ef38ac59d5cabc1a917549ea8f90b0ca7b4e84f6be/Flask_Migrate-2.5.2-py2.py3-none-any.whl\nRequirement already satisfied: Flask-SQLAlchemy==2.4.0 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nRequirement already satisfied: Flask-Script==2.0.6 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nRequirement already satisfied: Flask-Security==3.0.0 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nRequirement already satisfied: Flask-Testing==0.7.1 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nRequirement already satisfied: Flask-WTF==0.14.2 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nRequirement already satisfied: Flask==1.0.3 in /library/lokole/venv/lib/python3.5/site-packages (from opwen_email_client==0.4.3)\nCollecting SQLAlchemy==1.3.5 (from opwen_email_client==0.4.3)\n\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/library/lokole/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 594, in urlopen\n    chunked=chunked)\n  File \"/library/lokole/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 391, in _make_request\n    six.raise_from(e, None)\n  File \"<string>\", line 2, in raise_from\n  File \"/library/lokole/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 387, in _make_request\n    httplib_response = conn.getresponse()\n  File \"/usr/lib/python3.5/http/client.py\", line 1198, in getresponse\n    response.begin()\n  File \"/usr/lib/python3.5/http/client.py\", line 297, in begin\n    version, status, reason = self._read_status()\n  File \"/usr/lib/python3.5/http/client.py\", line 266, in _read_status\n    raise RemoteDisconnected(\"Remote end closed connection without\"\nhttp.client.RemoteDisconnected: Remote end closed connection without response\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/commands/install.py\", line 342, in run\n    requirement_set.prepare_files(finder)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/req/req_set.py\", line 554, in _prepare_file\n    require_hashes\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/req/req_install.py\", line 278, in populate_link\n    self.link = finder.find_requirement(self, upgrade)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/index.py\", line 792, in get_page\n    \"Cache-Control\": \"max-age=600\",\n  File \"/library/lokole/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 501, in get\n    return self.request('GET', url, **kwargs)\n  File \"/library/lokole/venv/lib/python3.5/site-packages/pip/download.py\", line 386, in request\n    return super(PipSession, self).request(method, url, *args, **kwargs)\n  File \"/library/lokole/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 488, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/library/lokole/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 609, in send\n    r = adapter.send(request, **kwargs)\n  File \"/library/lokole/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py\", line 423, in send\n    timeout=timeout\n  File \"/library/lokole/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 643, in urlopen\n    _stacktrace=sys.exc_info()[2])\n  File \"/library/lokole/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py\", line 315, in increment\n    total -= 1\nTypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n"}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=269  changed=58   unreachable=0    failed=1    skipped=92   rescued=0    ignored=0

Refs PRs #1803 #1804

holta, 2019-07-02:
@c-w I don't even understand this `pip install opwen_email_client` error arising from https://github.com/iiab/iiab/blob/master/roles/lokole/tasks/install.yml#L15-L24, do you?

(And/or if you believe this a very rare condition, perhaps we can close this ticket for now?)

c-w, 2019-07-03:
@holta I have a vague memory of seeing this behavior with old versions of pip and setuptools. Is there any specific guarantee on IIAB about the versions being up to date? If not, it may be useful to upgrade them ([as we do in the RaspberryPi installer](https://github.com/ascoderu/opwen-webapp/blob/01479e0d8f9a712c201e1baf6968911c880244a6/install.py#L423)).

holta, 2019-07-03:
> @holta I have a vague memory of seeing this behavior with old versions of pip and setuptools. Is there any specific guarantee on IIAB about the versions being up to date? If not, it may be useful to [...]

In this case that doesn't appear to be the issue, as Lokole 0.4.3 installs even with pip 8.1.1 and setuptools 20.7.0 on Ubuntu 16.04 (IIAB has not supported Ubuntu 16.04 in over a year now, just an observation!)

But on this topic, IIAB 7.0 is [mandating new OS's / distros](https://github.com/iiab/iiab/wiki/IIAB-Platforms) like these 3 especially &mdash; so there should be no concerns here:
- Raspbian Buster offers: pip 18.1 + setuptools 40.8.0
- Debian 10 offers: pip 18.1 + setuptools 40.8.0
- Ubuntu 18.04.2 offers: pip 9.0.1 + setuptools 39.0.1

(e.g. using commands `pip --version` and `python -m easy_install --version` to show version numbers.)

holta, 2019-07-04:
Let's blame cosmic rays for now ;)

(And re-open this tkt in future if/as it's determined to affect IIAB/Lokole installs more broadly...)

closed 2019-07-04 by holta