iiab/iiab #1807: Will Kolibri 0.12.6 be ready early July? Validate it after its .pex pip installer is published

holta, 2019-07-02:
Kolibri 0.17.6 is converging here @https://github.com/learningequality/kolibri/milestone/42 &mdash; hypothetically/possibly even being released in early July in time for [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes)?

Building off #1736

indirectlylit, 2019-07-24:
Just wanted to alert that Kolibri 0.12.6 is currently in beta. 

It's a pretty minor patch update and most changes are cosmetic, but there are a few performance improvements also.

You can see the issues closed here:
https://github.com/learningequality/kolibri/issues?q=is%3Aissue+milestone%3A0.12.6+is%3Aclosed

holta, 2019-07-25:
> Just wanted to alert that Kolibri 0.12.6 is currently in beta.

Thanks!  Time to move it from beta to gamma ;)

Likewise [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) RC1 (Release Candidate 1) will be published in coming hours~

holta, 2019-07-26:
0.12.6 released:
https://github.com/learningequality/kolibri/releases/tag/v0.12.6
https://github.com/learningequality/kolibri/blob/develop/CHANGELOG.md#0126

Hopefully its .pex file (for pip install) will appear here soon:
https://learningequality.org/r/kolibri-pex-latest

indirectlylit, 2019-07-27:
FYI, we had a regression in 0.12.6 and are quickly moving on to 0.12.7

holta, 2019-07-30:
0.12.7 pip install (.pex) is now available from:
https://learningequality.org/download/

holta, 2019-08-01:
@KW4NP @arky thanks all for continued testing of [Kolibri 0.12.7](https://github.com/learningequality/kolibri/releases/tag/v0.12.7) as part of [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes), all of which will be officially released in coming days!

closed 2019-08-01 by holta