iiab/iiab #1810: Python 2 -> Python 3 IIAB strategic planning to prepare for 2020+

holta, 2019-07-03:
https://pythonclock.org excerpts:

> Python 2.7 [will not be maintained past 2020](https://www.python.org/dev/peps/pep-0373/). Originally, there was no official date. Recently, that date has been updated to [January 1, 2020](https://github.com/python/devguide/pull/344).

> If the code you care about is still on Python 2, that's totally understandable. Most of PyPI's popular packages now [work on Python 2 and 3](https://python3wos.appspot.com/), and more are being added every day. Additionally, a number of critical Python projects have [pledged to stop supporting Python 2 soon](http://python3statement.org/). To ease the transition, the [official porting guide](https://docs.python.org/3/howto/pyporting.html) has advice for running Python 2 code in Python 3.

tim-moody, 2019-07-08:
biggest issue is switch from str and unicode types to bytes and str types.

traverseda, 2019-07-11:
Do we have a list of code that needs to be updated?

holta, 2019-07-12:
> Do we have a list of code that needs to be updated?

These are the 4+8=12 that matter, if you have time to assess any quick?

(1) Arising from `grep -rl '#!.*/python' /opt/iiab/iiab` we have:

- [roles/kiwix/templates/iiab-make-kiwix-lib.py](https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/iiab-make-kiwix-lib.py)
- [roles/kiwix/templates/iiab-make-apache-config.py](https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/iiab-make-apache-config.py)
- [roles/1-prep/templates/iiab_env.py.j2](https://github.com/iiab/iiab/tree/master/roles/1-prep/templates/iiab_env.py.j2)
- [roles/network/templates/captive-portal/captive-portal.py.j2](https://github.com/iiab/iiab/tree/master/roles/network/templates/captive-portal/captive-portal.py.j2)

(2) Arising from `grep -rl '#!.*/python' /opt/iiab/iiab-admin-console` we have:

- [roles/cmdsrv/files/iiab-cmdsrv](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/iiab-cmdsrv)
- [roles/cmdsrv/files/scripts/get_kiwix_lib](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/scripts/get_kiwix_lib)
- [roles/cmdsrv/templates/scripts/get_oer2go_catalog](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/get_oer2go_catalog)
- [roles/cmdsrv/templates/scripts/iiab_update_menus.py](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/iiab_update_menus.py)
- [roles/cmdsrv/templates/scripts/get_kiwix_catalog](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/get_kiwix_catalog)
- [roles/cmdsrv/templates/iiab-cmdsrv-ctl.j2](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/iiab-cmdsrv-ctl.j2)
- [roles/cmdsrv/files/scripts/get_inet_speed](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/scripts/get_inet_speed)
- [roles/js-menu/utilities/menu-def-converter](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/utilities/menu-def-converter)

P.S. @tim-moody has done a preliminary conversion of [cmdsrv](https://github.com/iiab/iiab-admin-console/tree/master/roles/cmdsrv).  As mentioned above (4 days ago) the next step there needs a knowledge of bytes vs str and ZeroMQ.

traverseda, 2019-07-12:
Well section 1 looks like it shouldn't be a problem, those can probably just be switched to python3 directly without any errors. The one I'd be most worried about is `captive-portal.py.j2`

I've been thinking about putting together an IIAB for myself for a while, so when I stand that up I should be able to confirm.

Cmdsrv does look pretty complicated, I'd rather avoid touching it if there's someone who's actually familiar with the code base working on it.

Section 2 is a lot more/denser code, and while nothing jumps out at me as python2 only it would need to be gone over with a finer comb.

Now that I know what files are relevant when I build my IIAB node I can work on that.

If I change those files from `#!/usr/bin/python` to `#!/usr/bin/python3` they will actually run under python 3? Do you know where they'd log to?

holta, 2019-07-12:
> The one I'd be most worried about is `captive-portal.py.j2`

@georgejhunt do you have any recommendations above?

> I've been thinking about putting together an IIAB for myself for a while, so when I stand that up I should be able to confirm.

We highly recommend the Raspberry Pi 4 (if those are available in Nova Scotia ;)  But a VM or old PC running Debian 10 or Ubuntu 18.04+ should also work &mdash; even a pre-release [daily build of Ubuntu 19.10](http://cdimage.ubuntu.com/daily-live/current/) if you're daring: #1714

> If I change those files from #!/usr/bin/python to #!/usr/bin/python3 they will actually run under python 3? Do you know where they'd log to?

Let me ask others like @tim-moody & @jvonau to see if they know-

traverseda, 2019-07-12:
Probably going to throw it on an odroid-xu4 I have kicking around.

holta, 2019-07-13:
> Probably going to throw it on an odroid-xu4 I have kicking around.

What OS likely?

Keep an eye on these HW/OS options that stand the best chance of working!
https://github.com/iiab/iiab/wiki/IIAB-Platforms

traverseda, 2019-07-13:
Armbian Sid if I can, armbian buster if I can't.

floydianslips, 2019-07-16:
Hey @traverseda, I have had some luck with DietPi as well on Orange Pi's. You could give that a shot on your Odroid. I know they have an image for it.

holta, 2019-09-26:
FYI @tim-moody is now working on converting [Admin Console](https://github.com/holta/iiab-admin-console) (and its ZMQ?) from Python 2 to Python 3.

holta, 2019-09-26:
@tim-moody is reviewing our options under "3. Python 3 Migration" under September 26, 2019 at http://minutes.iiab.io

Excerpt:

> a. It’s later than I realized
b. Start by only writing any new scripts in 3
c. Gain experience with 2to3
d. Gain experience with byte/string
e. Create /usr/local/lib/python3.x/dist-packages/iiab containing iiab_lib.py (all functions needed only by iiab) and iiab_admin_lib.py (containing functions needed by adm cons)
f. How can we avoid the version in the path?
g. Should we add docstrings?
h. Should we use conda?
i. Should we always use virtualenv?

tim-moody, 2019-09-27:
turns out that  /usr/lib/python3/dist-packages exists and is in python path

tim-moody, 2019-09-27:
George prefers #! /usr/bin/env python which is not caught by the above grep
grep -rl '#!.*python' /opt/iiab/iiab yields:

/opt/iiab/iiab/roles/kiwix/templates/iiab-make-apache-config.py
/opt/iiab/iiab/roles/kiwix/templates/iiab-make-kiwix-lib.py
/opt/iiab/iiab/roles/network/templates/captive-portal/captive-portal.py.j2
/opt/iiab/iiab/roles/captive-portal/templates/capture-wsgi.py
/opt/iiab/iiab/roles/1-prep/templates/iiab_env.py.j2
/opt/iiab/iiab/roles/activity-server/files/bin/xs-check-activities
/opt/iiab/iiab/roles/activity-server/files/bin/xs-regenerate-activities
/opt/iiab/iiab/roles/osm-vector-maps/templates/iiab-update-map
/opt/iiab/iiab/scripts/roles-needing-docs.py

grep -rl '#!.*python' /opt/iiab/iiab-admin-console yields:
/opt/iiab/iiab-admin-console/roles/js-menu/utilities/menu-def-converter
/opt/iiab/iiab-admin-console/roles/cmdsrv/files/iiab-cmdsrv
/opt/iiab/iiab-admin-console/roles/cmdsrv/files/scripts/get_inet_speed
/opt/iiab/iiab-admin-console/roles/cmdsrv/files/scripts/get_kiwix_lib
/opt/iiab/iiab-admin-console/roles/cmdsrv/templates/iiab-cmdsrv-ctl.j2
/opt/iiab/iiab-admin-console/roles/cmdsrv/templates/scripts/iiab_update_menus.py
/opt/iiab/iiab-admin-console/roles/cmdsrv/templates/scripts/get_oer2go_catalog
/opt/iiab/iiab-admin-console/roles/cmdsrv/templates/scripts/get_kiwix_catalog

grep -rl '#!.*python' /opt/iiab/iiab-factory/
/opt/iiab/iiab-factory/content/kiwix/zim-list.txt
/opt/iiab/iiab-factory/content/kiwix/cp-zim
/opt/iiab/iiab-factory/content/kiwix/zim-concatenator.py
/opt/iiab/iiab-factory/content/osm-tiles/import.py
/opt/iiab/iiab-factory/content/subset-osm/mk-subset.py
/opt/iiab/iiab-factory/content/khan/ka-find
/opt/iiab/iiab-factory/content/khan/ka-vids
/opt/iiab/iiab-factory/content/khan/ka-scan
/opt/iiab/iiab-factory/content/rachel/gen-rachel-menus
/opt/iiab/iiab-factory/content/rachel/gen-rachel-menus-html
/opt/iiab/iiab-factory/content/rachel/get-rachel-menu-blocks
/opt/iiab/iiab-factory/box/generic/merge_local_vars.sh

holta, 2019-10-20:
Below are two serious Python 3 issues we're now facing, with Python 2.7 being end-of-life'd in just over 2 months: (https://pythonclock.org)

- [ ] #2003 CMDSRV requires more than python-systemd -> python3-systemd to run on Ubuntu 19.10
- [ ] #2013 Can python3-geojson be made to work, as required by OS's like Ubuntu 19.10+? [osm-vector-maps]

@jvonau @tim-moody @georgejhunt @traverseda do you know of others that should be added to the list?

holta, 2019-10-23:
@jvonau recommends these 2 Ansible guides w.r.t. transitioning to Python 3:

- https://docs.ansible.com/ansible/latest/reference_appendices/python_3_support.html
- https://docs.ansible.com/ansible/latest/dev_guide/developing_python_3.html

holta, 2019-10-23:
FYI @jvonau even the very latest Ansible (2.9.0rc5) currently being installed as part of IIAB's master ends up using Python 2.7 not Python 3 -- see the notes just above for background (e.g. serious bugs in Ansible 2.8.3 & 2.8.6 mentioned at #1737 

In any case, here is the warning shown when running ./iiab-install on Ubuntu 19.10:

> [DEPRECATION WARNING]: Distribution Ubuntu 19.10 on host 127.0.0.1 should use
/usr/bin/python3, but is using /usr/bin/python for backward compatibility with
prior Ansible releases. A future Ansible release will default to using the
discovered platform python for this host. See https://docs.ansible.com/ansible/
2.9/reference_appendices/interpreter_discovery.html for more information. This
feature will be removed in version 2.12. Deprecation warnings can be disabled
by setting deprecation_warnings=False in ansible.cfg.

holta, 2019-10-24:
See the last 2 lines of PR #2019 where @jvonau is prototyping an `./iiab-install` command that uses Ansible with Python 3:

> ansible -m setup -i $INVENTORY localhost --connection=local -e ansible_python_interpreter=/usr/bin/python3 | grep python
> ansible-playbook -i $INVENTORY $PLAYBOOK --connection=local -e ansible_python_interpreter=/usr/bin/python3

@jvonau also mentions:

> so the -e would go into iiab-install, runrole, ... 
>
> we would need to install some python3 support debs that we have identified in the issues, in the 3 ansible install files
>
> maybe not
>> sudo apt list python3-distutils*
>> Listing... Done
>> python3-distutils-extra/stable 2.42 all
>> python3-distutils/stable,now 3.7.3-1 all [installed,automatic]
>
> maybe just
>> sudo apt list python3-pymysql
>> Listing... Done
>> python3-pymysql/stable 0.9.3-1 all
>
> yup
>> TASK [mysql-enable : Update MySQL root password for localhost root accounts, if mysql_enabled] **
>> fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "The PyMySQL (Python 2.7 and Python 3.X) or MySQL-python (Python 2.X) module is required."}
>>
>> PLAY RECAP **
>> 127.0.0.1                  : ok=25   changed=6    unreachable=0    failed=1    skipped=13   rescued=0    ignored=0   
>
> works now on rpi
>
> and runs to completion

jvonau, 2019-10-28:
> FYI @jvonau even the very latest Ansible (2.9.0rc5) currently being installed as part of IIAB's master ends up using Python 2.7 not Python 3 -- see the notes just above for background (e.g. serious bugs in Ansible 2.8.3 & 2.8.6 mentioned at #1737
> 
> In any case, here is the warning shown when running ./iiab-install on Ubuntu 19.10:
> 
> > [DEPRECATION WARNING]: Distribution Ubuntu 19.10 on host 127.0.0.1 should use
> > /usr/bin/python3, but is using /usr/bin/python for backward compatibility with
> > prior Ansible releases. A future Ansible release will default to using the
> > discovered platform python for this host. See https://docs.ansible.com/ansible/2.9/reference_appendices/interpreter_discovery.html
> > for more information. This feature will be removed in version 2.12. Deprecation
> > warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.

This is a result of having installed python2.7 prior to the mysql crash (due to the missing package python3-pymysql) on the first run of anisble. Now when run a second time ansible sees the newly installed python2.7 and fallsback to using that, resulting in the mysql crash mysteriously going away running python2.7.

holta, 2019-10-30:
> mod_wsgi for apache can only be python 2 OR 3, not both. it is used by
> captive portal
> cmdsrv
> pathagar

Good to know.  This relates to Captive Portal work at #1182 that @georgejhunt is looking into.

jvonau, 2019-10-30:
> mod_wsgi for apache can only be python 2 OR 3, not both. it is used by
> captive portal
> cmdsrv
> pathagar

In the short term we get python2 out of the box on Raspbian and apache is the default webserver. However on Ubuntu what is needed on a pure python3 distro is to change the pip call to use python3 and tell pip to use python2 in the venv. Now python2.7 is not installed but we install what is needed in packages.yml for kalite and others' venv that are installed later, those four packages should be changed to python3. What program needs python-passlib?  Perhaps cmdsrv could use a venv to contain the python2 dependencies? 

The nginx changes could make mod_wsgi python3 only as uwsgi is used with nginx, which I see has new python dependency. Think pathagar and all other unused/unmaintained roles that require python2 be moved out of the main line code.

jvonau, 2019-11-10:
#2034 addresses a short term need for vector-maps to install on Ubuntu-19.10 and provide backwards support for older python2 scripts till porting is complete.

holta, 2019-11-11:
"Python 2 removal in sid/bullseye: Progress and next steps"
https://lists.debian.org/debian-devel-announce/2019/11/msg00000.html

holta, 2019-11-20:
PR #2043 merged to facilitate this effort.

See @tim-moody's clarification of @jvonau's related work, at https://github.com/iiab/iiab/pull/2043#issuecomment-555832836

holta, 2019-11-20:
See also IIAB Admin Console's ongoing conversion to Python 3:

https://github.com/iiab/iiab-admin-console/pull/263 "0.4 python3 Convert Admin Console to Python 3 - WIP"

holta, 2019-11-20:
FYI in support of things like PR #2042, @jvonau has posted a request to IBM/Red Hat's Ansible developers here:

https://github.com/ansible/ansible/issues/57342#issuecomment-556219871 "Please provide packages for python3 in PPA"

holta, 2019-12-21:
At what stage do we declare victory?

holta, 2020-01-09:
@jvonau suggests leaving this tkt open pending Ansible & Nodesource.com full transition to Python 3.

holta, 2020-01-09:
Also see #2105

holta, 2020-02-03:
Related: #2231 "Should Calibre-Web use Python 3 instead of 2.7 ?"

jvonau, 2020-02-03:
Better question is what about kalite, also python2.

holta, 2020-02-03:
> Better question is what about kalite, also python2.

@benjaoming what's the summary status here do you know?

benjaoming, 2020-02-03:
@holta KA Lite will not support Python 3 in this current millennium :)

holta, 2020-04-08:
Moving this ticket from IIAB 7.1 milestone to 7.2, to keep organized around any residual Python 2 -> 3 issues.

holta, 2020-09-12:
Related:

#2479 "Validate Ansible 2.10 expected 2020-08-13 [with simplified auto-installation of "Collections" ETA 2020-09-22 ?]"
#2512 "Revamp Ansible for the future"